esp8266/Arduino #1: Update to SDK 1.0.0

4refr0nt, 2015-03-28:
Need to update from SDK 0.9.5 to SDK 1.0.0

rogerclarkmelbourne, 2015-03-29:
Sorry. I didn't see that issue before I added mine

BTW. I did briefly try replacing the 0.9 SDK in the repo with the 1.0 SDK but got lots of errors, missing at least 1 header and also missing some internal functions that were in 0.9 
(Hard to know if they have been removed of just are not in the headers in 1.0 :-(

igrr, 2015-03-29:
No problem. 
It seems there is a DHCP client issue in 0.9.5, so upgrading to 1.0 does make sense.
If you post the compiler output I may be able to guide you, alternatively I'll look into this myself in the next few days. The second route would probably be easier.

rogerclarkmelbourne, 2015-03-29:
OK. Looks like quite a few things are different

C:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -IC:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266/tools/sdk//include -c -Os -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10601 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -IC:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266\esp8266\cores\esp8266 -IC:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266\esp8266\variants\esp01 C:\Users\rclark\AppData\Local\Temp\build5447791988060577372.tmp\sketch_mar29a.cpp -o C:\Users\rclark\AppData\Local\Temp\build5447791988060577372.tmp\sketch_mar29a.cpp.o 
C:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-as -o C:\Users\rclark\AppData\Local\Temp\build5447791988060577372.tmp\cont.S.o C:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266\esp8266\cores\esp8266\cont.S 
C:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -D__ets__ -DICACHE_FLASH -IC:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266/tools/sdk//include -c -Os -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -MMD -std=c99 -DF_CPU=80000000L -DARDUINO=10601 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -IC:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266\esp8266\cores\esp8266 -IC:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266\esp8266\variants\esp01 C:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266\esp8266\cores\esp8266\cont_util.c -o C:\Users\rclark\AppData\Local\Temp\build5447791988060577372.tmp\cont_util.c.o 
C:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -D__ets__ -DICACHE_FLASH -IC:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266/tools/sdk//include -c -Os -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -MMD -std=c99 -DF_CPU=80000000L -DARDUINO=10601 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -IC:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266\esp8266\cores\esp8266 -IC:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266\esp8266\variants\esp01 C:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266\esp8266\cores\esp8266\core_esp8266_noniso.c -o C:\Users\rclark\AppData\Local\Temp\build5447791988060577372.tmp\core_esp8266_noniso.c.o 
C:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -D__ets__ -DICACHE_FLASH -IC:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266/tools/sdk//include -c -Os -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -MMD -std=c99 -DF_CPU=80000000L -DARDUINO=10601 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -IC:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266\esp8266\cores\esp8266 -IC:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266\esp8266\variants\esp01 C:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266\esp8266\cores\esp8266\core_esp8266_wiring.c -o C:\Users\rclark\AppData\Local\Temp\build5447791988060577372.tmp\core_esp8266_wiring.c.o 
In file included from C:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266/tools/sdk//include/ets_sys.h:11:0,
                 from C:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266\esp8266\cores\esp8266\core_esp8266_wiring.c:24:
C:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266/tools/sdk//include/c_types.h:15:29: error: conflicting types for 'uint32_t'
 typedef unsigned long       uint32_t;
                             ^
In file included from c:\users\rclark\documents\arduino\hardware\arduino-esp8266\tools\xtensa-lx106-elf\lib\gcc\xtensa-lx106-elf\4.8.2\include\stdint.h:9:0,
                 from C:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266\esp8266\cores\esp8266\Arduino.h:24,
                 from C:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266\esp8266\cores\esp8266\wiring_private.h:31,
                 from C:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266\esp8266\cores\esp8266\core_esp8266_wiring.c:23:
c:\users\rclark\documents\arduino\hardware\arduino-esp8266\tools\xtensa-lx106-elf\xtensa-lx106-elf\include\stdint.h:81:22: note: previous declaration of 'uint32_t' was here
 typedef unsigned int uint32_t;
                      ^
In file included from C:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266/tools/sdk//include/ets_sys.h:11:0,
                 from C:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266\esp8266\cores\esp8266\core_esp8266_wiring.c:24:
C:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266/tools/sdk//include/c_types.h:17:29: error: conflicting types for 'int32_t'
 typedef signed long         int32_t;
                             ^
In file included from c:\users\rclark\documents\arduino\hardware\arduino-esp8266\tools\xtensa-lx106-elf\lib\gcc\xtensa-lx106-elf\4.8.2\include\stdint.h:9:0,
                 from C:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266\esp8266\cores\esp8266\Arduino.h:24,
                 from C:\Users\rclark\Documents\Arduino\hardware\arduino-esp8266\esp8266\cores\esp8266\wiring_private.h:31,
[... cut ...]
c:\users\rclark\documents\arduino\hardware\arduino-esp8266\tools\xtensa-lx106-elf\lib\gcc\xtensa-lx106-elf\4.8.2\include\stdbool.h:35:0: note: this is the location of the previous definition
 #define false 0
 ^
Error compiling.

Unfortunatly its getting really late here (almost 10:30pm) and I'm not a night owl, so I will need to call it a day and take another look tomorrow about why there seems to be so many issues with typedefs

PS. I better check I have a clean version of the 1.0SDK as I got it from www.arduinoesp.com and his build for Arduino (which was instantly made redundant when you released your much better version ;0)

Thanks

Roger

closed 2015-03-29 by igrr

igrr, 2015-03-29:
Here's the new bundle: https://ci.appveyor.com/api/buildjobs/wcq3nf7ni2lfp2sm/artifacts/build/windows/arduino-1.6.1-windows.zip

igrr, 2015-03-29:
Will post the link to the new bundle when build completes.

rogerclarkmelbourne, 2015-03-30:
Many thanks

I'll clone your whole repo to my local machine and copy the necessary hardware files to my local repo and then push my hardware files only stuff back to my repo

Cheers
Roger

igrr, 2015-03-30:
@rogerclarkmelbourne Why do we need a separate repository? Wouldn't it be easier to add a simple shell script which would package the necessary files, and then run this as a build script? Every commit is built on Travis CI and Appveyor, so it should be easy to build the right package there.
Would you like to contribute such a script? :)

rogerclarkmelbourne, 2015-03-30:
I agree

We don't need a separate repo.

Where are your Xtensia compilor exe files kept? I can't see them in you repo. :-(

I'd need to build the script to bundle the compilor files into the tools/xtensa-lx106-elf folder etc

But I'm currently using a different set of compilor files, because of the issue in your initial Windows 1.6.1-p1 release with libiconv-2.dll not being found by gcc

igrr, 2015-03-30:
If you check the build.xml you'll see that during the build the archives are downloaded from the following URLs:

http://download.igrr.me/osx-xtensa-lx106-elf.tgz
http://download.igrr.me/linux64-xtensa-lx106-elf.tgz
http://download.igrr.me/win32-xtensa-lx106-elf.tgz

This looks like a good layout for all the tools: https://github.com/sandeepmistry/esp8266-Arduino

igrr, 2015-03-30:
@rogerclarkmelbourne 
I have created a new issue for building the package (#14), let's leave this one closed.

rogerclarkmelbourne, 2015-03-30:
Thanks.

I'll take a look at those links.

Please close this one ;-)

commit b64079f5 (2015-03-29): Update to SDK 1.0.0

Fixes #1