esp8266/Arduino #2: add support for Arduino Eclipse Plugin

Links2004, 2015-03-28:
add support for Arduino Eclipse Plugin (http://www.baeyens.it/eclipse/)

change the platform.txt, the Compiling is working.
the serial upload in current broken.

```diff
@@ -17,6 +17,9 @@ compiler.cpreprocessor.flags=-D__ets__ -DICACHE_FLASH "-I{compiler.sdk.path}/inc
 compiler.c.cmd=xtensa-lx106-elf-gcc
 compiler.c.flags=-c -Os -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -MMD -std=c99
 
+compiler.S.cmd=xtensa-lx106-elf-gcc
+compiler.S.flags=-c -g -x assembler-with-cpp -MMD
+
 compiler.c.elf.ldscript=eagle.app.v6.ld
 compiler.c.elf.flags=-nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static "-L{compiler.sdk.path}/lib" "-L{compiler.sdk.path}/ld" "-T{compiler.c.elf.ldscript}"
 compiler.c.elf.cmd=xtensa-lx106-elf-gcc
@@ -57,7 +60,7 @@ recipe.c.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.cpreprocessor.fla
 recipe.cpp.o.pattern="{compiler.path}{compiler.cpp.cmd}" {compiler.cpreprocessor.flags} {compiler.cpp.flags} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.cpp.extra_flags} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"
 
 ## Compile S files
-recipe.S.o.pattern="{compiler.path}{compiler.as.cmd}"  -o "{object_file}" "{source_file}"
+recipe.S.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.cpreprocessor.flags} {compiler.S.flags} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.c.extra_flags} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"
 
 ## Create archives
 recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}"
```

hope this help.

closed 2015-03-29 by igrr

czopas, 2015-06-29:
Can anyone help with tutorial how to build sketch using EclipseArduino ? I am trying and still have error like: 

"make all 
Cannot run program "make": Launching failed

Error: Program "make" not found in PATH"

Links2004, 2015-06-29:
may this helps:
https://github.com/esp8266/Arduino/blob/esp8266/docs/eclipse/eclipse.md

if make still missing try to install the make from mingw

czopas, 2015-06-29:
Ok, so I installed Eclipse with baeyens plugin and Arduino 1.6.1.
Next I set Arduino IDE Path to folder with Arduino. I also set Private Hardware Path to C:\Users\[username]\AppData\Roaming\Arduino15\packages\esp8266\hardware.
If I tried to compile I received error:
"make all 
Building file: ../.ino.cpp
Starting C++ compile
"/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:/Users/Marcin/AppData/Roaming/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-804-g2d340c7/tools/sdk//include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10601 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266   -I"C:\Users\Marcin\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-804-g2d340c7\cores\esp8266" -I"C:\Users\Marcin\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-804-g2d340c7\variants\nodemcu" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"   -Wall
/usr/bin/sh: /bin/xtensa-lx106-elf-g++: No such file or directory
make: *** [.ino.cpp.o] Error 127"

Next fix was to change path to:
"runtime.tools.xtensa-lx106-elf-gcc.path={runtime.platform.path}/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9
runtime.tools.esptool.path={runtime.platform.path}/tools/esptool/0.4.5"

The error was chnged to:
"make all 
Building file: ../.ino.cpp
Starting C++ compile
"C:/Users/Marcin/AppData/Roaming/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-804-g2d340c7/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:/Users/Marcin/AppData/Roaming/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-804-g2d340c7/tools/sdk//include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10601 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266   -I"C:\Users\Marcin\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-804-g2d340c7\cores\esp8266" -I"C:\Users\Marcin\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-804-g2d340c7\variants\nodemcu" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"   -Wall
      0 [main] sh 5640 handle_exceptions: Exception: STATUS_ACCESS_VIOLATION
    147 [main] sh 5640 open_stackdumpfile: Dumping stack trace to sh.exe.stackdump
make: *** [.ino.cpp.o] Error 101120"

Any suggestion where is a problem ?

commit cb799b67 (2015-03-29): fix platform file to support Arduino Eclipse Plugin

fixes #2
commit 52be27df (2021-06-16): Release process fix #2 (#8127)

* release script: fix variable name error
* Release process in README is decribed twice. Syncing the two walk-throughs.
* adding ARDUINO_ESP8266_VERSION reflecting exactly what's passed to `makecorever.py -v`
* remove "unix-" in version name (which was present even for windows but never used)