esp8266/Arduino #3: Got an error when flashing

ly0, 2015-03-28:
Sketch uses 196,284 bytes (37%) of program storage space. Maximum is 524,288 bytes.
java.lang.NullPointerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingProgrammer(SerialUploader.java:246)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:75)
	at processing.app.debug.Compiler.upload(Compiler.java:157)
	at processing.app.Sketch.upload(Sketch.java:1259)
	at processing.app.Sketch.exportApplet(Sketch.java:1201)
	at processing.app.Sketch.exportApplet(Sketch.java:1173)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2432)
	at java.lang.Thread.run(Thread.java:745)

igrr, 2015-03-29:
@ly0 did you select "esptool" in Tools > Programmer menu?
What OS are you running?

ly0, 2015-03-29:
sorry, I forgot that, but after I chose esptool, another error occured.
```bash
Sketch uses 196,284 bytes (37%) of program storage space. Maximum is 524,288 bytes.
Uploading 35952 bytes from /var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build6484424438996743417.tmp/WiFiClient.cpp_00000.bin to flash at 0x00000000
warning: espcomm cmd: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
```
OS: mac os x 10.9.5

4refr0nt, 2015-03-29:
GPIO0, GPIO15 pulldown?

ly0, 2015-03-29:
yeah, only failed on Mac, flash in Ubuntu works fine.

svdgraaf, 2015-03-29:
I have the same issue on osx (10.10.3 (14D113c)), works fine in ubuntu on virtualbox. If you need debug info for the osx issue, I'd be more than happy to help out (but don't know where to look) :)

igrr, 2015-03-29:
@svdgraaf could you please enable verbose output in Arduino > Preferences and send me the log?

ly0, 2015-03-29:
Here is my log
```
/Users/latyas/Desktop/Arduino.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -I/Users/latyas/Desktop/Arduino.app/Contents/Java/hardware/tools/esp8266/sdk//include -c -Os -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10601 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -I/Users/latyas/Desktop/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266 -I/Users/latyas/Desktop/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/variants/esp01 -I/Users/latyas/Desktop/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src /var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/WiFiClient.cpp -o /var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/WiFiClient.cpp.o 
使用已经编译的文件：/var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/ESP8266WiFi/ESP8266WiFi.cpp.o
使用已经编译的文件：/var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/ESP8266WiFi/WiFiClient.cpp.o
使用已经编译的文件：/var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/ESP8266WiFi/WiFiServer.cpp.o
使用已经编译的文件：/var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/ESP8266WiFi/WiFiUdp.cpp.o
/Users/latyas/Desktop/Arduino.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-as -o /var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/cont.S.o /Users/latyas/Desktop/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/cont.S 
使用已经编译的文件：/var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/cont_util.c.o
使用已经编译的文件：/var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/core_esp8266_noniso.c.o
使用已经编译的文件：/var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/core_esp8266_wiring.c.o
使用已经编译的文件：/var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/core_esp8266_wiring_analog.c.o
使用已经编译的文件：/var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/core_esp8266_wiring_digital.c.o
使用已经编译的文件：/var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/core_esp8266_wiring_pulse.c.o
使用已经编译的文件：/var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/core_esp8266_wiring_shift.c.o
使用已经编译的文件：/var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/abi.cpp.o
使用已经编译的文件：/var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/core_esp8266_main.cpp.o
使用已经编译的文件：/var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/HardwareSerial.cpp.o
使用已经编译的文件：/var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/i2c.cpp.o
使用已经编译的文件：/var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/IPAddress.cpp.o
使用已经编译的文件：/var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/Print.cpp.o
使用已经编译的文件：/var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/Stream.cpp.o
使用已经编译的文件：/var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/Tone.cpp.o
使用已经编译的文件：/var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/WMath.cpp.o
使用已经编译的文件：/var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/WString.cpp.o
/Users/latyas/Desktop/Arduino.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar cru /var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/core.a /var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/cont.S.o 
/Users/latyas/Desktop/Arduino.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar cru /var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/core.a /var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/cont_util.c.o 
/Users/latyas/Desktop/Arduino.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar cru /var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/core.a /var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/core_esp8266_noniso.c.o 
/Users/latyas/Desktop/Arduino.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar cru /var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/core.a /var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/core_esp8266_wiring.c.o 
/Users/latyas/Desktop/Arduino.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar cru /var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/core.a /var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/core_esp8266_wiring_analog.c.o 
/Users/latyas/Desktop/Arduino.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar cru /var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/core.a /var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/core_esp8266_wiring_digital.c.o 
/Users/latyas/Desktop/Arduino.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar cru /var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/core.a /var/ [... cut ...]
	espcomm_open
Uploading 35952 bytes from /var/folders/rs/c45vj5c5433gw3cmpt641yhr0000gn/T/build8916872152345362035.tmp/WiFiClient.cpp_00000.bin to flash at 0x00000000
	erasing flash
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	setting timeout 5000
	setting character timeout 50
	done
	setting timeout 1
	setting character timeout 1
	done
warning: espcomm cmd: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
closing bootloader
```

svdgraaf, 2015-03-29:
```
Sketch uses 197,436 bytes (37%) of program storage space. Maximum is 524,288 bytes.
/Users/svdgraaf/Desktop/esp2866.app/Contents/Java/hardware/tools/esp8266/esptool -vv -cd none -cb 115200 -cp /dev/tty.usbserial-A9YLP7NJ -ca 0x00000 -cf /var/folders/5q/yzssc7fs2999ym3g3b5v_w480000gn/T/build7603229010055182128.tmp/mDNS_Web_Server.cpp_00000.bin -ca 0x40000 -cf /var/folders/5q/yzssc7fs2999ym3g3b5v_w480000gn/T/build7603229010055182128.tmp/mDNS_Web_Server.cpp_40000.bin 
esptool v0.4.2 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to none
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/tty.usbserial-A9YLP7NJ
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat /var/folders/5q/yzssc7fs2999ym3g3b5v_w480000gn/T/build7603229010055182128.tmp/mDNS_Web_Server.cpp_00000.bin success
opening port /dev/tty.usbserial-A9YLP7NJ at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	espcomm cmd: receiving 2 bytes of data
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_open
Uploading 36112 bytes from /var/folders/5q/yzssc7fs2999ym3g3b5v_w480000gn/T/build7603229010055182128.tmp/mDNS_Web_Server.cpp_00000.bin to flash at 0x00000000
	erasing flash
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	setting timeout 5000
	setting character timeout 50
	done
	serialport_receive_C0: 08 instead of C0
	setting timeout 1
	setting character timeout 1
	done
warning: espcomm cmd: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
closing bootloader
```

sticilface, 2015-03-29:
I've been having exactly the same problem.  

If I copy the paths for the binaries I can flash using the esptool.py... but using esptool does not work, and I can't get it to work with usb-serial devices.  I can via a bluetooth HC-06 though.

I have set up another programmer pyesptool, but for what ever reason the arduino IDE is refusing to run / find a python script...  I've tried lots of variations...  

Here are the contents of the files

1) - platform.txt

tools.pyesptool.cmd=esptool.py
tools.pyesptool.cmd.windows=esptool.exe
tools.pyesptool.path={runtime.ide.path}/hardware/tools/esp8266/pyesptool
tools.pyesptool.program.params.verbose=""
tools.pyesptool.program.params.quiet="" 
tools.pyesptool.program.pattern="python {path}/{cmd}" {program.verbose} --port {serial.port} write_flash 0x00000 "{build.path}/{build.project_name}_00000.bin" 0x40000 "{build.path}/{build.project_name}_40000.bin"

2) - programmers.txt 

pyesptool.name=pyesptool
pyesptool.communication=serial
pyesptool.protocol=esp
pyesptool.program.protocol=esp
pyesptool.program.tool=pyesptool
pyesptool.program.extra_params=

This give me the extra programmer pyesptool... and it generates this, which works if i cut and paste it into the command line...

Sketch uses 185,856 bytes (35%) of program storage space. Maximum is 524,288 bytes.
python /Users/amelvin/Downloads/Arduino-espfinal.app/Contents/Java/hardware/tools/esp8266/pyesptool/esptool.py --port /dev/tty.wchusbserialfa130 write_flash 0x00000 /var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gn/T/build6204336032621940391.tmp/WiFiScan.cpp_00000.bin 0x40000 /var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gn/T/build6204336032621940391.tmp/WiFiScan.cpp_40000.bin

igrr, 2015-03-29:
Okay, thanks all. Looks like esptool needs a bit of tuning.

svdgraaf, 2015-03-29:
Hmmm, interestingly, it does work when I use a arduino as an intermediate instead of my regular usbtoserial ftdi thingy (which did work under ubuntu).

I connected up my arduino (uno), and loaded the 'bare minimum' example, hooked up my esp8266 (yes, using the regular serialports), and boom, it worked... Might be a coincidence, but here's the log:

```
Sketch uses 197,436 bytes (37%) of program storage space. Maximum is 524,288 bytes.
/Users/svdgraaf/Desktop/esp2866.app/Contents/Java/hardware/tools/esp8266/esptool -vv -cd none -cb 115200 -cp /dev/tty.usbmodem1451 -ca 0x00000 -cf /var/folders/5q/yzssc7fs2999ym3g3b5v_w480000gn/T/build7603229010055182128.tmp/mDNS_Web_Server.cpp_00000.bin -ca 0x40000 -cf /var/folders/5q/yzssc7fs2999ym3g3b5v_w480000gn/T/build7603229010055182128.tmp/mDNS_Web_Server.cpp_40000.bin 
esptool v0.4.2 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to none
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/tty.usbmodem1451
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat /var/folders/5q/yzssc7fs2999ym3g3b5v_w480000gn/T/build7603229010055182128.tmp/mDNS_Web_Server.cpp_00000.bin success
opening port /dev/tty.usbmodem1451 at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	serialport_receive_C0: 1C instead of C0
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	espcomm cmd: receiving 2 bytes of data
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_open
Uploading 36112 bytes from /var/folders/5q/yzssc7fs2999ym3g3b5v_w480000gn/T/build7603229010055182128.tmp/mDNS_Web_Server.cpp_00000.bin to flash at 0x00000000
	erasing flash
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	setting timeout 5000
	setting character timeout 50
	done
	setting timeout 1
	setting character timeout 1
	done
	espcomm cmd: receiving 2 bytes of data
	writing flash
....................................
	setting address from 0x00000000 to 0x00040000
	espcomm_upload_file
	stat /var/folders/5q/yzssc7fs2999ym3g3b5v_w480000gn/T/build7603229010055182128.tmp/mDNS_Web_Server.cpp_40000.bin success
	espcomm_open
Uploading 161368 bytes from /var/folders/5q/yzssc7fs2999ym3g3b5v_w480000gn/T/build7603229010055182128.tmp/mDNS_Web_Server.cpp_40000.bin to flash at 0x00040000
	erasing flash
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	setting timeout 5000
	setting character timeout 50
	done
	setting timeout 1
	setting character timeout 1
	done
	espcomm cmd: receiving 2 bytes of data
	writing flash
..............................................................................................................................................................
starting app without reboot
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	espcomm cmd: receiving 2 bytes of data
closing bootloader
```

sticilface, 2015-03-30:
OK, got an intermediate fix.. that uses https://github.com/themadinventor/esptool.git

put this in your IDE packages directory ./Contents/Java/hardware/tools/esp8266/pyesptool 
make sure its working by flashing some binaries using it...

add this to files located here
./Contents/Java/hardware/esp8266com/esp8266/

platform.txt
`````
tools.pyesptool.cmd=esptool.py
tools.pyesptool.cmd.windows=esptool.exe
tools.pyesptool.path={runtime.ide.path}/hardware/tools/esp8266/pyesptool
tools.pyesptool.program.params.verbose=""
tools.pyesptool.program.params.quiet="" 
tools.pyesptool.program.pattern=python "{path}/{cmd}" {program.verbose} --port {serial.port} write_flash 0x00000 "{build.path}/{build.project_name}_00000.bin" 0x40000 "{build.path}/{build.project_name}_40000.bin"
```````

and this to programmers.txt
``````
pyesptool.name=pyesptool
pyesptool.communication=serial
pyesptool.protocol=esp
pyesptool.program.protocol=esp
pyesptool.program.tool=pyesptool
pyesptool.program.extra_params=
``````

Then reboot the IDE, and you should have an extra programmer pyesptool that works... (at least I do)

GiorgioAresu, 2015-04-01:
I can confirm what said by @svdgraaf , in particular the Arduino needs to have the BareMinimum sketch flashed (I had to reflash to get rid of the ArduinoISP it had)

jarkman, 2015-04-14:
I've got what looks like the same issue on OSX with a FTDI USB dongle - I see 
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
whatever I do. The log is below. 

I'm happy to dig into this some more, but I'm not really sure where to start. Any suggestions would be very welcome.

/Applications/"Arduino 161 esp.app"/Contents/Java/hardware/tools/esp8266/esptool -vvv -cd none -cb 115200 -cp /dev/tty.wchusbserial1410 -ca 0x00000 -cf /var/folders/71/9gbg1bj16jjg0h_7r49pbfv00000gn/T/build5091595574695456475.tmp/HelloServer.cpp_00000.bin -ca 0x40000 -cf /var/folders/71/9gbg1bj16jjg0h_7r49pbfv00000gn/T/build5091595574695456475.tmp/HelloServer.cpp_40000.bin
esptool v0.4.2 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to none
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/tty.wchusbserial1410
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat /var/folders/71/9gbg1bj16jjg0h_7r49pbfv00000gn/T/build5091595574695456475.tmp/HelloServer.cpp_00000.bin success
opening port /dev/tty.wchusbserial1410 at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
		espcomm cmd: didn't receive command response
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	espcomm cmd: receiving 2 bytes of data
		espcomm_cmd: received 2 bytes: 
		0x00 
		0x00 
		espcomm cmd: response 0x20120707
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_open
Uploading 35904 bytes from /var/folders/71/9gbg1bj16jjg0h_7r49pbfv00000gn/T/build5091595574695456475.tmp/HelloServer.cpp_00000.bin to flash at 0x00000000
	erasing flash
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	setting timeout 5000
	setting character timeout 50
	done
	setting timeout 1
	setting character timeout 1
	done
warning: espcomm cmd: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
closing bootloader

jarkman, 2015-04-15:
And, I've tried sticilface's recipe from above, and that worked for me immediately. Maybe esptool hates FTDI adapters ?

rei-vilo, 2015-04-19:
@sticilface  

Thank you very much for the solution with `esptool.py`. It works fine on Mac OS X where `esptools` doesn't work.

>
$ python /Users/ReiVilo/Downloads/esptool-master/esptool.py --port /dev/tty.wchusbserial1410 write_flash 0x00000 /var/folders/5d/dky9xc691m32kztqnhkn8bhw0000gn/T/build719426854255008627.tmp/Blink.cpp_00000.bin 0x40000 /var/folders/5d/dky9xc691m32kztqnhkn8bhw0000gn/T/build719426854255008627.tmp/Blink.cpp_40000.bin
Connecting...
Erasing flash...
Writing at 0x00008800... (100 %)
Erasing flash...
Writing at 0x00066000... (100 %)
>
Leaving...
>

The only thing I did was to replace in `platform.txt`

> tools.pyesptool.program.pattern=python "{path}/{cmd}" {program.verbose} --port {serial.port} write_flash 0x00000 "{build.path}/{build.project_name}_00000.bin" 0x40000 "{build.path}/{build.project_name}_40000.bin"

by

> tools.pyesptool.program.pattern="{path}/{cmd}" {program.verbose} --port {serial.port} write_flash 0x00000 "{build.path}/{build.project_name}_00000.bin" 0x40000 "{build.path}/{build.project_name}_40000.bin"

achingbrain, 2015-05-23:
The Python flasher @sticilface mentions above seems to work much more reliably for OS X - it'd be great if it could be added to the distribution.

achingbrain, 2015-05-23:
Actually looks like there's already a PR for this #100 but it's a little out of date.

talhakamran, 2015-09-27:
Same issue :(

Build options changed, rebuilding all

Sketch uses 199,406 bytes (19%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 32,708 bytes (39%) of dynamic memory, leaving 49,212 bytes for local variables. Maximum is 81,920 bytes.
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_sync failed
error: espcomm_open failed

dogmatic69, 2015-09-30:
I have just installed the package to Arduino IDE and have the same issue.

Trying to burn:
```
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

output

```
Sketch uses 201,034 bytes (19%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 44,592 bytes (54%) of dynamic memory, leaving 37,328 bytes for local variables. Maximum is 81,920 bytes.
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_sync failed
error: espcomm_open failed
```

talhakamran, 2015-10-06:
Ah! its working now for me after I plugged in the GPIO0 to GND.
So if I understand it correctly NodeMCU framework write to a file in ROM. But Arduino IDE flashes the entire ESP8266 and then pushes in a new firmware with your code. With that said.
If any issues with your code you may need to reflash the ESP8266 to original firmware :)

pai168, 2015-10-07:
I've got below error messages always, but it works fine after restarted Arduino IDE 1.6.5 when esp8266 was still connected.
.......
warning: espcomm_sync failed
error: espcomm_open failed

drmpf, 2015-10-07:
I have the same experience.  When I get those errors I restart the IDE and that seems to solve it.

tablatronix, 2015-12-17:
Same problem here.

I have to hold `GPIO 0`  to gnd during uploads to avoid this error when using `GPIO 0` as output
(adafruit huzzah, ftdi)

Which makes no sense since nothing should be executing during bootloading, so not sure how that matters, but I barely have an issue when not using GPIO0... take from that what you will.

Links2004, 2015-12-17:
@tablatronix this problem shut be fixed in git
see: #1212

tablatronix, 2015-12-17:
No change, I also already tried generic and changed this.

I am having a hard time finding the details on resetmode but i am assuming this is auto reset bootload related.

I am using manual reset.

tablatronix, 2015-12-18:
There are 3 different main issues here, mine and a few others probably should be broken out.

There is the issue with ecomm sync failed, usually occurs after an erased firmware and upload failure, it looks like bootload but it will never work until reset, sometimes an actual power cycle is needed when in this state.

Another is wrong direction, i never got this personnally but i have had the chip get stuck in rom dump mode where it just streams out data and is hard to reset or get out of it.

And there is the failed to receive error, which is what I get. Upload success seems to be random.

dave-prosee, 2015-12-19:
At first I thought I had the same problem  but I support Tablaronix in that there are different issues.
However by trying the esptool.py I came across the protocol and looked into it by capturing the debug output.
I had "wrong direction/command" and times where communication just didn't start. Then it took a little time to found out that my adapter was not giving proper signals (low was just 1Volt below high). 
So "failed to receive" and "wrong direction/command" could be related to weak signals from adapters.
I posted some more info on #1145. Maybe this helps someone.

tablatronix, 2015-12-19:
Could this be an internal pullup state problem?

dave-prosee, 2015-12-19:
I don't know how to tell. All my ESP12E's were fresh and I suppose the bootloader in the chip is the same for all. Maybe some other load on the ESP board? I wanted to know too, but couldn't find a clue.

tablatronix, 2015-12-20:
I tried switching to epstool.py and changed programmer to it, but it seems to be ignored when I compile, could the board.txt entry be overriding it as it always still calls esptool.exe, does something else have to change.

tablatronix, 2015-12-21:
I think I was having the same issue, got everything working fine with a new adapter.

Gonna choke this up to a voltage levels issue, the adapter i was using is made for industrial control and  5v level, so it has massive filtering on it (ferrite beads and a lot of caps), I was seeing voltage across the FTDI 232 of 4.95v from a 5.10 supply, coupled with the adafruit huzzah which has a dropper on the RX line. It worked perfect for everything , but it would not get reliable responses from flash erase. ( perhaps the baud rate used at that point, or the baud negotiate, is tripping the curves up.

Ill look into this later when I have a scope or serial inspector. meh.

dave-prosee, 2015-12-21:
Was away for a day and could not react on your question yesterday. I think you got it already. The instructions above were missing boards.txt change. There is <yourboardname>.upload.tool=<your programmer> to be changed for the board in question.
Glad yo see you have it solved. I am reading 5V in your story. Be aware that this could harm the ESP.

tablatronix, 2015-12-21:
The adafruit huzzah dev board has regulator and level shifters, works with 5v serial.

dave-prosee, 2015-12-21:
OK.

tablatronix, 2016-01-03:
Quick question about using pyesptool as custom programmer on mac.

My IDE does not create  ino.00000.bin or 40000, it only compiles a .ino.bin
So the platform cmd posted here does not work.

Has this changed at some point since ?

tablatronix, 2016-01-03:
looks like you can write the entire bin , you dont need 2 blobs. not sure what the addresses specify.

Since i am using git this is my platform entry that works.

```
tools.pyesptool.cmd=esptool.py
tools.pyesptool.cmd.windows=esptool.exe
tools.pyesptool.path={runtime.ide.path}/hardware/esp8266com/esp8266/tools/pyesptool/
tools.pyesptool.program.params.verbose=""
tools.pyesptool.program.params.quiet="" 
tools.pyesptool.program.pattern=python "{path}{cmd}" {program.verbose} --port {serial.port} write_flash 0x00000 "{build.path}/{build.project_name}.bin" 
```

benoit332, 2016-05-19:
Hi all, 

The command line :
`python "{path}{cmd}" {program.verbose} --port {serial.port} -b {upload.speed} write_flash 0x00000 "{build.path}/{build.project_name}.bin"`
...will assume default flash size by esptool.py (and it is 512K).

I've uncountered issues with OTA in this case: flash write error above the 512K limit on a 4M board.

So I recommand to add the following esptool.py options to the command line:
`-fs (flash size), -ff (flash frequency), -fm (flash mode)`

I successfully tested it on my Huzzah board (4MB = 32Mb) and it fixed my OTA issues with this: 
`{runtime.platform.path}/tools/esptool.py" --port "{serial.port}" write_flash -fs 32m -ff 80m -fm dio 0x00000 "{build.path}/{build.project_name}.bin"`

I don't know if there is a way to copy the parameters from the Arduino user interface to work with all boards?

Benoît.

devyte, 2017-09-04:
@ly0 @tablatronix Is this issue still valid? The last comment was over a year ago.
Also, I understand that all options need to be specified to esptool, along the lines of what @benoit332 said.

tablatronix, 2017-09-04:
I personally have not encountered it in some time. There are alot of commit refs to this issue though, might want to check through it a few times, there seems to be alot of refs here.

devyte, 2017-10-20:
Closing due to age and can't reproduce. If anyone thinks there is still a problem, please open a new issue and include a reference to this one.

closed 2017-10-20 by devyte

Trospector, 2017-12-17:
Shorting "RESET" to "GND" on Arduino Uno solves the following problem when connecting to ESP8266 via Arduino (even with the Blank Sketch on the Arduino):
I was able to flash to ESP01 from Arduine IDE 1.8.3 running on Lubuntu 16.10, but not from esptool.py running on the same Lubuntu. Turns out shorting RESET to GND on the Arduino with jumper cable solved the issue. Otherwise I was getting the same error message as ly0 and svdgraaf above.

Also thanks to sticilface above for pointing to the esptool.py github location. Installing esptool on Lubuntu 16.10 via
sudo apt-get install esptool
gave me some other esptool other than the esptool.py mentioned above, that hung me up for a while and wasted a good amount of time. Finally I did a manual install of esptool.py from the link provided above by sticilface, and along with the RESET ground on the Arduino, esptool.py has started working.