esp8266/Arduino #4: Latest windows version 1.6.1-p1.zip doesn't seem to compile

rogerclarkmelbourne, 2015-03-28:
There seems to be some sort of compiler build issue in the latest windows zip.

You just get a message saving Error compiling

But if you look at the verbose output and cut and paste it into the Windows command line, and run it, the error is that the xtensia gcc compilor can't locate libiconv-2.dll

4refr0nt, 2015-03-28:
temp solve: copy this file from arduino-1.6.1\hardware\tools\avr\libexec\gcc\avr\4.8.1 to arduino-1.6.1

rogerclarkmelbourne, 2015-03-29:
Sorry. I'm not sure what you mean.

Which files need to be copied libiconv-2.dll ?

PS. I have replaced the tools folder with a working copy from another repo, and it now compiles, but it gets a java runtime null pointer error something to do with esptool, but I'm not sure what the platforms.txt is missing, I suspect its missing a paramater that is required for 1.6.1

4refr0nt, 2015-03-29:
Yes.
Copy file libiconv-2.dll 
From: arduino-1.6.1\hardware\tools\avr\libexec\gcc\avr\4.8.1
To: arduino-1.6.1

rogerclarkmelbourne, 2015-03-29:
I'll try that, but I still think I will get this error when uploading 

java.lang.NullPointerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingProgrammer(SerialUploader.java:246)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:75)
	at processing.app.debug.Compiler.upload(Compiler.java:157)
	at processing.app.Sketch.upload(Sketch.java:1259)
	at processing.app.Sketch.exportApplet(Sketch.java:1201)
	at processing.app.Sketch.exportApplet(Sketch.java:1173)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2432)
	at java.lang.Thread.run(Thread.java:745)

rogerclarkmelbourne, 2015-03-29:
Well, you change did fix the compile bug, but the upload bug still exists

I'd already fixed the compile bug by using another build of the Xtensia compiler

Does upload work for you ??

4refr0nt, 2015-03-29:
yes, I down't have any upload troubles. All work very well.

rogerclarkmelbourne, 2015-03-29:
OK
Worked out what it is

The Programmer menu default selection is empty.

You have to select "esptool" otherwise it goes wrong

4refr0nt, 2015-03-29:
delete Arduino 1.6.1 folder
unpack Arduino 1.6.1 again from github archive to up level folder (on my Win7 x64 - C:\)
copy file libiconv-2.dll from: C:\arduino-1.6.1\hardware\tools\avr\libexec\gcc\avr\4.8.1 to: C:\arduino-1.6.1
thats all.

rogerclarkmelbourne, 2015-03-29:
This whole method of needing to deploy the whole IDE is the old way

You should not be using this method.

The new method is to put files in the user's hardware folder

I have re-configure all the files to do this now on my local installation

4refr0nt, 2015-03-29:
and now all working good?

rogerclarkmelbourne, 2015-03-29:
Yes. It is working. However I will upload a much better way to do this, and I have a better compiler version which doesnt need that hack.

closed 2015-03-29 by rogerclarkmelbourne

rogerclarkmelbourne, 2015-03-29:
OK.
My version using your core code is in github

Instructions

In your MyDocuments/ Arduino  folder

make a hardware folder (if you don't already have one)

Then clone the repo into the hardware folder

Download the official Arduino IDE 1.6.2 and install it

Run the Official IDE, (first run seems to take some time, I suspect its building a file cache)

quit the official IDE as the first run doesn't seem to be able to see comm ports correctly

Restart the official IDE

Looking in the boards menu and you will see the Generic ESP8266  - select this

Under the programmer menu, select esptool (as the default appears to be no selection, which causes issues for upload)

Compile upload and enjoy ;-)

igrr, 2015-03-29:
@rogerclarkmelbourne 
Distributing the "hardware" folder is a good option, in my opinion, and I'll certainly add the packaged .zip to the project's releases page.
The only issue is that once I will get WiFi uploads working, users will need to have modified Java code as well, so distributing the full installer will be the way to go.

igrr, 2015-03-29:
That is, unless there is a way to integrate Java plugins into the IDE.

rogerclarkmelbourne, 2015-03-29:
What do you need to do that requires Java ?

Ah. Uploading via Wifi ??

You can run anything you like as an upload script, so I presume you can just execute java from a batch file from windows

And run shell scripts from OSX and Linux

I manage a repo for the STM32 on Arduino and we have support for Linux and OSX and have upload scripts for both of those OS's 

But on Linux, users have to get and compile their own uploaders, because we can't distribute a binary uploader that works for all Linux distro's 

(Actually its quite complex as on the STM32, there are various different boards, some with bootloaders that need specific uploaders e.f. DFU (like android uses) and also Serial and also anoher protocol called STLink

Anyway. If you let me know what the wifi uploader would need to do, I may be able to assist

igrr, 2015-03-29:
Uploading over Wifi requires adding items to the list of serial ports, one
item for each esp device on the user's wifi network. On Windows, the list
of serial ports is provided by an external tool, while on Linux and Mac
this list is generated in Java.
So if you know how to populate the ports list without modifying Java code,
that would be great indeed.

rogerclarkmelbourne, 2015-03-29:
Ivan

I'm not entirely sure what you mean.

Do you mean you want to update the list of serial ports that the IDE is showing in its menu ?

If so, I don't think there is a way to do that without recompiling the IDE, or at least modifying part of it.

You can add menu's to the IDE, by defining them in boards.txt, but as far as I know they are static, e.g. you can add a menu which selects the baud rate for the upload, and you can use the params from the menu and pass them to the upload script

But I don't think you can dynamically add items to any of these menus 

You can run any external script you like, when the user presses Upload (control U), but I'm not sure whether this would help.

As it stands you already pass the comm port to the esptool, so I guess what you require is far more complicated

Perhaps if you can write a description of how the wifi upload works, I may be able to help

igrr, 2015-03-29:
Roger,

Wifi upload works as follows:
- esp modules send multicast UDP packets over wifi network.
- some tool running on the users PC receives these packets and builds a
list of esp modules that are ready to receive the sketch
- the list of modules is added to the list of ports, allowing the user to
select which esp module he would like to upload to
- when the user presses upload, some tool sends the sketch over TCP to the
selected module

So given that we need to add items to the ports list I don't see other way
than modifying the java code.

igrr, 2015-03-29:
this is, by the way, similar to the wifi uploads that are implemented for
Yun boards, except that we can't use Yun's upload method as it is
hard-coded to use ssh for upload itself.

rogerclarkmelbourne, 2015-03-29:
Ivan,

At the moment, I agree that it sounds like you need to have a modified version of the IDE

However the IDE developers are very active at the moment, and if you can propose a new system e,g which runs as a script when the IDE is started, this may avoid the need for a completely custom version

The problem is that the IDE team seem to release a new version very month, e.g when I started the arduino stm32 project last November, the IDE was 1.5.7 and now its 1.6.2 

So having to rebuild a new release each time the IDE team makes a new version takes a lot of time.

rogerclarkmelbourne, 2015-03-29:
Just a thought...

But you could get the wifi uploader to display a popup, with the list of devices.

The user could then select a device, and the uploader would remember that selection for the current session.

However this is not ideal, as if the user is allowed to change the selection each time, you would need some way to allow this, 

Perhaps a small delay and if no change of selection, the upload would continue

igrr, 2015-03-29:
I agree that it would be better to distribute a plugin, not the IDE.
Gerarding rebuilding, I have CI servers that build distributions on each
commit, so it's not a huge problem right now.