esp8266/Arduino #10: Issue compiling MODBUS library

iMarvinS, 2015-03-29:
Hello!
I just wanted to compile the arduino-modbus-slave library (https://code.google.com/p/arduino-modbus-slave/downloads/detail?name=MODBUS.zip&can=2&q=) which seems to be perfectly compatible with the ESP8266 Arduino SDK because it only uses the Serial library which is fully adopted.
When I try to compile my sketch which depends on the MODBUS library I get the following error: 
[image]

Does anybody know where the issue is?
It seems to be an issue with the xtensa toolchain but I cannot find out what it is.

Kind Regards
Marvin

igrr, 2015-03-29:
The issue is because the library seems to use malloc function, which is not
available. There are three ways to fix that:
- replace malloc and free with new and delete in the library
- replace malloc and free with os_malloc and os_free which are defined in
ESP SDK
- fix ulibc to use os_malloc instead of malloc.

igrr, 2015-03-29:
A lot of people seem to be having issues related to malloc, so I created #11 to track it.

iMarvinS, 2015-03-29:
Thank you! 
I replaced all malloc and free calls with their corresponding C++ equivalents (new and delete).
Now the library works flawless on the ESP as well as the Uno.
This issue is resolved now.

Thanks

closed 2015-03-29 by iMarvinS

gkrananth, 2015-08-02:
Can you please share the library