esp8266/Arduino #11: libc-provided malloc and free need to be replaced by os_malloc and os_free

igrr, 2015-03-29:
References to malloc and other functions that use malloc (like sprintf) pull in _sbrk, which isn't available.
Need to tweak libc build so that it would use os_malloc and os_free instead.

dmcinnes, 2015-04-01:
Running into the same issue here. Error output:

```
Arduino: 1.6.1 (Mac OS X), Board: "Generic ESP8266 board"

/Applications/Arduino-esp8266.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-mallocr.o):(.literal+0x1c): undefined reference to `_sbrk_r'
/Applications/Arduino-esp8266.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-mallocr.o): In function `malloc_extend_top':
mallocr.c:(.text+0x5e): undefined reference to `_sbrk_r'
mallocr.c:(.text+0x127): undefined reference to `_sbrk_r'
/Applications/Arduino-esp8266.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-freer.o): In function `_malloc_trim_r':
mallocr.c:(.text+0x366): undefined reference to `_sbrk_r'
mallocr.c:(.text+0x38f): undefined reference to `_sbrk_r'
/Applications/Arduino-esp8266.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-freer.o):mallocr.c:(.text+0x3a0): more undefined references to `_sbrk_r' follow
collect2: error: ld returned 1 exit status
Error compiling.
```

igrr, 2015-04-01:
I did try disabling built-in malloc and free in ulibc at some point. But I guess I wasn't able to pass correct parameters via crosstool-NG because the resulting library still had malloc and free.
One hack that I just thought of is to use ```ar``` to remove malloc and free from ```libc.a```, and then provide os_malloc and os_free as replacements in the linker script (perhaps they are already there).

closed 2015-04-11 by igrr

oystein-devicedrive, 2016-10-10:
Also make sure you use os_sprintf instead of sprintf.