esp8266/Arduino #12: cbuf.h read() and peek() wrong return type! (not int)

Links2004, 2015-03-29:
read() and peek() in cbuf.h hast return type char this is wrong!

need to be int
 int read()
 int peek()

if not readBytesUntil for example is not working.

igrr, 2015-03-29:
What exactly is not working? Could you please post an example sketch?

I think in WiFiClient.cpp I always call getSize() before calling read and peek, so this shouldn't be a problem.

Will change the return type, just wondering why this would be an issue.

igrr, 2015-03-29:
Oh, right, that's HardwareSerial. Forgot to add a check there.
WiFiClient doesn't use cbuf now.
Sure. Will fix.

Links2004, 2015-03-29:
example:
```cpp
char buf[16];
Serial.begin(9600);
int c = Serial.readBytesUntil((char) 0x03, buf, 16); //<- instant return buffer full of gabage (0xFF)
```
reason:
```cpp
size_t ICACHE_FLASH_ATTR Stream::readBytesUntil(char terminator, char *buffer, size_t length)
{
  if (length < 1) return 0;
  size_t index = 0;
  while (index < length) {
    int c = timedRead(); // <--- int here
    if (c < 0 || c == terminator) break;
    *buffer++ = (char)c;
    index++;
  }
  return index; // return number of characters, not including null terminator
}
```
```cpp
// private method to read stream with timeout
int ICACHE_FLASH_ATTR Stream::timedRead()
{
  int c;
  _startMillis = millis();
  do {
    c = read();  // <--- int here
    if (c >= 0) return c;
    yield();
  } while(millis() - _startMillis < _timeout);
  return -1;     // -1 indicates timeout
}
```

```cpp
int ICACHE_FLASH_ATTR HardwareSerial::read(void)
{
    return _rx_buffer->read();  // <--- no int here
}
```

```cpp
char read()  // <--- no int here
    {
        if (getSize() == 0)
            return -1;  // <--- no int type chat to char --> result 0xFF
        
        char result = *_begin;
        if (++_begin == _bufend)
            _begin = _buf;
        return result;
    }
```

commit 260aee75 (2015-03-29): Fix cbuf::read and cbuf::peak return types

Fixes #12
closed 2015-03-29 by igrr