esp8266/Arduino #6: cyborg5/IRLib gives an error when compiling

probonopd, 2015-03-29:
I am trying to use https://github.com/cyborg5/IRLib/

```
IRLib/IRLib.h:74:7: error: '__FlashStringHelper' does not name a type
 const __FlashStringHelper *Pnames(IRTYPES Type); //Returns a character string that is name of protocol.
```

How would I have to change https://github.com/cyborg5/IRLib/blob/master/IRLib.h to make it work on the esp8266?

igrr, 2015-03-29:
F() macro definition lost during merge, need to add it back again.

igrr, 2015-03-29:
Looking at the code again, there's also some AVR specific stuff here that's needs a rewrite: https://github.com/cyborg5/IRLib/blob/master/IRLib.cpp#L926

juanpintom, 2015-06-24:
There are news about this? I found some post on forum talking about, but not success for now :(

http://www.esp8266.com/viewtopic.php?f=11&t=3035&p=21524#p21524
http://www.esp8266.com/viewtopic.php?f=33&t=3357&p=19670

This one talks about one RTOS example, but I haven't knowledge to get it working on Arduino IDE ESP:
http://www.esp8266.com/viewtopic.php?f=32&t=2239&p=20127&hilit=IR#p20127

Regards

probonopd, 2015-06-24:
See https://github.com/markszabo/IRremoteESP8266/
Infrared remote library for ESP8266: send and receive infrared signals with multiple protocols. Based on: https://github.com/shirriff/Arduino-IRremote/

Unfortunately it only supports sending so far and not yet receiving but it is a start.
Still, I would prefer to see IRLib which has more functionality. Maybe someone who has the skills can port the changes that have been made to IRremote to run on the ESP8266 to the IRLib library, and implement receiving?

ildr3him, 2015-09-16:
Any progress on this end? Is it impossible to receive IR signals on the ESP8266?

devyte, 2017-10-20:
pgmspace.h has been reworked several times since this was opened.
Closing due to age and 3rd party lib.

closed 2017-10-20 by devyte

C0rn3j, 2017-12-25:
I was wondering about the current status of IR on the ESP8266 - and it seems to work with the project linked before, including sending.

https://github.com/markszabo/IRremoteESP8266/

Here is also some very crude receiver that will just output to serial(since the examples I found out needed at least some modification to run on current library/arduino IDE versions):
```
int RECV_PIN = 4; // D2, IR OUT
unsigned int timeOfIncoming;
IRrecv irrecv(RECV_PIN);
decode_results results;

void setup() {
  Serial.begin(115200);
  Serial.println("Starting up");
  irrecv.enableIRIn(); // Start the receiver
}
void loop() {
   if (irrecv.decode(&results)) {
    Serial.println((long int)results.value, HEX);
    irrecv.resume(); // Receive the next value
  }
  delay(100);
}
```

Also https://github.com/esp8266/Arduino/issues/3371 seems to be related(came up when googling the include error), so I'm linking it through for posterity.