esp8266/Arduino #7: ESP8266 mDNS: How to advertise on the network?

probonopd, 2015-03-29:
I would like to use mDNS in order to advertise the web server running on the ESP8266 to the network, so that it is easily found by guests and other devices on the network. How can this be done?

igrr, 2015-03-29:
I would suggest you to check the [example](https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/ESP8266mDNS/examples/mDNS_Web_Server/mDNS_Web_Server.ino) provided with ESP8266mDNS library.

probonopd, 2015-04-04:
Others seem to have the same issue:

"Also used the App Bonjour Browser, but the ESP is not seen as Bonjour device.
Anyone having success with this example on a Mac?"

http://www.esp8266.com/viewtopic.php?f=29&t=2270

igrr, 2015-04-04:
To be seen as a bonjour service we need to support DNS-SD, not just bare
mDNS. DNS-SD may be implemented on top of mDNS, but I haven't yet done this.

gregslomin, 2015-04-18:
I can also confirm that the mDNS example is not working at all for me.

closed 2015-04-27 by igrr

igrr, 2015-04-27:
I've added missing interface address setup for multicast PCBs, so this example works properly now.

Note that it only works in station mode, see my comment [here](https://github.com/esp8266/Arduino/issues/74#issuecomment-96730305) for details.

commit 99510e31 (2015-04-27): UDP multicast APIs fix

fix #74, fix #7
commit 3665c874 (2015-04-28): UDP multicast APIs fix

fix #74, fix #7