esp8266/Arduino #8: Support Servo library

probonopd, 2015-03-29:
So far, the Servo library is not supported. https://github.com/esp8266/Arduino/commit/4b7302692c1c0f5491c05f31e6b84b4087d619e2#commitcomment-10456654
It would be nice if the library could be expanded to support the esp8266 as well; this is probably dependent on the PWM code being implemented.

igrr, 2015-03-29:
Looking at the code, this library heavily uses AVR timer registers. It doesn't even support Arduino Due (ARM) boards. Supporting esp8266 would require some heavy rework of this library.

Makuna, 2015-05-23:
I was looking at this also.  The rework is basically abstracting the timers.  The SAM version has better timer abstraction due to the existing timer APIs.

Makuna, 2015-05-29:
https://github.com/esp8266/Arduino/pull/358

igrr, 2015-06-02:
Merged

closed 2015-06-02 by igrr

vpr1995, 2015-06-18:
[image]
Please help me out,I have installed latest version of Arduino IDE(1.6.5) and also from Board manager(..3697) ,i had the same problem for Arduino IDE(1.6.4).

Makuna, 2015-06-18:
From the preferences menu turn on verbose compiler output, then build and copy the complete output.  From what I see above, it is including the wrong header file as the path is wrong.

vpr1995, 2015-06-19:
How to change the path?

Makuna, 2015-06-19:
What version are you using of the Esp8266?  If you include the path that is similar to what I had shown above, you can see the version I am running is 1.6.4-816-<blah>.  The 816 is the important part here.  
How did you install the esp8266 Board support?  The servo library is in the staging install at the moment waiting for other stuff to stabilize before being put into the normal install.
To use the staging install, you need to change the json path to the following.
``` http://arduino.esp8266.com/staging/package_esp8266com_index.json  ```

vpr1995, 2015-06-19:
I just followed the instructions given in readme.Now its working thank you for sorting me out.

Lizatre, 2015-06-29:
I am still having this problem even though I followed all README instructions. I also tried adding the staging install but I still get the [avr, sam] architecture error message. Any advice?

igrr, 2015-06-29:
Please see #463 and comment from @Makuna above — you might not be using the version which contains the changes for the Servo library.

Lizatre, 2015-06-29:
Yes that link was very helpful. Now I'm just having a problem with the timecallback functions for some reason...

Arduino: 1.6.4 (Windows 7), Board: "NodeMCU (ESP8266 ESP-12 Module), 80 MHz, 115200"

In file included from C:\Users\Valued Customer\Documents\Arduino\libraries\Servo\src/Servo.h:67:0,
                 from sketch_jun27c.ino:1:
C:\Users\Valued Customer\Documents\Arduino\libraries\Servo\src/esp8266/ServoTimers.h:59:24: error: 'timercallback' has not been declared
     void InitInterrupt(timercallback handler) 
                        ^
C:\Users\Valued Customer\Documents\Arduino\libraries\Servo\src/esp8266/ServoTimers.h: In member function 'void ServoTimer0::InitInterrupt(int)':
C:\Users\Valued Customer\Documents\Arduino\libraries\Servo\src/esp8266/ServoTimers.h:61:25: error: 'timer0_isr_init' was not declared in this scope
         timer0_isr_init();
                         ^
C:\Users\Valued Customer\Documents\Arduino\libraries\Servo\src/esp8266/ServoTimers.h:62:39: error: 'timer0_attachInterrupt' was not declared in this scope
         timer0_attachInterrupt(handler);
                                       ^
C:\Users\Valued Customer\Documents\Arduino\libraries\Servo\src/esp8266/ServoTimers.h: In member function 'void ServoTimer0::StopInterrupt()':
C:\Users\Valued Customer\Documents\Arduino\libraries\Servo\src/esp8266/ServoTimers.h:69:32: error: 'timer0_detachInterrupt' was not declared in this scope
         timer0_detachInterrupt();
                                ^
C:\Users\Valued Customer\Documents\Arduino\libraries\Servo\src/esp8266/ServoTimers.h: In member function 'void ServoTimer0::SetPulseCompare(uint32_t)':
C:\Users\Valued Customer\Documents\Arduino\libraries\Servo\src/esp8266/ServoTimers.h:74:49: error: 'timer0_write' was not declared in this scope
         timer0_write(ESP.getCycleCount() + value);
                                                 ^
C:\Users\Valued Customer\Documents\Arduino\libraries\Servo\src/esp8266/ServoTimers.h: In member function 'void ServoTimer0::SetCycleCompare(uint32_t)':
C:\Users\Valued Customer\Documents\Arduino\libraries\Servo\src/esp8266/ServoTimers.h:79:41: error: 'timer0_write' was not declared in this scope
         timer0_write(_cycleStart + value);
                                         ^
C:\Users\Valued Customer\Documents\Arduino\libraries\Servo\src/esp8266/ServoTimers.h: At global scope:
C:\Users\Valued Customer\Documents\Arduino\libraries\Servo\src/esp8266/ServoTimers.h:147:24: error: 'timercallback' has not been declared
     void InitInterrupt(timercallback handler)
                        ^
C:\Users\Valued Customer\Documents\Arduino\libraries\Servo\src/esp8266/ServoTimers.h: In member function 'void ServoTimer1::InitInterrupt(int)':
C:\Users\Valued Customer\Documents\Arduino\libraries\Servo\src/esp8266/ServoTimers.h:150:39: error: invalid conversion from 'int' to 'void (*)()' [-fpermissive]
         timer1_attachInterrupt(handler);
                                       ^
In file included from C:\Users\Valued Customer\Documents\Arduino\libraries\Servo\src/Servo.h:52:0,
                 from sketch_jun27c.ino:1:
C:\Users\Valued Customer\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.4-673-g8cd3697\cores\esp8266/Arduino.h:106:6: error:   initializing argument 1 of 'void timer1_attachInterrupt(void (*)())' [-fpermissive]
 void timer1_attachInterrupt(void (*userFunc)(void));
      ^

Makuna, 2015-06-30:
If you examine your output, you will notice that you are running 673 version (using my instructions above to help identify the build).

build 673 doesn't support Servo as it was included after that.  So nothing will work until you get a newer build.

It looks like build 673 was the last "public" build.  If you want to use the staging build, it will include the library.  You can get the staging build (albeit you will be using a possible unstable build) you follow the normal instructions, but use the following for the path ```http://arduino.esp8266.com/staging/package_esp8266com_index.json```

chrisallick, 2015-10-18:
Hi,
I know this is closed, but I installed the staging package, and I got the servo working fine on by attaching to "5". However, it will not connect to WiFi. I'm assuming there is a conflict of pins between ESP8266WiFi.h and Servo.h?

Is there is solve for this? Thanks. I'd really love to be able to use the servo/wifi on the ESP8266 rather than relying on another MCU. Here is the error that happens when I have the "attach" command in my sketch. I also tried A0.

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

chrisallick, 2015-10-18:
Just following up on my comment. I moved "myservo.attach(5)" to the end of my setup function and it works! Connected to WiFi then attached my servo and it's good to go. Thanks!

Gargo, 2018-02-27:
nice. Now nodemcu "supports" servo library. Which means the code is compiled but won't work