esp8266/Arduino #9: Support the esp8266 in Firmata Boards.h

probonopd, 2015-03-29:
Firmata errors with
```
libraries/Firmata/src/Boards.h:337:2: error: #error "Please edit Boards.h with a hardware abstraction for this board"
 #error "Please edit Boards.h with a hardware abstraction for this board"
  ^
```

Possibly https://github.com/esp8266/Arduino/blob/esp8266/libraries/Firmata/src/Boards.h could be changed to support the esp8266?

scryb3, 2015-06-02:
This seems to be an important one, I'm researching adding a definition, but am starting at nothing, please feel free to help!

scryb3, 2015-06-02:
I tried this hack around and it seems to have worked for now (IOW I'm getting other errors now):

(in sketch for now)
#define __ESP8266__

(in boards.h before "// anything else")

// ESP8266
#elif defined(__ESP8266__)
#define TOTAL_ANALOG_PINS       15
#define TOTAL_PINS              15 // 16 GPIO
#define VERSION_BLINK_PIN       13
#define IS_PIN_DIGITAL(p)       ((p) >= 0 && (p) <= 15)
#define IS_PIN_ANALOG(p)        ((p) >= 0 && (p) <= 15)
#define IS_PIN_PWM(p)           digitalPinHasPWM(p)
#define IS_PIN_SERVO(p)         (IS_PIN_DIGITAL(p) && (p) - 2 < MAX_SERVOS)
#define IS_PIN_I2C(p)           ((p) >= 0 && (p) <= 15)
#define PIN_TO_DIGITAL(p)       (p)
#define PIN_TO_ANALOG(p)        ((p) - 14)
#define PIN_TO_PWM(p)           PIN_TO_DIGITAL(p)
#define PIN_TO_SERVO(p)         ((p) - 2)

scryb3, 2015-06-19:
I think that my cavalier copying of the text above, which did fix my compilation errors, is causing me other problems.

If someone more experienced can help me figure out the right config for this board I would appreciate it.

mralexgray, 2016-01-24:
is there any update on this?  support for firmata is really missed/would be great!

devyte, 2017-10-20:
3rd party lib. The referenced file above is no longer hosted in this repo.
Closing.

closed 2017-10-20 by devyte