esp8266/Arduino #15: Wire.setClock maximum frequency

ManiackCZ, 2015-03-30:
This is more a observation then issue report since you have already written in description that Wire.setClock gives weird frequency. I played with it a while and found that at 400 it gives frequency around 1khz. At 300000 and more it doesnt work at all since the scl is all distorted and non consistent (more like some noise). Maximum frequency I got was ~307kHz with set values around 180000-200000.

igrr, 2015-03-31:
Thanks for investigating.

The code that sets the frequency is [here](https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/cores/esp8266/i2c.cpp#L77):
```
void i2c_freq(int freq_hz)
{
	s_i2c_delay = 1000000 / freq_hz / 4 - 1;
	if (s_i2c_delay < 0)
		s_i2c_delay = 0;
}
```
So this function needs a tweak to calculate the correct delay, maybe by using a static lookup table.
@ManiackCZ Since you have started experimenting with this, would you mind hacking this a bit to measure the frequencies we get for each value ```s_i2c_delay``` in range from 1 to 16. Then we can add convert that data into a LUT and fix this.

ManiackCZ, 2015-04-02:
Oh, havent found that function when I was looking for it. Ok, I have few free days this week so I will try to poke around and let you know.

ManiackCZ, 2015-04-02:
Seems little bit rough to use delayMicroseconds(). It has big gaps between individual delay values due to operation with integers. What frequency are the ESP8266 modules usually running at? Are there any other sources to generate delays? (timers, way to use NOPs...)

igrr, 2015-04-02:
80MHz or 160MHz most of the time.
Maybe we can use a calibrated delay loop?
There's a CCOUNT special register which counts the CPU clock cycles. It can
be read in asm using RSR instruction.

ManiackCZ, 2015-04-02:
That seems promising. Can you send me source of your info about registers and assembly instructions?

ManiackCZ, 2015-04-04:
I thought I got it, but it was bahaving absolutely without any logic so I looked at my conversion algorithm itself and it seems to give 0 all the time. I thing there is some issue with calculation with decimal places.

ManiackCZ, 2015-04-21:
<b>UPDATE:</b> I have my final leaving exams coming in few weeks, so I will have to (unforunately) pause my work on the calibration but I plan to get right back into it after a month or so.
Thank you for understanding

igrr, 2015-05-05:
Fixed with the new i2c library.

closed 2015-05-05 by igrr