esp8266/Arduino #16: add support for "serial1" (TX only)

Links2004, 2015-03-30:
TX is working RX is disabled (only posible on GPIO 8 but we have no access to it)

in HardwareSerial.cpp in uart0_interrupt_handler add:
```CPP
    status = READ_PERI_REG(UART_INT_ST(1));
    if (status & UART_TXFIFO_EMPTY_INT_ST)
    {
        WRITE_PERI_REG(UART_INT_CLR(1), UART_TXFIFO_EMPTY_INT_CLR);
        Serial1._tx_empty_irq();
    }
```

HardwareSerial1.cpp
```CPP
/* 
  HardwareSerial1.cpp - esp8266 UART support

  Copyright (c) 2014 Ivan Grokhotkov. All rights reserved.
  This file is part of the esp8266 core for Arduino environment.
 
  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General Public
  License along with this library; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
  
  Modified 30 March 2015 by Markus Sattler (esp8266 platform support for UART1)
*/

#include <stdlib.h>
#include <stdio.h>
#include <string.h>
#include <inttypes.h>
#include "Arduino.h"
#include "cbuf.h"

extern "C" {
#include "osapi.h"
#include "ets_sys.h"
#include "mem.h"
#include "uart_register.h"
#include "user_interface.h"
}

#include "HardwareSerial1.h"

HardwareSerial1 Serial1;

uart_t* uart1_init(int baud_rate);
void uart1_set_baudrate(uart_t* uart, int baud_rate);
int  uart1_get_baudrate(uart_t* uart);
void uart1_uninit(uart_t* uart);
void uart1_transmit(uart_t* uart, const char* buf, size_t size);    // may block on TX fifo
void uart1_wait_for_transmit(uart_t* uart);
void uart1_transmit_char(uart_t* uart, char c);  // does not block, but character will be lost if FIFO is full

void uart1_set_debug(bool enabled);
bool uart1_get_debug();

struct uart_
{
    int  baud_rate;
[... 379 lines cut ...]
    inline size_t write(long n) { return write((uint8_t)n); }
    inline size_t write(unsigned int n) { return write((uint8_t)n); }
    inline size_t write(int n) { return write((uint8_t)n); }
    using Print::write; // pull in write(str) and write(buf, size) from Print
    operator bool() const;

    void setDebugOutput(bool);

    void _tx_empty_irq(void);

protected:
    friend void uart1_interrupt_handler(uart_t* uart);
    void _rx_complete_irq(char c);

protected:
    uart_t* _uart;
    cbuf*   _tx_buffer;
   // cbuf*   _rx_buffer;
    bool    _written;
};

extern HardwareSerial1 Serial1;

#endif
```

igrr, 2015-03-30:
Thanks!
I'm sure this can be done better than duplicating all that code? Maybe
passing an argument to the HardwareSerial constructor (UART0/UART1) will do
the trick?

Links2004, 2015-03-30:
yes i think this is possible, 
may i do this at weekend, today I simple do a fast implementation to test if it possible.

reaper7, 2015-04-01:
a good source of examples with multi uart/spi/i2c
(of course for different platforms like msp430, lm4f, cc3200)
https://github.com/energia/Energia/tree/master/hardware

igrr, 2015-04-01:
Fixed in #31

closed 2015-04-01 by igrr