esp8266/Arduino #17: Error on upload - Windows

pucbaldwin, 2015-03-30:
Any suggestions on what is going wrong here? Using the Arduino ESP8266 IDE to try and upload the HelloServer example.

ESP-01 is on COM4 on Windows via a FTDI TTL converter.
Board is set to Generic ESP8266. Programmer to ESPtool.

Any assistance gratefully received.

Cheers

Pete
```
esptool v0.4.2 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to none
	setting baudrate from 115200 to 115200
	setting port from COM1 to COM4
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat C:\Users\peter\AppData\Local\Temp\build4325786157015904206.tmp/HelloServer.cpp_00000.bin success
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	serialport_receive_C0: 00 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	serialport_receive_C0: 00 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	serialport_receive_C0: 00 instead of C0
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	serialport_receive_C0: 00 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	serialport_receive_C0: 00 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	serialport_receive_C0: 00 instead of C0
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	serialport_receive_C0: 00 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	serialport_receive_C0: 00 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	serialport_receive_C0: 00 instead of C0
warning: espcomm_sync failed
error: espcomm_open failed

```

igrr, 2015-03-31:
Can you try flashing something with [esptool.py](https://github.com/themadinventor/esptool) and the same hardware setup? Also, what USB-TTL chip are you using?

pucbaldwin, 2015-03-31:
USB-TTL is a FTDI232 from here

http://cgi.ebay.co.uk/ws/eBayISAPI.dll?ViewItemVersion&item=181576067020&view=all&tid=1303776937008

rogerclarkmelbourne, 2015-04-01:
@pucbaldwin 

Can you confirm this is purely an Arduino / esptool issue.

Uploading can fail if the PSU can't deliver enough current. 
e.g. board will function for AT command set, etc , if that firmware is installed, but its not possible to upload

sticilface, 2015-04-01:
I could not get the esptool methods to work with any of my usb-tty devices.  I ended up adding another programmer that was configured to use esptool.py, and it has not failed on me since!  

I'm happy as not it uploads in seconds.. with the HC06 it would take 10min...
Not sure if this helps you guys though

pucbaldwin, 2015-04-01:
Having done some more testing, I don't think this is an Arduino issue. While I can talk to the ESP8266 over the USB TTL and give it AT commands, I can't flash it using any of the available tools (esptool, NodeMcuflasher etc). I tried it with 2 AA batteries, and with a 3 v power supply so I don't think it's a power issue.

When I put GPIO0 low, and reboot, get the blue led, then using ESPlorer on 74880 baud I read repeated Fatal Errors from the ESP8266. I don't know why this is but is presumably related to the issue. I've got another ES8266 ESP-12 on order and will try again when it arrives.

4refr0nt, 2015-04-01:
Latest ESPlorer have new features: firmware autodetection. When GPIO0
in LOW state this not good idea, because ESPlorer send <CR><LF> and
waiting some answer (">" for NodeMCU and "ERROR" for AT firmawre). In
next version ESPlorer will be have option on Settings tab for disable
autodetection for using with any firmwares, and with GPIO0 pulldown
too.

pucbaldwin, 2015-04-07:
This now works - received new ESP-12. Initially had same problems but have mastered the reset / upload timing. Upload of code is however temperamental.

closed 2015-04-07 by pucbaldwin

jindrichsirucek, 2015-07-07:
It works for me, after I figured out I have to press the reset button on esp8266 when uploading at right time.. stupid, but it works in that way with basic Firmaware(AT commands)

bogzao, 2017-05-26:
> It works for me, after I figured out I have to press the reset button on esp8266 when uploading at right time.. stupid, but it works in that way with basic Firmaware(AT commands)

Same happended with me! I only managed to upload if the reset button is pressed before the uploading procedure. :)

shubhi35, 2017-08-19:
jindrichsirucek/bogzao, many many thanks! I killed my more than 5-6 hrs to get this resolved, did complete installation of Arduino from 1.6 to 1.8.3, instead of using BoardManager to get esp8266 support libraries used git version still was stuck ....finally it is your post which has helped me to see the light on the other side of the tunnel.

I echo your sentiments ...its really stupid ;-)

NischalBhandari, 2018-05-11:
having the same issue ... can anyone tell me any AT commands to communicate with the esp8266 inside the sonoff basic .??