esp8266/Arduino #18: Create API reference

igrr, 2015-03-30:
Describe all the basic API's and the libraries by adding documentation to the repository or to the wiki.

cimba007, 2015-04-01:
Is there a wiki yet?

igrr, 2015-04-01:
Yes there is!
http://www.esp8266.com/wiki/doku.php?id=arduino-docs

romanseidl, 2017-06-13:
I don't get why some autogenerated API-Doc should not be possible.

I never had to read as much code for using a well established library before. It makes me consider migrating to sming.

devyte, 2017-10-20:
I think the last comment is still rather valid. While we do have readthedocs, I don't think we have an automated way to generate docs for an api reference, and we probably should.

davisonja, 2017-10-22:
Doxygen! 😊

igrr, 2017-10-22:
We have some success with Doxygen + Breathe + Sphinx/readthedocs setup, i
think it can be replicated here. The only (obvious) thing is that all
header files need to receive actual API documentation.

davisonja, 2017-10-22:
Presumably we're thinking more basic api parameter/return/side effect rather than usage examples in the source? Keeping main docs for that?

igrr, 2017-10-22:
Not usage examples, just what Doxygen normally produces from header file
comments.
E.g.
http://esp-idf.readthedocs.io/en/latest/api-reference/system/log.html#api-reference

davisonja, 2017-10-25:
I've seen some people use Doxygen as full documentation. My experiments with that haven't led me to believe it has major advantages though. So it sounds as though we are thinking the same way.

atesin, 2022-09-03:
i think we should have 4 separate documentations here (as i do in arduino)

- arduino api, on which esp8266-arduino api is "familiar" (available at https://www.arduino.cc/reference/en/)
- xtensa c/c++ documentation on which espressif sdk api was built from
- espressif sdk api, on which esp8266 arduino was built from
- esp8266 arduino documentation (core and libs)

maybe we can build some documentation from source code with doxygen, but unfortunately source code header files (".h" files) does not cointain enough comments... until i could be wrong, with {esp-path} as = `%localappdata%\Arduino15\packages\esp8266\`, the involved folders i had found so far are:

- xtensa = {esp-path}\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include
- core = {esp-path}\hardware\esp8266\3.0.2\cores\esp8266
- builtin libs = {esp-path}\hardware\esp8266\3.0.2\libraries

check this post also = https://www.esp8266.com/viewtopic.php?p=95142#p95142