esp8266/Arduino #22: enable reset control for generic esp8266 boards - enhancement

nerdralph, 2015-03-31:
change
esp01.upload.resetmethod=none
in esp8266/boards.txt to:
esp01.upload.resetmethod=ck

rogerclarkmelbourne, 2015-03-31:
What does this change achieve ?

nerdralph, 2015-03-31:
It is documented in esptool-ck; connect RTS to reset and DTR to GPIO0, then
upload automatically resets the ESP and triggers the bootloader.
https://github.com/igrr/esptool-ck

rogerclarkmelbourne, 2015-03-31:
I guess there is no harm to change boards.txt, but this form of resetting is very dependent on the type of USB to serial board you have.

nerdralph, 2015-03-31:
I tried it with a cheapie pl2303hx under Windows 7, and it works fine.

rogerclarkmelbourne, 2015-03-31:
OK. I have one of those, so I'll test, and I can generate a pull request if its easier for Ivan than for him to do the change himself

igrr, 2015-04-01:
The reason I didn't set this as default is that when you open Serial Monitor in Arduino IDE, it pulls both RTS and DTR low, which puts the chip into bootloader mode. In fact some other terminals (screen?) also do that by default.
I can probably make this change but some people may get confused by this behaviour.

rogerclarkmelbourne, 2015-04-01:
Ivan

I think this can be dependent on what USB to Serial adapter you have. There are loads of postings on various embedded forums where people have tried to make these adapters to both reset and also to enable the bootloader, but the consensus is that its not reliable and only works for some people.

There was a big and heated discussion on Arduino STM32 forum, and in the end I decided not to try to implement it.

igrr, 2015-04-01:
For one esp8266 board I used the following solution:
[image]
This is supported in my fork of esptool using ```resetmethod=wifio```. Seems to be very reliable and works with all the terminals I have. But for most people having to find a PNP and a capacitor may be an issue.

nerdralph, 2015-04-01:
It makes no sense to disable it for everyone because it *might* not work
for some.  We're talking a one-line change here that won't affect anyone
that is using manual reset.

igrr, 2015-04-01:
> We're talking a one-line change here that won't affect anyone
> that is using manual reset.

Okay, makes sense. Will fix.

nerdralph, 2015-04-01:
"when you open Serial Monitor in Arduino IDE, it pulls both RTS and DTR low, which puts the chip into bootloader mode."
That is a minor nuisance if you want to do serial logging to have to unplug the RTS (and maybe DTR) line.  It allows you to get started programming an ESP-01 with only a USB-TTL module and some dupont cables.  I've been using mine that since it arrived in the mail a week ago.

igrr, 2015-04-01:
Option to disable DTR and RTS manipulation in serial monitor has also been added, so no issues with that now!

closed 2015-04-01 by igrr

JeroenBeemster, 2015-04-05:
iggr, seems THE solution to build an easy way to upload. "Where can I find the option to disable DTR and RTS in serial monitor. (I currently use version 1.6.3 of arduino).

rogerclarkmelbourne, 2015-04-05:
Jeroem

From what I recall the change was in board.txt

What I think may be a better solution for everyone is to have a menu defined in boards.txt that allows you to switch whether you want to enable the reset or not.

JeroenBeemster, 2015-04-06:
Hi Roger,
I found it in board.txt: 
generic.serial.disableDTR=true 
generic.serial.disableRTS=true
It is not in yours and in mine fork from you. 
Any idea when this will be get in sync. 
I am pretty new with github and I am loosing track now.
Anyway good work. I will test it asap and trying to update arduino esp8266 for beginners at http://www.arduinesp.com/

JeroenBeemster, 2015-04-06:
I tested the 
generic.serial.disableDTR=true
generic.serial.disableRTS=true 
on Arduino 1.6.3 (downloaded from arduino.cc) and it is not working. 
I assume this is a change also in the arduino java code. Will this change also be available in the "Original" build of arduino. Or should we use the build from igrr? 
Again, I can't wait for this change.  :)

ozayturay, 2015-04-07:
Does current Windows release from igrr's repository (arduino-1.6.1-p1-windows.zip) have this
generic.serial.disableDTR=true
generic.serial.disableRTS=true 
options enabled. They are in the esp8266/boards.txt but I cannot use Serial Monitor when DTR and RTS are connected.

igrr, 2015-04-07:
@ozayturay no this was added in 0dabb25. You need to clone the repository and build the modified Java  source to get this feature at the moment, sorry about that.

ozayturay, 2015-04-07:
Is there a simple tutorial for compiling the source for windows?

igrr, 2015-04-07:
@ozayturay For windows builds you need MinGW, JDK, ant, and some mingw utilities (untar i think).
Once you've cloned the repository, do the following in the MinGW prompt:
```
cd "your arduino git directory"/build
ant dist < NUL
```
If you find that something extra needs to be installed, please post an update here so I can add that knowledge into the readme. Thanks.

ozayturay, 2015-04-07:
Answering my own question; build/howto.txt takes me to https://code.google.com/p/arduino/wiki/BuildingArduino page and it looks simple enough. I'll try tonight. :)

ozayturay, 2015-04-08:
Here are simple instructions for building Windows version:

* Download *jdk-8u40-windows-i586.exe* from http://www.oracle.com/technetwork/java/javase/downloads/jdk8-downloads-2133151.html and install with default settings.

* Make an enviroment variable **JAVA_HOME** that points to **C:\Program Files (x86)\Java\jdk1.8.0_40** from computer properties. 

* Download *https://cygwin.com/setup-x86.exe* and install adding latest versions of **git, make, mingw-gcc-core, mingw-gcc-g++, perl, zip, unzip** and making sure the already default packages **coreutils, gzip, tar** are also selected.

* Download *http://ftp.itu.edu.tr/Mirror/Apache//ant/binaries/apache-ant-1.9.4-bin.zip* and extract it's contents to folder **C:\Cygwin\opt** (create folder opt if not exists)  and using your favourite editor (mine is NotePad++) at the end of file **C:\Cygwin\home\yourwindowsusername\\.bash_profile** add these two lines:

>export ANT_HOME=/opt/apache-ant-1.9.4
>export PATH=/opt/apache-ant-1.9.4/bin:$PATH

* Enter Cygwin enviroment using the icon from desktop and use these commands to clone and build the source:

>$ mkdir Source
>$ cd Source
>$ git clone https://github.com/esp8266/Arduino.git
>$ cd Arduino/build
>$ ant dist

After a long downloading and building process you can find your freshly build **arduino-1.6.1-windows.zip** file in **C:\Cygwin\home\yourwindowsusername\Source\Arduino\build\windows** folder. ;)

JeroenBeemster, 2015-04-10:
ozayturay, Could you share your executable. I am dying for this solution. 
Currently waiting for ivan that he update the zip. But I assume he working something great now.
I am working on some simple examples and this makes it much more easy to use.

ozayturay, 2015-04-10:
Here it is: https://dl.dropboxusercontent.com/u/18575245/ESP8266/Arduino-1.6.1_ESP8266.zip

Copy the included files over your current ESP8266 Arduino (probably arduino-1.6.1-p1-windows.zip) and have fun. ;)

igrr, 2015-04-11:
@JeroenBeemster  The next release is currently blocked by availability of a proper windows toolchain, as discussed in #23 and elsewhere.
But there is really no need to wait for the release if you need the latest changes. You can always grab the latest build from the CI server:
https://ci.appveyor.com/project/igrr/Arduino/build/artifacts
As mentioned above,
>Don't forget to copy libiconv-2.dll from hardware\tools\avr\libexec\gcc\avr\4.8.1 to the root (side by side with arduino.exe)

ozayturay, 2015-04-11:
Wow that CI sever is really cool. Is the build process automatic or manually activated by you when required?

igrr, 2015-04-11:
@ozayturay It's configured to build whenever there is a new commit, but it's also possible to run the build manually.

JeroenBeemster, 2015-04-12:
@ivan Thanks. Its working
@ozayturay Thanks for your quick reply

Toshik, 2015-04-26:
@igrr 
Your schematics to autoflash is interesting, but it will strictly limit GPIO0 usage.
For example, now I use ESP-01 as I2C device on ports GPIO0 and GPIO2, in case I would attach your schematics, I will kill I2C when ESP uses UART

probonopd, 2015-06-28:
Now that we are using the Boards Manager, I think the change to `Arduino/arduino-core/src/processing/app/Serial.java` from https://github.com/esp8266/Arduino/commit/0dabb25963852c072f6d8428c5b3e2884532f282 needs to be made in https://github.com/arduino/Arduino in order to become effective.

If you agree, please create a pull request to https://github.com/arduino/Arduino containing the change.

JeroenBeemster, 2015-07-23:
2 NPN transistors did the trick See #533
I have made an example over here http://www.arduinesp.com/blink

d-a-v, 2015-11-07:
Hi,
I really like this, I can now dev from home on the chip at work.
I updated the patch above and submitted to arduino. 
https://github.com/arduino/Arduino/pull/4102
Credit goes not to me, but I want this included. Maybe some of you could +1 right there, or comment better than I did.

tablatronix, 2015-12-22:
Looks like 
https://github.com/arduino/Arduino/issues/4102
is in limbo ?

d-a-v, 2016-01-14:
It is now getting somewhere else. I don't know about politics, people in charge, global orientations, ...
What about a specific arduino fork for esp for making it easier to people to get things working fast ?
Would it be too much of work for maintaining ?
Has it been tried before ?

ab3nd, 2016-06-09:
Is the version of the development environment that had configurable DTR/RTS lines available anymore? I notice that the option is in the ESP8266 for arduino package's boards.text, but it doesn't have any effect on the IDE.

ozayturay, 2016-08-26:
Compilation instuctions in the comment https://github.com/esp8266/Arduino/issues/22#issuecomment-91043388 above updated, 1.6.11 compiled without problems and works as expected.

d-a-v, 2017-07-18:
one of the patches submitted to original arduino has just been merged (https://github.com/esp8266/Arduino/issues/22#issuecomment-154594053)
2 years later :)

tablatronix, 2017-07-18:
w00t!

probonopd, 2017-07-18:
Since there is no `resetmethod=none` anymore in boards.txt it looks like this will start to work in the Arduino IDE nightly builds...

beicnet, 2017-07-18:
@d-a-v Don't worry, it's all up to date with 108 poll request! ;)

commit 8b081ebe (2015-04-01): Enable reset control for generic esp8266 boards

Close #22