esp8266/Arduino #23: build gcc with lto support - enhancement

nerdralph, 2015-03-31:
The included gcc wasn't built with the linker plugin, so enabling link-time optimization (-flto) fails.
dlopen:'.../arduino-1.6.1/hardware/tools/esp8266/xtensa-lx106-elf/bin/../libexec/gcc/xtensa-lx106-elf/4.8.2/liblto_plugin-0.dll': The specified module could not be found.
Lto not only reduces code size, it can significantly improve performance by doing constant folding for function calls like digitalWrite.
http://nerdralph.blogspot.ca/2014/04/gcc-link-time-optimization-can-fix-bad.html

rogerclarkmelbourne, 2015-03-31:
I think there is also an issue with the way gcc was compiled, so that the libiconv-2.dll, is required to be in the root of the Arduino IDE

This would prevent the repo being deployed by the new Boards Manager in IDE 1.6.2 or newer

igrr, 2015-04-01:
Yes, thanks, looks like there are currently 3 issues that need to go into the toolchain: 
- this one, adding lto support
- [the dependency on libiconv](https://github.com/rogerclarkmelbourne/arduino-esp8266/issues/2)
- [need to rebuild ulibc without malloc and free](https://github.com/esp8266/Arduino/issues/11)

For Windows I used this build script:
[https://gist.github.com/fpoussin/7ae55b5a5bd9a28ce21d]

For Linux and Mac I used crosstool-NG (branch lx106-g++).

@nerdralph what options do we need to pass to build gcc with lto support?

rogerclarkmelbourne, 2015-04-01:
Ivan,

Did you manage to fix the libiconv-2 dependency issue ?

PS. How did you run the script under windows Mingw?

igrr, 2015-04-01:
No I didn't have a chance yet — I'm running OS X and for all the Windows stuff I have use my old laptop which I usually can access on weekends.

For the build, yes, I did use mingw.

rogerclarkmelbourne, 2015-04-01:
OK
No worries
Thanks
Roger

nerdralph, 2015-04-01:
It was a lot of work to figure out lto support for avr-gcc, so I imagine it
won't be any easier for the lx106.  From what I recall, it depended not
only on enabling a configure option, but also on the "gold" linker plugin
version.
Right now I'm trying to figure out the lx106 instruction set including
which are 16/24 bit so I can write some code in hand tuned asm.  After that
I'll look at the lto build if nobody figures it out in the meantime.
Ideally the expressif folks should release a new sdk with GCC 4.9.2 and
working lto.

igrr, 2015-04-02:
There should be a way to tell gcc's configure script to link statically to libiconv. But I don't know which exact options are required for this though...

igrr, 2015-04-02:
Perhaps this one might be useful: http://stackoverflow.com/a/14112368

rogerclarkmelbourne, 2015-04-02:
Ivan

I've tried building on 2 different Windows 7 x 64 machines (more core i5 and one core2) and both just hang after a while in gcc :-(

This part of the script gives a warning ./ error 
```
  find $XTDLP -maxdepth 1 -type d -name gmp-* | xargs -i mv -v {} $XTDLP/$GMP
  find $XTDLP -maxdepth 1 -type d -name mpfr-* | xargs -i mv -v {} $XTDLP/$MPFR
  find $XTDLP -maxdepth 1 -type d -name mpc-* | xargs -i mv -v {} $XTDLP/$MPC
```

but I think its OK, because its just a clumsy way to rename the file to what its already called (and fails on MinGW).

I guess I could try installing XP as a VM and see if it doesn't hang when I do that. 
(I have an XP machine but its old and slow and doesnt have much ram so would take ages to compile I suspect)

:-(

igrr, 2015-04-03:
Oh, right, it does hang quite a lot.
For me turning off parallel compilation helped a lot:
```JOBS=-j6``` => ```JOBS=```

rogerclarkmelbourne, 2015-04-03:
Thanks 

I changed to -j1 , and its no longer hanging

I'll see if it manages to complete a compile and then see what static link settings need to change, if any

I'll also need to do some research about Lto as recommended by @nerdralph

rogerclarkmelbourne, 2015-04-03:
Ivan
Sorry to keep bothering you, but I'm now getting a build failure on line 164 because (the ../configure line)
because configure has not been created in 

```
echo "Building first stage GCC"
cd $XTDLP/gcc-xtensa/build-1

dl\gcc-xtensa

did you make any other changes to the script that you can remember ?

if [ $RECONF -gt 0 -o ! -f .configured ]; then
  rm -f .configured
  ../configure --prefix=$XTTC --target=$TARGET --enable-multilib --enable-languages=c --with-newlib --disable-nls --disable-shared --disable-threads --with-gnu-as --with-gnu-ld --with-gmp=$XTBP/gmp --with-mpfr=$XTBP/mpfr --with-mpc=$XTBP/mpc  --disable-libssp --without-headers --disable-__cxa_atexit
  touch .configured
fi
```

igrr, 2015-04-03:
Maybe I did... I'll check on my windows laptop tomorrow.

rogerclarkmelbourne, 2015-04-04:
I'm not sure why this works, but Sampreet's repo works on IDE 1.6.3 without needing libiconv-2.dll in the IDE root directory.

Anyway. On the build front. g++ and a few other things are still not getting built for me, so I'll need to track down where in build.sh creates them.

Re: LinkTimeOptimization

It looks like currently adding -flto makes no difference to the output size, but it still compiles OK.

But as @nerdralph  has indicated, gcc would need to be built with LTO enabled as part of the build, and at the moment, its not too clear how to do this, or whether its possible on the Xtensa version of gcc

I found this reference https://gcc.gnu.org/wiki/LinkTimeOptimization

Which seems to say that the LTO version is a branch from the main GCC repo

But I can't see a branch for LTO on https://github.com/jcmvbkbc/  so I guess the only thing would be to try adding the  --enable-lto and see if it gets passed down into the actual compile in the same way as --enable-static is supposed to.

However at the moment I think this is not a top priority, as its probably a lot of work / waiting for compiles to finish etc, for possibly not too much difference for most people

nerdralph, 2015-04-04:
Gcc 5.0 has a lot of new IPA optimizations, even without lto:
https://gcc.gnu.org/gcc-5/changes.html
Hopefully Max moves to 5.0 from 4.8.2 soon.
If you haven't read Honda's blog, it explains his work on lto.
http://hubicka.blogspot.ca/2014/04/linktime-optimization-in-gcc-2-firefox.html

igrr, 2015-04-11:
@rogerclarkmelbourne were you able to make some progress with the gcc build?

rogerclarkmelbourne, 2015-04-11:
Hi Ivan

I tried on a Window XP VM, but it kept hanging like it did before I change the jobs settings

But the jobs settings was -j1 
I also tried
 jobs=

But nothing worked :-(

I have been too busy with the Boards Manager issues on my other repo STM32

If you send me your XP build script I can try it, but apart from that I don't know how to proceed.

PS. I'm not sure if we also need to try gcc 5, but I think it will probably have its own problems ;-)

igrr, 2015-04-11:
I checked my build script and it's no different from the one linked above, aside from -j1 option.
Guess i need to get myself a windows machine, perhaps some cheap option from Azure Cloud will do.

rogerclarkmelbourne, 2015-04-11:
Ivan,

I have been using Azure for work, but its expensive and not very fast unless you pay $$$$ .

I'm not sure if its possible to request a 32 bit machine on Azure.

I think you would be better off dual booting your computer or using a VM. 

Do you run Linux or OSX?

I ran an vm on OSX with XP int it. ( I think I used virtual box )

But I also tried XP 32 bit under VM on windows 7 x64, and the compiler keeps hanging :-(

I guess I should find a copy of Windows 7 , 32 bit and run that in the  VM and see if the compiler still hangs.

But perhaps the compiler doesn't work in the VM ( virtual box), and perhaps I need to try VirtualPC

rogerclarkmelbourne, 2015-04-12:
@nerdralph 

What version of MinGW are you using ?

I have been using http://sourceforge.net/projects/mingw/postdownload?source=dlp but I noticed there is another version called MinGW-W64 which seems to be gaining traction and claims to be contain the latest / later releases of gcc

However, it doesn't seem to have bash shell so I can't fun the script that Ivan uses

Anyway, I've made a new Windows 7 X86 VM using Microsoft Virtual PC and I will see if that runs Ivans' script any better than the Oracle Virtual Box VM running XP 32 bit.

nerdralph, 2015-04-12:
Roger,
I'm using the same version.  Yesterday I built avr-gcc-4.9.2 on mingw
(since I'm experimenting I figured I'd build something I can use).  Builds
can hang if you use parallel make (-j>1) or don't use -D__MINGW_ACCESS.
I also built the 20150405 snapshot on Linux.  My build script downloads
gmp, mpc, and mpfr sources and links them in the gcc source dir.  I still
haven't figured out the xtensa target configuration options from the gcc
docs, and may have to dig into the source.

nerdralph, 2015-04-12:
That should be __USE_MINGW_ACCESS

nerdralph, 2015-04-12:
Just tried another avr-gcc build on MinGW, with __USE_MINGW_ACCESS, and make -j2, and it locked up again.  This is with the stock mingw gcc-4.8.1 and binutils.  Looks like parallel builds are out for now.
Looking through gcc's config files, I see config/mh_mingw already defines __USE_MINGW_ACCESS.

rogerclarkmelbourne, 2015-04-12:
Ralph

Thanks for the info.

I tried using Microsoft Virtual PC and a copy of Windows 7 x86, instead if virtual box, but the shared drives with the host are not working, ( aaarrrghhh), which makes it a pain to use, so I may go back to Oracle Virtualbox but with Window 7 x86 rather than XP.

Where in the script do you add __USE_MINGW_ACCESS ?

nerdralph, 2015-04-12:
See my other comment about mh_mingw.
If you want to add some other flag, add CFLAGS="..." to the configure
command.

rogerclarkmelbourne, 2015-04-12:
Ralph
Thanks.

BTW. Just turned on my W7 machine (i.e cold reboot) and the shared host drives are now working on Windows Virtual PC, so I'm now copying over all the downloads from when I tried it on my host machine (W7 x 64) and I will add that flag and try again.

rogerclarkmelbourne, 2015-04-13:
Ralph,

I've managed to get the build to work on my Windows Virtual PC / hosting Windows 7 32 bit.

However it took around 5 hours to complete. I'm only running it on a core 2 duo machine, ie its only got 2 cores, but the clock speed on the cpu is quite fast (3.15Ghz)

I do have a Core i5 machine I could run it on, but I don't know if it would be any faster as its only single tasking the compile (jobs=-j1) and my i5 machine is clocked at a slower speed (around 2.5G I think)

I'll now need to copy the files from the VM back to my Arduino ESP8266 installation and see if it actually compiles ;-)

But of course, we still have not made any changes eg not done anything to get around the static linking issue etc

The git repo for xtensa gcc

https://github.com/jcmvbkbc/gcc-xtensa/commits/call0-4.8.2, 
nor has newlib or the esp-binutils repo (as far as I can see )

Looking at the build config, it does pass in --enable-static, but if this was the build config that Ivan used, it doesn't seem to statically link so that libiconv-2.dll is no longer required to be in the root of the windows IDE :-(

However I will double check later in case my build behaves differently

PS. One other avenue is to see where arduinesp.com got their compiler from, as it definitely has different build settings, (it doesn't need libiconv-2.dll , but from what I understand from Ivan there are other settings which are incorrect in the build  www.arduinesp.com are using.

I'll see if I can contact arduinesp.com and find out

nerdralph, 2015-04-13:
Good to hear you got a build going.  I know how slow going it is; I don't
think my PC is much faster and my Linux VM is only 512M and a single CPU
instance.  After seeing Ivan's reference to Appveyor I tried out Drone and
Travis.  My Drone builds timeout at 15min.  Travis is smokin' fast but I'm
hitting their 4MB log output limit 4min into my builds.
I know about the gcc xtensa fork, but I see commits from Max on the gcc
latest official branch, so thats the route I want to try.  Gcc 5 not only
has improvements to lto, but also optimization improvements that should
mean guys like me that are obsessed with code performance and size won't
have to delve into assembler so often.

Regarding static builds, I'd like to be able to use a local shared lib path
since it means big code size savings.  With static, nm, ar, objdump, etc
all have huge amounts of duplicate code.

rogerclarkmelbourne, 2015-04-13:
Ralph

Re: Static builds. 

The issue was just libiconv-2.dll, but the only references I can find to problems with it, related to static builds. 
But it could be a completely different problem, as there was very little info on the web that I could find about it.

When I copied my freshly built xtensa gcc into the Arduino hardware folder, I get the same error that I get with Ivans build

i.e xtensa-lx106-elf-g++.exe
"The program can't start because libiconv-2.dll is missing"    etc etc

The problem is that the dll currently needs to be copied into the IDE install folder, which is hard for some people to do and means this can't be a seamless install  e.g. via the boards manager
Actually the boards manager can run a post install bat (apparently), which could copy the file, but it would be hack to do so. 

Better if we can remove this error by changing a build setting.

Re:GCC 5

Any idea who we update the GCC repo file to version 5, as its currently v4.8.2 as far as I can tell

nerdralph, 2015-04-13:
Roger,
If you look on the gcc mailing list and in the svn repo you'll see commits
approved that came from Max, so I think the mods he made to his 4.8.2
branch should be in the gcc5 snapshots.  Once I have the binutils/gcc
configure&build process mastered I'll try building them with target=xtensa.

nerdralph, 2015-04-13:
The timing for gcc5 is working out well...
https://www.mail-archive.com/gcc@gcc.gnu.org/msg76338.html

rogerclarkmelbourne, 2015-04-13:
Ralph

OK. Let me know how you get on.

Re: Libiconv-2.dll

I found this,  http://www.gurucoding.com/en/rpi_cross_compiler/compacting_and_testing.php  which basically says the issue is because its compiled in MinGW and effectively assumes the compilor is going to run under MinGW and hence references MinGW dll's (like libicon-2.dll)

However the fix listed on that page doesn't work. (e.g. copying libiconv-2.dll to a specific location)

I did try copying the dll to the folder containg the compiler exe's and every other folder containing an exe of a binary, but unless I missed a folder, that method to fix doesnt work :-(

However if its a MinGW issue, then I can try searching for solutions to dll path related to MinGW files on installations that don't have MinGW installed on them

re: GCC 5.
I'm downloading it now.

I'm not an expert in compiling GCC, so I've no idea if the build target of "xtensa-lx106-elf" is something that is defined in GCC by default ?

nerdralph, 2015-04-13:
I'm not an expert in compiling gcc either, but I'm slowly getting there.
You need to build binutils first (go with 2.24 instead of 2.25), since the
gnu assembler is in binutils.  Then build gcc.
I think the configure option for the target should be just xtensa, and when
invoking the compiler a -mmcu=lx106 or something like that should be used.
You can get an idea of the build process if you look at the avr-gcc build
scripts I'm working on:
Github.com/nerdralph/nerdralph/avr/make-avr-gcc

nerdralph, 2015-04-13:
Here's the direct link:
https://github.com/nerdralph/nerdralph/tree/master/avr/make-avr-gcc

rogerclarkmelbourne, 2015-04-13:
Ralph

Thanks.

I've done some more research on the libiconv issue and its supposed to be fixed by a linker flag of -static-libgcc or -static  but looking in the Makefiles for gcc build1 and build2 those flags are defined)

see http://stackoverflow.com/questions/4984612/program-cant-find-libgcc-s-dw2-1-dll

re: BinUtils  
Ivans script uses 
https://github.com/fpoussin/esp-binutils

which was last updated 6 months ago

There are no docs in that repo about what makes it an esp version :-(

igrr, 2015-04-13:
@rogerclarkmelbourne i think you may want to try building libiconv from source as a static library and make sure the gcc is linked to the static version of libiconv (libiconv.lib instead of libiconv.dll).

rogerclarkmelbourne, 2015-04-13:
Ivan

Thanks. That sounds like it could be the issue.

I will investigate what I need to change in order to compile it that way. It must be something in the automake templates, because I think the Makefile is generated with automake

rogerclarkmelbourne, 2015-04-13:
Ivan and Ralph

It looks like I need to get lib iconv from here. 
https://www.gnu.org/software/libiconv/#downloading

And build it as a static library ( which on windows doesn't seem to be the default build target)

Then the Makefile for gcc needs to be changed to link the static library. 

Which will mean somehow changing the gcc Makefile, but the Makefile is actually created by Automake.

So I presume the correct way to do this is to somehow change the Automake template, but this is all new to me.
I have a basic understanding of Make files, but Automake is new to me

nerdralph, 2015-04-13:
Roger, first look for a configure option for gcc... Perhaps
--disable-shared.
https://gcc.gnu.org/install/configure.html
Also look at Max's crosstool-NG xtensa repo, which seems to encapsulate the
build of binutils, gcc, and possibly libc.

rogerclarkmelbourne, 2015-04-13:
Ralph,
I'll need to double check, but I was pretty sure that --disable-shared was already in the build scripts

It looks like esp-newlib contans iconv sources

BTW. 
The other possibility is to use PFalcon's build process as outlined here 

https://github.com/pfalcon/esp-open-sdk

However its not clear how to build this on Windows.
PFalcon mentions "Windows with Cygwin"

I guess perhaps I should post an issue on that repo to see if there are windows build instructions

igrr, 2015-05-05:
@rogerclarkmelbourne I got my windows laptop back and looks like I'm making progress.
Downloaded and build libiconv with ```--disable-shared``` flag, added an ```--with-libiconv-prefix=/path/to/my/libiconv``` option to gcc configure call and also removed libiconv-dev package from MinGW (the last one was the key - with this package installed, configure would pick the shared libiconv from MinGW). Checked the resulting xtensa-lx106-elf-gcc with Dependency Walker, and looks like it doesn't depend on libiconv-2.dll any more!

rogerclarkmelbourne, 2015-05-05:
Ivan

OK.

I'm not sure if you saw the issue has been fixed in Sanpreet's repo, as someone found if you add the dll in 2 different places, it solves the problem

But if you have totally removed the dependencies that's an even better solution

igrr, 2015-05-05:
No I haven't seen that solution, will check.

rogerclarkmelbourne, 2015-05-05:
https://github.com/sandeepmistry/esp8266-Arduino/commit/0ce775c41ec79e64895a4e288441bb539f60bf4e

igrr, 2015-05-05:
That way is actually preferrable because the code is not statically linked into multiple gcc executables (gcc, gcc-version, g++, c++, etc). Will update my toolchain archive accordingly.

rogerclarkmelbourne, 2015-05-05:
Cool... Glad to be vaguely useful ;-)  (as I failed to recompile the compiler )

nerdralph, 2015-05-05:
Based on builds I've done of gcc on Linux, static linking had less of an
impact on size than I expected.  Only the parts of the shared library that
are used by the executable get linked in, whereas with shared libs, you
have the whole library including the unused parts.  Also many of the gcc
executables are links to the same file (like c++ and g++).

kireevco, 2015-12-16:
I'm working on building an alternative sdk: kireevco/esp-alt-sdk, which is based on similar concept as esp-open-sdk, but is meant to work on all plarforms, including Windows. It's not using crosstool-ng, and should work on mingw32,64 and probably cygwin. Currently I have a build working, some people reported errors, some people reported it working. I'm experimenting with https://github.com/fpoussin/esp-binutils vs https://github.com/jcmvbkbc/binutils-gdb-xtensa (latter has gdb in it).

Any thoughts/testing/contributions are welcome!

nerdralph, 2015-12-16:
I think the ideal would be taking Max's work and creating a script to add
the lx106 files to a stock gcc source tree.  4.8 is no longer maintained.
4.9 is the minimum version you should be at, and preferably 5.2.

kireevco, 2015-12-16:
@nerdralph not sure what do you mean, can you clarify please?

nerdralph, 2015-12-16:
Max's gcc-xtensa is based on gcc 4.8.2.
The lx106 is not a supported target in the stock gcc sources.
Take Max's mods to the 4.8.2 tree and create a script or patch file so you
can add lx106 support to the stock 4.9.3 and 5.2 sources.

kireevco, 2015-12-16:
So, you want to take https://gcc.gnu.org/releases.html, and add lx106? That's pretty tough step, I think it's been maintained here https://github.com/jcmvbkbc/gcc-xtensa/tree/lx106-5.1?

kireevco, 2015-12-16:
By the way, how crosstool-ng is structured? I can't find a clear set of instructions how they build binutils?

nerdralph, 2015-12-16:
You were asking for feedback; you didn't say it had to be easy. :-)
Optimization in 4.8 without lto is pretty lame.
If ipa-ra in 5.2 can work with the lx106, I expect you'd see 10-25%
improvement in speed and code size.

kireevco, 2015-12-16:
@nerdralph 
> You were asking for feedback; you didn't say it had to be easy. :-) fair enough:)

nerdralph, 2015-12-16:
I didn't get very far into crosstool-ng before I decided it was an option I
didn't want to pursue.

kireevco, 2015-12-16:
@nerdralph that's the issue I'm trying to solve with esp-alt-sdk. Crosstoon-ng is way far from working ok on Windows, and I need to re-do the logic that is under `./ct-ng xtensa-lx105`...

kireevco, 2015-12-16:
@nerdralph just for fun, I'm trying to build binutils, gcc for xtensa target... we'll see how it goes.

milkpirate, 2017-06-16:
Dont know if this issue is still valid but since I was searching for a newer GCC version (better `constexpr` handling) for the ESP8266 I stumbled across it and a little later across that: 

Binutils | GCC | libc | GDB | Download link
|:-|:-|:-|:-|:-|
2.24 | 5.2.0 | Newlib 2.0.0 | 7.11 | esp8266-gcc5.2.0-r10.exe

Link: [http://gnutoolchains.com/esp8266/](http://gnutoolchains.com/esp8266/)

That might be the desired GCC 5.2.

devyte, 2017-10-20:
@igrr we discussed briefly moving to a newer toolchain, namely 5.2, which is the same as mentioned here. Would that address the issues discussed above?
If so, I suggest tracking here, but I'd change the issue title.

[1 more reply]