esp8266/Arduino #27: Floating Point Converstions

sticilface, 2015-04-01:
I need to convert numbers entered in a browser (Strings) or sent by MQTT  (char array) to a floating point number (double).

I've been trying and failing various ways, but I'm pretty new to C (only since i got hold of the beta for this) so i've failed.  

Igrr has said that atof has not been implemented ... this is what is in core_esp8266_noniso.c
``````
float atof_internal(const char* s)
{
  float result = 0;
  return result;
}
``````

atol does work but I'd kinda like the decimals...  ( i know i can just enter an integer and then divide...)

Does anyone have any ideas... or fancy giving me a quick patch for this.. I'd try.... infact i spent a few hours on it.. but I'm so new to arrays, pointers, and all this that i just fail....

Cheers

A

igrr, 2015-04-01:
You can try this:
```
double os_atof(const char* s)
{
	double rez = 0, fact = 1;
	while (*s && (*s == ' ' || *s == '\t' || *s == '\r' || *s == '\n'))
		s++;
	if (*s == 0)
		return 0; // can't read

	if (*s == '-')
	{
		s++;
		fact = -1;
	};
	for (int point_seen = 0; *s; s++)
	{
		if (*s == '.')
		{
			point_seen = 1;
			continue;
		};
		int d = *s - '0';
		if (d >= 0 && d <= 9)
		{
			if (point_seen)
				fact /= 10.0;
			rez = rez * 10.0f + (double) d;
		};
	};
	return rez * fact;
}
```

[source](https://github.com/anakod/Sming/blob/master/Sming/system/stringconversion.cpp#L93)

I'm not 100% positive this is LGPL though.

igrr, 2015-04-01:
Yeah, and you obviously need to change the function name and change double to float.

sticilface, 2015-04-01:
Thanks for the speedy reply..  If I cut and paste that function straight into my sketch, it works perfectly.  

I tried what you suggested, changed the name of the function to atof_internal, changed return type to float, -/+ changing internal types to float, and it does not compile...

/Users/amelvin/Downloads/Arduino-espfinal.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: MQTT-Melvide_-_v1.0.2.cpp.elf section `.text' will not fit in region `iram1_0_seg'
/Users/amelvin/Downloads/Arduino-espfinal.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-mallocr.o):(.literal+0x1c): undefined reference to `_sbrk_r'
/Users/amelvin/Downloads/Arduino-espfinal.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-mallocr.o): In function `malloc_extend_top':
mallocr.c:(.text+0x5e): undefined reference to `_sbrk_r'
mallocr.c:(.text+0x127): undefined reference to `_sbrk_r'
/Users/amelvin/Downloads/Arduino-espfinal.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-freer.o): In function `_malloc_trim_r':
mallocr.c:(.text+0x366): undefined reference to `_sbrk_r'
mallocr.c:(.text+0x38f): undefined reference to `_sbrk_r'
/Users/amelvin/Downloads/Arduino-espfinal.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-freer.o):mallocr.c:(.text+0x3a0): more undefined references to `_sbrk_r' follow
collect2: error: ld returned 1 exit status
Error compiling.

its out old friend _sbrk_...

Im happy though as the function works just fine inserted into the sketch.  cheers for your help

commit eeabbf45 (2015-04-02): Add atof implementation

Fix #27
closed 2015-04-02 by igrr