esp8266/Arduino #28: ESP8266WiFiClass::begin Starts in STA + AP mode?

Links2004, 2015-04-01:
is there a reason for starting in WIFI_AP_STA mode?
in the default config the AP is "open" (no WPA2, no password), this can be a big risk of the security for the application or the network.  

```cpp
int ESP8266WiFiClass::begin(const char* ssid, const char *passphrase)
{
    if (wifi_get_opmode() == WIFI_AP)
    {
        // turn on AP+STA mode
        mode(WIFI_AP_STA);
    }
```

i preferred to only open the AP if the user really need it.

```cpp
 mode(WIFI_STA);
```

igrr, 2015-04-01:
We're only going from AP to AP_STA, we don't enable AP if it wasn't enabled previously.

Imagine the following code:
```
WiFi.softAP("esp", "password");
WiFi.begin("other_network", "whatever");
```
If we would just do ```mode(WIFI_STA)``` in ```begin``` this code would break.

Links2004, 2015-04-01:
i never call softAP in my code only begin and i get an AP?

igrr, 2015-04-01:
Yes, because the SDK libs store the WiFi configuration.
But I understand your point, I will add a flag to check whether softAP was called.

For now, please use ```WiFi.mode(WIFI_STA);``` as a workaround.

Links2004, 2015-04-01:
ok thanks.

Links2004, 2015-05-21:
fixed https://github.com/Links2004/Arduino/commit/3c7f046009e5d65fc6dfecbb3c945bd7a61c68cc

closed 2015-05-21 by Links2004