esp8266/Arduino #34: WDT fires if setup() takes more than 1 sec

Sermus, 2015-04-02:
Should WDT be disabled during setup()?
It's quite common situation when setup() executes quite long time.

tomaja, 2015-04-02:
Yes, I also have this problem. I would like to disable WDT during setup() because setup waits for WiFi connection that is used to deliver settings for further operation.

igrr, 2015-04-02:
Can you call yield() or delay() in your setup function periodically in the
same way it is done in the sample sketches for the ESP8266WiFi library?
Both will indirectly reset the watchdog.

tomaja, 2015-04-02:
Oh, yes - delay works! Sorry for that, I didn't get that from README :(
Спасибо, Иван!

igrr, 2015-04-02:
@Sermus 
Maybe adding methods to enable/disable WDT would be sufficient for most cases?

Sermus, 2015-04-02:
Not sure this is a way out because i got it in a 3rd party library which i tried to use as is. Its initialization tooks long time. In that sense if we want to have one-to-one Arduino library portability we must let them initialize as long as they need. Otherwise i'd have to modify a lib inserting watchdog feeds (if it's possible at all)

igrr, 2015-04-02:
I meant something like this:
```
void setup() {
  wdt_disable();
 /// some initialization
  ThirdPartyLibrary.begin();
/// more stuff
  wdt_enable();
}
```
Would that work?

What exactly is this library doing, by the way? I have added workarounds for some of the common usage patterns like ```while (!wifiClient.available()){}``` which detect such blocking loops and call ```yield()``` internally.

Sermus, 2015-04-02:
it's ILI9341 lib and it fills the screen black in quite a longy way.
What is the issue with nesting setup() call in these wdt_disable()/wdt_enable(), except not working WDT itself?

tomaja, 2015-04-04:
delay() doesn't seem enough. I probably have more than 1s in between two delay calls.
Yes, your proposition looks good (adding wdt_enable and wdt_disable).
Can you add those two functions?

Thanks!

anteph, 2015-04-05:
It would also be usefull to have a function to set the wtd reset time!
Thanks!

tomaja, 2015-04-10:
@igrr Do you think this feature will be added?

`void setup() {
  wdt_disable();
 /// some initialization
  ThirdPartyLibrary.begin();
/// more stuff
  wdt_enable();
}`

igrr, 2015-04-11:
@tomaja Added WDT APIs: ```ESP.wdtEnable()```, ```ESP.wdtDisable()```, ```ESP.wdtFeed()```.
You can either grab the latest version from git, or download the full .zip when it is ready from https://ci.appveyor.com/project/igrr/Arduino/build/artifacts

tomaja, 2015-04-11:
Great! Thanks, @igrr !
I just built Mac version from source and I checked that I can use `ESP.wdtDisable()` - script compiles just fine.
My Prolific USB to Serial stopped working after OSX update this morning so I'm unable to flash the new image to ESP8266 until I resolve that. I'll let you know how it works.

igrr, 2015-04-11:
@tomaja I have also installed the Yosemite update today and had to reinstall the FTDI driver to make serial uploads work. I suppose the driver got replaced by Apple's version during the upgrade.

marvinroger, 2015-05-15:
Sorry to trigger a notification, but I would like to understand what is the purpose of the watchdog? I can't find any informations. Thanks!

AndyKirby, 2015-05-15:
A Watchdog timer is a safeguard against infinite loops or crashed
processors.

You periodically reset the watchdog timer. If you don’t it forces a
reset of the processor.

Crashed processors are usually stuck in a giberish loop of some kind
that is unlikely to have a watchdog reset command in it.

Given that embedded processors are often stuck in the middle of
electrically noisy environments,  or placed inaccessibly this is a very
important feature.

Processors can crash for reasons other than poor code. Single event
upsets triggered by background and cosmic radiation are examples. As
well as glitches from electrical noise and RFI and EMI.

The AVR ATmega has one built in but most folk don't actually enable or
use it.

marvinroger, 2015-05-15:
Thanks, I thought it was something specific to the ESP8266. So I suppose the watchdog resets at the end of a loop() iteration and on a yield/delay call?
I think the concept should be documented in the README, because as you said it is not enabled by default on official Arduino boards, so newcomers might be lost as I was.

AndyKirby, 2015-05-15:
Agreed

The ESP8266 is a bit of an odd ball case.

Consider that the other Arduino/Ethernet/Wifi offerings use an ATmega as
the arduino and a second embedded processor of some type or other to do
the networking bit.

In the ESP's case we are expecting one processor to do it all. there is
a lot going off in the background that we are unaware of. The
pre-written envrionments typical of arduino style coding don’t help with
seeing what is going off behind the scenes. But does get folk into it
quicker by hiding the complexity. Something of a trade off.

Not only do we need to de-schedule to kick or reset the watchdog timer,
but we also need to do the same to let the background networking stuff
do it's thing.

This knowledge is obvious to the guys at the cutting edge doing the
coding but is part of a whole bunch of context that other folk coming to
it new do not have.

Part of the challenge of turning out open source is the wide range of
skill levels that will be encountered.

Some clear statements and maybe an example snippet might be useful to
help with this. Or at least a statement of what the time period for the
WDT defaults to, so folk can deal with it by issuing a specific call to
it at intervals.

Whilst it will be possible to modify the WDT period and even to disable
it all together, my best advice is don't do it. Learn to program with
WDT and de-scheduling if you are planning on working with the ESP.

igrr, 2015-05-15:
We've got something along these lines in the third paragraph here:
https://github.com/esp8266/Arduino#timing-and-delays

@AndyKirby if you can help to expand/improve this readme section I would really appreciate this.

AndyKirby, 2015-05-15:
Yup will do.

What is the default timeout for WDT ??

igrr, 2015-05-15:
Around 1 second. Unfortunately there is no info at this point how to change that value.

AndyKirby, 2015-05-15:
Cool leave it with me.

igrr, 2015-05-19:
WDT APIs (ESP.wdtEnable(), ESP.wdtDisable(), ESP.wdtFeed()) are now included in the latest release.

closed 2015-05-19 by igrr

adrionics, 2015-11-13:
ESP.wdtEnable(), ESP.wdtDisable() seem to have no effect?

Links2004, 2015-11-14:
the SDK allow only to disable the software WDT, so the hardware WDT is still aktive.
try to add some ```delay```.

adrionics, 2015-11-14:
Thanks, I was experimenting and trying to understand the behavior. What is the difference between the software WDT and hardware one? When would one be triggered or the other?

Links2004, 2015-11-14:
the software WDT is implemented inside the SDK from espressif, and is using a Timer.
the Hardware WDT is implemented in the silicon of the ESP chip, and is running independent form every software. 

the basic mechanic is for both the same if they not reseted  before they run out, the chip makes a reboot.

the different from developer view is that the SDK gives us the possibility to manipulate the software WDT but not the hardware WDT.
Software WDT api:
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Esp.h#L86-L92

Unfortunately we not have any clear documentation what the times for both are.
after around 1sec we get a WDT reset.

adrionics, 2015-11-14:
Thanks, that makes sense. It did not seem like the timeout changed anything :)

igrr, 2015-11-14:
We can manipulate hardware WDT. setting exact timeouts is not clear, but
enabling/disabling is not difficult. I didn't include those functions in
the core because I believe disabling WDT (which many will attempt) is a
lousy idea. I can provide more info on WDT registers though.

electronicsguy, 2016-04-25:
@igrr  @tomaja  @adrionics   I tried ESP.wdtDisable() and adding a delay. No effect. It still resets. did anyone get a long setup() working reliably?

peteroid, 2016-05-06:
ESP.wdtDisable() seems no effect. It will still reset the chip if delay takes long.

But ESP.wdtEnable(ms) works fine. It seems that the default time can be rewritten by this function. The following code would not reset the chip:
```c
ESP.wdtEnable(15000); // make the watch dog timeout longer
delay(10000);         // long delay
```

I am using NodeMCU v3 ESP-12E module with Arduino IDE 1.6.7.

Connor-F, 2016-08-18:
`wdtEnable()`, `wdtEnable(time)` and `wdtDisable() `don't seem to be working. Did anyone find a way around this?

Using NodeMCU v2 ESP-12E with Arduino IDE 1.6.7

jhows, 2016-08-29:
> Connor-F commented 11 days ago
> wdtEnable(), wdtEnable(time) and wdtDisable()don't seem to be working. Did anyone find a way around this?
> 
> Using NodeMCU v2 ESP-12E with Arduino IDE 1.6.7

Same problem here, with Wemos D1 and Wemos D1 Mini,

romansavrulin, 2016-09-20:
@igrr Hello, Ivan. can you provide any info about HW WDT registers? we implement a low-power application and need to sleep for a long period of time. It seems that HW wdt resets the CPU, so we have to wakeup and feed the WDT to workaround this.

derpicknicker1, 2016-10-25:
Hey guys...im also looking for some information about HW WDT.

vapor99, 2016-12-10:
Same here.  I am using the RFM69 library and waiting for RX sets off the WDT, its very difficult to debug.

hemangjoshi37a, 2017-06-06:
anyone have found any solution to this problem??
wdtEnable() or wdtDisable() does not work.
I am  facing this from a week or so. Please help me.
Thank you.

vwroad87, 2017-06-08:
I facing the same problem using serial.available() in **the arduino loop** while running an async webserver build.   Constant serial input more that 1 second is causing an WDT timeout.

hemangjoshi37a, 2017-06-09:
@vwroad87 So have you found any solution??

vwroad87, 2017-06-09:
No I just posted today,  hopefully someone can comment on the WDT issue.

hemangjoshi37a, 2017-06-09:
@vwroad87 okay man, but this bug  is tough one. Because this issue was opened 2.5 years ago but its still not solved. I also hope someone someday come with solution.

vwroad87, 2017-06-09:
Well I'm gonna try a couple more things,  delay(0) for one to see if this
resets the WDT.

I'll post and try to keep the thread alive and perhaps a core dev will take
interest.

freeck, 2017-06-13:
Did you try ESP.wdtFeed() ?

hemangjoshi37a, 2017-06-13:
@freeck I will try that and let you know if it works

vwroad87, 2017-06-13:
Here is my Loop, everything works except when the attached micro controller
sends back a text file to display which could be many lines. It fails at
random lines though, sometimes the first few others after maybe 15-20
lines?

void loop() {
  bool      led_state ;
  uint16_t fade_speed;

  // are we in web server call back mode?
    if (webCallbackMode){
      // NOP
    }else{
      // we are in websocket mode
      // Got one serial char ?
      //if (PROP_SERIAL.available()) {
      while (PROP_SERIAL.available()) {
        ESP.wdtFeed();     //reset watchdog
        // Read it and store it in buffer
        char inChar = (char)PROP_SERIAL.read();
        inputString += inChar;

        // New line char ?
        if (inChar == '\n') {
          // Send to all client
          ws.textAll(inputString);
          inputString = "";
          break;      // get out of the loop
        }

      }
    }

  // Handle remote Wifi Updates
  ArduinoOTA.handle();

  //get the current miliseconds
  unsigned long currentMillis = millis();
  // handle the PSI reading here
  //if (currentMillis - previousMillis >= interval) {
    // save the last time you read the PSI
    //previousMillis = currentMillis;
    // read the PSI sensor
    //PROP_SERIAL.println(getPSIMA());
    //getPSIMA();  // update the psi sensor
  //}

}

adrionics, 2017-06-13:
Did you try using yield() instead of wdtFeed()?

hemangjoshi37a, 2017-06-14:
@adrionics Thanks man. I will try that and let you know if it works fine.

vwroad87, 2017-06-14:
Not yet, I will and let the list know,  I've read using yeild() is more art
when working with asynchronous systems. The other part of this project has
already started an async webserver on top of the async IP stack.

Yield() didn't work but I did fix the problem by putting a 1 ms delay
between each line in  the attached micro routine that was sending the lines.

I did keep yield() in the receive routine since a to "pump the queue"
during the receive routing delay from the attached micro.

hemangjoshi37a, 2017-06-14:
@vwroad87 , hey man can you put the code here?? The one that succeeded .

vwroad87, 2017-06-14:
Joshi,   like I said the micro (propeller chip) that is  attached to the
ESP8266 was flooding the ESP RX buffer or something I guess.  What I did on
the micro was to insert a 1 ms delay into each line that was being sent to
the ESP8266 serial port.   The code on the micro is FORTH and I don't think
you wanna see that right?    So my bug fix was to slow down the data being
sent to the esp8266 and I did keep yield in my receive loop like this:

void loop() {
  bool      led_state ;
  uint16_t fade_speed;

  // are we in web server call back mode?
    if (webCallbackMode){
      // NOP
    }else{
      // we are in websocket mode
      // Got one serial char ?
      if (PROP_SERIAL.available()) {
        // Read it and store it in buffer
        char inChar = (char)PROP_SERIAL.read();
        inputString += inChar;
        yield();     //let other stuff run

        // New line char ?
        if (inChar == '\n') {                       // each line ending in
\n  from the propeller chip has a 1 ms delay now coded in the propeller
micro routine
          // Send to all client
          ws.textAll(inputString);
          inputString = "";
        }

      }
    }

  // Handle remote Wifi Updates
  ArduinoOTA.handle();

}

lrmoreno007, 2017-06-21:
I've never had this problem but today I had to fight with it, because the sketch I'm creating uses the Ticker.h library (It have timers inside and i get continous wdt resets).

I have solved it by creating the following function:
```
void delaySeconds(int seconds) { 
	for (int i = 0; i <= seconds*10; i++){
		delay(100);
		ESP.wdtFeed();
	}
}
```
And calling her with:
```
delaySeconds(2);  // two seconds 
```
instead of using:

```
delay(2000);  // two seconds 
```

I hope this helps someone.

hemangjoshi37a, 2017-06-22:
@lrmoreno007 yo man, you are going to live for 1000 years, you have my blessings, you saved my day....

vwroad87, 2017-06-22:
What else are you doing on the ESP8266?
My problem was/is that I'm using the Async TCP/HTTP stack/server and
receiving RX serial data in the  sketch loop was causing WDT resets.   I
fixed the problem by having the attached serial sender pause 2 ms between
each line it sent.   I tried yield(), ESP.wdtFeed() in the RX loop, slowing
down the baud rate.....but only after the 2 ms delay in the sender did the
system go stable again.

subham9777, 2017-09-23:
I have this problem using a dht11 sensor and wemos d1 mini to connect to adafruit io server , The code is 

#include <ESP8266WiFi.h>
#include "Adafruit_MQTT.h"
#include "Adafruit_MQTT_Client.h"
#include "DHT.h"

// DHT 11 sensor
#define DHTPIN 8
#define DHTTYPE DHT11 

// WiFi parameters
#define WLAN_SSID       "ssid"
#define WLAN_PASS       "pass"

// Adafruit IO
#define AIO_SERVER      "io.adafruit.com"
#define AIO_SERVERPORT  1883
#define AIO_USERNAME    "xcvnxcb"
#define AIO_KEY         "xyzredgbd"

// DHT sensor
DHT dht(DHTPIN, DHTTYPE, 15);

// Functions
void connect();

// Create an ESP8266 WiFiClient class to connect to the MQTT server.
WiFiClient client;
// Store the MQTT server, client ID, username, and password in flash memory.
Adafruit_MQTT_Client mqtt(&client, AIO_SERVER, AIO_SERVERPORT, AIO_USERNAME, AIO_KEY);
// Setup feeds for temperature & humidity
Adafruit_MQTT_Publish temperature = Adafruit_MQTT_Publish(&mqtt, AIO_USERNAME "/feeds/temperature");

Adafruit_MQTT_Publish humidity = Adafruit_MQTT_Publish(&mqtt, AIO_USERNAME "/feeds/humidity");

/*************************** Sketch Code ************************************/

void setup() {

  // Init sensor
  dht.begin();

  Serial.begin(115200);
  Serial.println(F("Adafruit IO Example"));

  // Connect to WiFi access point.
  Serial.println(); Serial.println();
  delay(10);
  Serial.print(F("Connecting to "));
  Serial.println(WLAN_SSID);

  WiFi.begin(WLAN_SSID, WLAN_PASS);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(F("."));
  }
  Serial.println();

  Serial.println(F("WiFi connected"));
  Serial.println(F("IP address: "));
  Serial.println(WiFi.localIP());

  // connect to adafruit io
  connect();

}

void loop() {

//  // ping adafruit io a few times to make sure we remain connected
//  if(! mqtt.ping(3)) {
//    // reconnect to adafruit io
//    if(! mqtt.connected())
//      connect();
//  }

  // Grab the current state of the sensor
  int humidity_data = (int)dht.readHumidity();
  int temperature_data = (int)dht.readTemperature();

  // Publish data
  if (! temperature.publish(temperature_data))
    Serial.println(F("Failed to publish temperature"));
  else
    Serial.println(F("Temperature published!"));

  if (! humidity.publish(humidity_data))
    Serial.println(F("Failed to publish humidity"));
  else
    Serial.println(F("Humidity published!"));

  // Repeat every 10 seconds
  delay(10000);

}

// connect to adafruit io via MQTT
void connect() {

  int8_t ret;

  // Stop if already connected.
  if (mqtt.connected()) {
    return;
  }

  Serial.print("Connecting to MQTT... ");

  uint8_t retries = 3;
  while ((ret = mqtt.connect()) != 0) { // connect will return 0 for connected
       Serial.println(mqtt.connectErrorString(ret));
       Serial.println("Retrying MQTT connection in 5 seconds...");
       mqtt.disconnect();
       delay(5000);  // wait 5 seconds
       retries--;
       if (retries == 0) {
         // basically die and wait for WDT to reset me
         while (1);
       }
  }
  Serial.println("MQTT Connected!");
}

It continuously show the following thing : -
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3ffe85ec
~ld

devyte, 2017-09-23:
@subham9777 you can't use such huge delays on the ESP, the wdt will trigger. This is on purpose, because the wifi stack has to be serviced frequently.
If you need more info, please ask at a forum like esp8266.com or stackoverflow.

martinius96, 2017-11-01:
#ISSUE
@igrr
Hello there, i tried these ESP. functions for watchdog but i dont know how it works.. I put in setup that:
`ESP.wdtEnable(WDTO_2S);`
At the end of loop i take that: 
` ESP.wdtFeed(); `
My loop working time is circa 6 seconds for one loop. I enabled wdt for 2 seconds and it didnt reset my NodeMCU v3 Lolin. I only want to reset my board to see if it is working if code is running more than 2 seconds.. I want to put in to 8S, but i dont know what i am doing bad. Where is problem? Maybe i bad think in that. I want if my board lags (stops working) it will reset. I am using HTTPS request to webpage (IoT house) for upload and download values and full automatization. Thanks for help!

Baxtex, 2018-01-08:
Got this issue aswell on my Wemos D1.  ESP.wdtDisable(); doesn't seem to do anything.

romansavrulin, 2018-01-08:
@Baxtex, that's [expected](https://github.com/esp8266/Arduino/issues/34#issuecomment-88799636)

electronicsguy, 2018-01-09:
@devyte  is there a document which describes why the wifi stack needs frequent servicing? (and what exactly it does as part of the servicing?)

adrionics, 2018-01-09:
Hello Sujay,

Please clarify what you mean by frequent servicing.  If you provide more
detail we may be able to help answer this.

Kind Regards,
Amanda

electronicsguy, 2018-01-09:
@adrionics  I'm referring to @devyte 's comment on 24th Sept above.

[12 more replies]