esp8266/Arduino #35: Issue with Olimex ESP8266 module

marcoschwartz, 2015-04-02:
Hello, great work on this project! I tested it and it works fine with the simple 8-pin ESP8266 module (the $5 one).

However, I've been trying with this module:

https://www.olimex.com/Products/IoT/MOD-WIFI-ESP8266-DEV/

And it doesn't work. I can upload the sketch but after that it fails and only return strange characters in the Serial monitor. I guess this is either because the flash memory is different from the other module, and because the clock is also different. Do you think that might be the cause? If yes, do you have plans to support different ESP8266 flash sizes/clocks in the future? Thanks

igrr, 2015-04-02:
Probably this is due to a different clock frequency.
I don't have any modules other than the basic esp-01 and esp-12 ones, so can not test this myself. I suppose we need to write some value into flash to set the correct chip size and clock. On the IDE side, this can be easily integrated by adding another board variant. Just need to add an option for esptool-ck to generate binaries for different flash clock/size.
Maybe this could help?
https://github.com/espressif/esp_iot_rtos_sdk/blob/master/tools/gen_appbin.py#L166-L192

marcoschwartz, 2015-04-02:
Yes, I think it would be great to integrate the choice of clock & memory size into the IDE. I am doing that with the Python esptool I am using the program the other board with NodeMCU. 

I am asking because I am currently writing a book around the ESP8266 and it would be great for my readers to use the Arduino IDE instead of NodeMCU which is more complicated. How long do you think it would take you to integrate this functionality? As soon as it's ready I could test it. Thanks!

dangamba, 2015-04-02:
Hello, just tried some basic sketches on this one:
https://www.olimex.com/Products/IoT/ESP8266-EVB/open-source-hardware
which is an eval board for MOD-WIFI-ESP8266-DEV, and everything works like a charm.

marcoschwartz, 2015-04-03:
I don't know what changed but now it works with one of my Olimex module too :) However the other one still doesn't work, but works with NodeMCU. Anyway ... I saw you added an option for the crystal speed, I'll let you decide to close this issue or not. Would be good to have also flash size options (512k, 1M, 2M ...) and more CPU speed (20 MHz, 26 MHz, 40 MHz, ...). Thanks!

igrr, 2015-04-03:
@marcoschwartz This option that was added is the "software-controlled" kind of CPU clock selection. The crystal frequency is still fixed at the default value.
Esptool-ck needs some changes to generate the firmware for a given crystal frequency and flash size. I will look at it! It would be nice if this toolkit worked with a wide range of boards.

igrr, 2015-04-30:
@marcoschwartz  I have added support for flash size/mode/frequency configuration to [esptool-ck](https://github.com/igrr/esptool-ck). The updated boards.txt file for the IDE with flash size menu is coming up. I think if the board needs CPU speed and CPU mode options then it would be better to add it as an additional item to the boards menu (e.g. "Olimex MOD-WIFI"). Unfortunately I don't own that module myself so don't know what the correct options are.

marcoschwartz, 2015-05-01:
Amazing, thanks! I'll have a look at the boards.txt when it's available.

igrr, 2015-05-01:
boards.txt has been commited as well. You will need esptool updated to 0.4.3. Hope this will solve some of the upload issues as well.

duncan-a, 2015-05-01:
*Marco*

I know you have a few varieties if the ESP module, might it be worth
compiling a list of known crystal frequencies and Flash chip capacities and
adding it to the WiKi? Pinouts and dimensions would also be very helpful to
me, what do you think?

My eyes aren't good enough to read the crystals and chip numbers, even with
my strongest magnifier - if they were, I'd start the ball rolling. All I
know at the moment is that the Flash chio on the Olimex board looks
physically larger than that fitted to the 01 boards I have and, according
to the Olimex schematic it's a W25Q16BVSSIG, which is 16MB.

Is there not a way that the IDE can check the Flash size during compile and
create an error message if it's inadequate? It seems a bit bizarre to have
manually selectable options when there seem to be so many variations?

It would be nice if the CPU frequency could also be autodetected, but I
suspect that's not possible - I can't read the crystals because I have poor
eyesight and, when it's all inside a 'tin can', nobody can read the crystal
or the Flash chip...

igrr, 2015-05-19:
Olimex MOD-WIFI is supported in the latest release.

closed 2015-05-19 by igrr

duncan-a, 2015-05-19:
I've just installed 1.6.4 and ESP via the Boards Manager and noticed that
the Olimex option doesn't show the size of the Flash, unlike the Generic.

Might I suggest that the Olimex board should have the Flash size options,
with a default of 16M (Olimex may downsize at some point in the future -
who knows).

Thanks for all the hard work that everybody has put into this project...