esp8266/Arduino #37: Unsupported baud rate 115200

scargill, 2015-04-02:
The examples use 115200 baud ratge - yet the compiler complains that it's an invalid baud rate - it still works of course. Can this be removed?

igrr, 2015-04-02:
Could you please post the sketch itself and the verbose compiler output when you submit issues? Verbose output can be enabled in the Arduino preferences. Thanks.

```
void setup() {
  Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(1000);
  Serial.println("Hello ESP!");

}
```
This sketch with 115200 seems to work.

scargill, 2015-04-02:
Absolutely it compiles no problem and runs no problem but when flashing the chip this is the message I see.. unsupported baud rate – it still works…

 

 

 

Sketch uses 189,636 bytes (36%) of program storage space. Maximum is 524,288 bytes.

warning: unsupported baud rate: 115200, using 115200

Uploading 31488 bytes from C:\Users\Peter\AppData\Local\Temp\build71807972327932918.tmp/sketch_apr02b.cpp_00000.bin to flash at 0x00000000

...............................

Uploading 158184 bytes from C:\Users\Peter\AppData\Local\Temp\build71807972327932918.tmp/sketch_apr02b.cpp_40000.bin to flash at 0x00040000

...........................................................................................................................................................

igrr, 2015-04-02:
Hmm. How is that possible?
```
void serialport_setbaudrate(unsigned int baudrate)
{
	DWORD br = 0;
	switch(baudrate)
	{
		case 9600: 		br = CBR_9600; 		break;
		case 14400: 	br = CBR_14400; 	break;
		case 19200: 	br = CBR_19200; 	break;
		case 38400: 	br = CBR_38400; 	break;
		case 56000: 	br = CBR_56000; 	break;
		case 57600: 	br = CBR_57600; 	break;
		case 115200: 	br = CBR_115200; 	break;
		case 128000: 	br = CBR_128000; 	break;
		case 256000:	br = CBR_256000;	break;
	}
	if (br == 0)
	{
		LOGWARN("unsupported baud rate: %d, using 115200", baudrate);
		br = CBR_115200;
	}
```
Puzzled...

cimba007, 2015-04-02:
Not sure if this is related but the LOGWARN message looks the same:

Setting the upload speed to 921600 does not work, although 256000  is fine

Der Sketch verwendet 184.858 Bytes (35%) des Programmspeicherplatzes. Das Maximum sind 524.288 Bytes.
D:\Users\Jinja_Admin\Downloads\arduino-1.6.1-p1-windows\arduino-1.6.1/hardware/tools/esp8266/esptool.exe -vv -cd ck -cb 256000 -cp COM23 -ca 0x00000 -cf C:\User
s\Jinja\AppData\Local\Temp\build2025833383679790865.tmp/sketch_apr02a.cpp_00000.bin -ca 0x40000 -cf C:\Users\Jinja\AppData\Local\Temp\build2025833383679790865.t
mp/sketch_apr02a.cpp_40000.bin
esptool v0.4.2 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
        setting board to ck
        setting baudrate from 115200 to 256000
        setting port from COM1 to COM23
        setting address from 0x00000000 to 0x00000000
        espcomm_upload_file
        stat C:\Users\Jinja\AppData\Local\Temp\build2025833383679790865.tmp/sketch_apr02a.cpp_00000.bin success

Der Sketch verwendet 184.858 Bytes (35%) des Programmspeicherplatzes. Das Maximum sind 524.288 Bytes.
D:\Users\Jinja_Admin\Downloads\arduino-1.6.1-p1-windows\arduino-1.6.1/hardware/tools/esp8266/esptool.exe -vv -cd ck -cb 921600 -cp COM23 -ca 0x00000 -cf C:\User
s\Jinja\AppData\Local\Temp\build2025833383679790865.tmp/sketch_apr02a.cpp_00000.bin -ca 0x40000 -cf C:\Users\Jinja\AppData\Local\Temp\build2025833383679790865.t
mp/sketch_apr02a.cpp_40000.bin
esptool v0.4.2 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
        setting board to ck
        setting baudrate from 115200 to 921600
        setting port from COM1 to COM23
        setting address from 0x00000000 to 0x00000000
        espcomm_upload_file
        stat C:\Users\Jinja\AppData\Local\Temp\build2025833383679790865.tmp/sketch_apr02a.cpp_00000.bin success
warning: unsupported baud rate: 921600, using 115200
        setting serial port timeouts to 1000 ms

igrr, 2015-04-02:
@cimba007 as you see from the code snippet above, 921600 is not supported on Windows (because windows doesn't have ```CBR_921600``` defined), so the output you have posted makes sense.
I think 921600 only works on Linux currently.

What I don't understand is how we can possibly get a message
> warning: unsupported baud rate: 115200, using 115200

from the code snippet above...

cimba007, 2015-04-02:
Is there a way to increase the baud rate above CBR_256000 with windows? 256000 is already a huge speed improvement but more = better ;-)

It seems odd, since I can do the following:

Serial.begin(512000);

- Open the com-port with e.g. http://www.der-hammer.info/terminal/
- Add Custom baud-rate to config file: 512000

And now I can receive the messages perfectly fine.

igrr, 2015-04-02:
Well, just found this:
```

/* These baudrates work on windows, but are not part of the windows headers */
#ifndef CBR_230400
#define CBR_230400 230400
#endif
#ifndef CBR_460800
#define CBR_460800 460800
#endif
#ifndef CBR_921600
#define CBR_921600 921600
#endif
```

[source](http://purple.mxnode.net/git/repostitory/SnoreCopter/qgroundcontrol.git/raw/587238c201ada3b90ca5bf7adb4bf974a500b591/src/lib/qextserialport/win_qextserialport.h)

So maybe it's okay to define those on Windows as well.

4refr0nt, 2015-04-02:
Ivan!
ESPlorer (written on Java too) has support up to 921600 on any
platform. And on Windows too. I'm using jssc java library for
low-level communication (written on C). But for Arduino IDE may be
other way for supporting high speed.

igrr, 2015-04-03:
I've got a new esptool build ready which adds support for higher baud rates on Windows:
https://ci.appveyor.com/api/buildjobs/i3ih98t8cyby87j6/artifacts/esptool-0.4.2-3-g1e36d49-win32.zip
Could you guys please give it a try? You can unzip this esptool.exe into **<Arduino dir>/hardware/tools/esp8266/esptool.exe**
Thanks.

cimba007, 2015-04-03:
Perfect! I tryed some different baud-rates and here are the results for my 200kb sketch:

115200 ~21 seconds
256000 ~10.5 seconds
921600 ~ less then 4 seconds

Only rough estimations with stopwatch.

No error messages, works fine

esptool v0.4.2-3-g1e36d49 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 921600
	setting port from COM1 to COM23
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat C:\Users\Jinja\AppData\Local\Temp\build3547006662176080286.tmp/esp8266_serial_test.cpp_00000.bin success
	setting serial port timeouts to 1000 ms
opening bootloader

scargill, 2015-04-03:
Erm, I've just realised I have no idea how to set the baud rate in this Arduino environment for ESP8266.  Someone enlighten me?

cimba007, 2015-04-03:
Go to the following path:

arduino-1.6.1-p1-windows\arduino-1.6.1\hardware\esp8266com\esp8266

and update the **boards.txt**

with the current one from the repository:

https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/boards.txt

Now you can use Tool > UploadSpeed in the Arduino IDE

scargill, 2015-04-03:
That looked impressive until I compiled and..
C:\Users\Peter\Documents\Arduino\hardware\arduino-esp8266-master\esp8266\cores\esp8266/Arduino.h:181:26: fatal error: pins_arduino.h: No such file or directory

 #include "pins_arduino.h"
                          ^

cimba007, 2015-04-03:
What Version are you using? I took the ~350mb download from here

+ https://github.com/igrr/Arduino/releases/download/1.6.1-esp8266-1/arduino-1.6.1-p1-windows.zip

and started the arduino_debug.exe with administrator priviledges (otherwise the compiler will crash).

The path suggests you are using the esp8266 library with the Arduino unmodified standalone?

scargill, 2015-04-03:
I'm using Arduino 1.62 - and all was well.... until.... I tried this mod (wish I'd kept the original) - I'm assuming all I need is a special pins_arduino.h ???

igrr, 2015-04-03:
Taking the most recent revision from git was likely the problem — I renamed board variants a few hours ago.
Instead, take this older version of boards.txt:
https://github.com/esp8266/Arduino/blob/2c7a0245b0c5b79d89423c4b890666e57f5d9339/hardware/esp8266com/esp8266/boards.txt

scargill, 2015-04-03:
Yes that’s it thanks… as it happens my (fast) PC won’t do beyond 115kb, at the top speed nothing happens, at 230k it fails, 115k works a treat.

 

:)

 

https://github.com/esp8266/Arduino/blob/2c7a0245b0c5b79d89423c4b890666e57f5d9339/hardware/esp8266com/esp8266/boards.txt

cimba007, 2015-04-04:
hmm, after i was just happy, that the upload baud-rate increased i now can now longer upload new sketches. i do not get any kind of error but after flashing there is just the old content on the esp.

doing some investigation

nerdralph, 2015-04-04:
I tried it out, and 921,600 works with a pl2303hx on Win7E/64.  It seems even more reliable than when I tried 256,000.
I still sometimes get sync failed, but I was getting that at 115,200 as well.
warning: espcomm_sync failed
error: espcomm_open failed

Both the PL2303 and FT232 support 3mbps.  I've even tested the pl2303 at 6mbps:
http://nerdralph.blogspot.ca/2014/05/high-speed-uart-pumping-out-data-at.html

So adding 3mbps support to esptool would probably work.

nerdralph, 2015-04-06:
I tried adding 3000000 to serialport_setbaudrate, and it didn't work.  It did set the port to 3mbps, but gave a sync failed error every time I tried.  I'm guessing the esp8266 bootloader won't autobaud above 921,600bps.

circuitFlower, 2015-04-11:
any updates on this? I'm getting 
warning: espcomm_sync failed
error: espcomm_open failed
when I try to upload example files to the esp8266 on Yosemite.

igrr, 2015-04-11:
@circuitFlower 
Are you able to upload any other firmware using esptool.py?
Usually "espcomm_sync failed" means the chip was not put into bootloader mode prior to upload. Did you try pulling GPIO0 low and toggling power? Also may need to pull GPIO2 high, GPIO15 low, CH_PD high.

circuitFlower, 2015-04-11:
Hi, 
I ran an esptool command and received this error

mac-ae:esptool-master ae$ sudo ./esptool.py read_mac
Traceback (most recent call last):
  File "./esptool.py", line 470, in <module>
    esp = ESPROM(args.port, args.baud)
  File "./esptool.py", line 66, in __init__
    self._port = serial.Serial(port, baud)
  File "build/bdist.macosx-10.10-intel/egg/serial/serialutil.py", line 282, in __init__
  File "build/bdist.macosx-10.10-intel/egg/serial/serialposix.py", line 289, in open
OSError: [Errno 2] No such file or directory: '/dev/ttyUSB0’

I noticed that there’s a default usb set in the python file, should I change that?
aside from that, I’m a little confused. does the arduino-esp8266 IDE automatically use the esptool, or do I need to put it into bootloader mode manually before I run the script in the IDE? I don’t see a command specifically for that? 
Also, it’s probably important to note that I’m using the ESP8266 WI07c Wifi Module purchased from Electrodragon roughly January 24th. It doesn’t have a GPIO15? but CH_PD is high, GPIO0 IS LOW, GPIO2 IS HIGH (TRIED BOTH), GND is ground, VCC is +, I don’t have reset connected.

I did not have GPIO0 LOW, so I connected that, and now I’m getting this error (through the arduino-esp8266 ide).:
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
closing bootloader

igrr, 2015-04-13:
If you use esptool.py, then the port should be specified, as in ```esptool.py --port=/dev/tty.usbserial read_mac```.
The IDE uses esptool ([another one](https://github.com/igrr/esptool-ck)) and it can reset the board for you, but you need to connect DTR and RTS lines to control GPIO0 and RESET pins of ESP.

If you get "FLASH_DOWNLOAD_BEGIN failed" error, try uploading once more (I get this error sometimes, but on the second try it usually works). I also have seen a case when an FTDI-based USB-serial converter did not work well with FTDI driver provided by Apple. Installing drivers from FTDI helped resolve the issue.

igrr, 2015-05-05:
High baud rates support is included into the latest esptool (which also resolves some of the upload issues).

closed 2015-05-05 by igrr

santhosh8438, 2017-07-13:
Arduino: 1.8.3 (Windows 7), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 921600, 4M (3M SPIFFS)"

I'm getting this ERROR when i use 921600 baudrate.... but its works on another Same module....

Build options changed, rebuilding all
Archiving built core (caching) in: C:\Users\Sandy\AppData\Local\Temp\arduino_cache_882395\core\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,UploadSpeed_921600,FlashSize_4M3M_7e4b3b36485033ea551fda2776e87af1.a
Sketch uses 224201 bytes (21%) of program storage space. Maximum is 1044464 bytes.
Global variables use 31736 bytes (38%) of dynamic memory, leaving 50184 bytes for local variables. Maximum is 81920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

scargill, 2017-07-13:
Why on EARTH would anyone want to try serial comms at that speed... surely that is just asking for trouble?

santhosh8438, 2017-07-13:
That's the maximum speed henceforth i don't want to waste my time by using other br, I have 3 modules with same specification model all models supports that br, but only one is not compatible!!!

scargill, 2017-07-13:
Whatever works for you - but clearly - it doesn't always. I personally do all my stuff at 230400 and never have any issues...

devyte, 2017-07-13:
@scargill <--- same as him, I use 230Kbps. Going above that is asking for trouble.

Whether higher speeds work or not depends on a bunch of things, including manufacturer, wear, noise, power supply, etc. In some cases, what gets written to the flash could even become corrupted on reboot.

santhosh8438, 2017-07-14:
Yeah, You may b right, i didn't considered the physical things...