esp8266/Arduino #38: PubSub missing

scargill, 2015-04-02:
PubSub appears to be missing from the libraries and the original pubsub won't compile.

igrr, 2015-04-02:
The [original one](https://github.com/knolleary/pubsubclient) was reported to work fine.
Please try [this sample](https://gist.github.com/igrr/7f7e7973366fc01d6393) for a start.

closed 2015-04-03 by igrr

scargill, 2015-04-03:
I thought I had it there - compiles ok, publishes ok, subscribes ok but will not take any notice of subscribed topics - and if you keep firing subscribed topics at it, the program eventually dies.

Here's my modified test - I even used millis() in case the 5 second delay was causing problems - made no difference...

#include <PubSubClient.h>
#include <ESP8266WiFi.h>
 
void* __dso_handle;
 
const char* ssid = "loft-east";
const char* password = "mypass";
 
char* topic = "blah";
char* server = "192.168.0.15";
 
 
unsigned long mymillis;
 
WiFiClient wifiClient;
PubSubClient client(server, 1883, callback, wifiClient);
 
void callback(char* topic, byte* payload, unsigned int length) {
  // handle message arrived
  Serial.print("Got message "); Serial.print(topic); Serial.print(" length "); Serial.println(length);
}
 
 
String macToStr(const uint8_t* mac)
{
  String result;
  for (int i = 0; i < 6; ++i) {
    result += String(mac[i], 16);
    if (i < 5)
      result += ':';
  }
  return result;
}
 
void setup() {
  Serial.begin(115200);
  delay(10);
  
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
 
  // Generate client name based on MAC address and last 8 bits of microsecond counter
  String clientName;
  clientName += "esp8266-";
  uint8_t mac[6];
  WiFi.macAddress(mac);
  clientName += macToStr(mac);
  clientName += "-";
  clientName += String(micros() & 0xff, 16);
 
  Serial.print("Connecting to ");
  Serial.print(server);
  Serial.print(" as ");
  Serial.println(clientName);
  
  if (client.connect((char*) clientName.c_str()),"admin","mypass") {
    Serial.println("Connected to MQTT broker");
    Serial.print("Topic is: ");
    Serial.println(topic);
    
    if (client.publish(topic, "hello from ESP8266")) {
      Serial.println("Publish ok");
      if (client.subscribe("hmm")) Serial.println("Subscribed ok"); else ("Subscribed failed");
    }
    else {
      Serial.println("Publish failed");
    }
  }
  else {
    Serial.println("MQTT connect failed");
    Serial.println("Will reset and try again...");
    abort();
  }
  mymillis=millis()+5000;
}

 
 
void loop() {
  static int counter = 0;

 if (mymillis<millis())
{
 mymillis=millis()+5000; 
  String payload = "{\"micros\":";
  payload += micros();
  payload += ",\"counter\":";
  payload += counter;
  payload += "}";
  
  if (client.connected()){
    Serial.print("Sending payload: ");
    Serial.println(payload);
    
    if (client.publish(topic, (char*) payload.c_str())) {
      Serial.println("Publish ok");
    }
    else {
      Serial.println("Publish failed");
    }
  }
  ++counter;
}
}
 
Any ideas?

igrr, 2015-04-03:
oh, you need to call client.loop(); from your loop function for the
callback to fire.
will update the gist.

scargill, 2015-04-03:
Right – that accounts for a lot – thanks for the speedy response – that appears to work a treat.  Presumably my use of millis() is the right solution as a 5 second delay would similarly stop incoming responses. Shame the incoming subs can’t be done under interrupt…

igrr, 2015-04-03:
There's a standard Arduino library, called [Time](https://github.com/PaulStoffregen/Time), which can make stuff like "call this from a loop every N seconds" easier.

scargill, 2015-04-03:
You are a life-saver – indeed that is of course correct and that now works..

 

But I’ve now found another issue…. While I was experimenting I realised my MQTT broker didn’t have security (something to do with a second mosquito.conf and an include that didn’t work)…  and in the process of getting THAT working – which it does perfectly I’ve realised that….

 

  if (client.connect((char*) clientName.c_str()),"my_username","my_password") {

 

Doesn’t work.  It does not matter what username and password I put in there it SAYs it has connected – but now will not publish (and rightly so)…

 

Any ideas?

igrr, 2015-04-03:
Frankly I've never tried using PubSubClient with auth... Need to
investigate what's happening inside the library.

scargill, 2015-04-03:
Doooohhhh. Parentheses…

 

Wrong..

 

if (client.connect((char*) clientName.c_str()),"my_username","my_password") {

 

Right..

 

if (client.connect((char*) clientName.c_str(),"my_username","my_password")) {

 

It works!

commit 2c7a0245 (2015-04-03): Update readme

Be more clear about PubSubClient
Close #38
[ci skip]