esp8266/Arduino #39: _DSO_HANDLE isn't defined

scargill, 2015-04-02:
When I tried to use the original PubSub I got a message to say _dso_handle isn't defined.

This time I've tried a simple web server page - using nothing special and the compilation has come back with the same error..

 C:\Users\Peter\AppData\Local\Temp\build71807972327932918.tmp/WiFiWebServer.cpp.elf: hidden symbol `__dso_handle' isn't defined

I used the same startup code as the server example that comes with this package - but this main loop

void loop(){
  // Create a client connection
  WiFiClient client = server.available();
  if (client) {
    while (client.connected()) {
      if (client.available()) {
        char c = client.read();

        //read char by char HTTP request
        if (readString.length() < 100) {

          //store characters to string 
          readString += c; 
          //Serial.print(c);
        } 

        //if HTTP request has ended
        if (c == '\n') {

          ///////////////
          Serial.println(readString); //print to serial monitor for debuging 

          client.println("HTTP/1.1 200 OK"); //send new page
          client.println("Content-Type: text/html");
          client.println();

          client.println("<HTML>");
          client.println("<HEAD>");
          client.println("<TITLE>Arduino GET test page</TITLE>");
          client.println("</HEAD>");
          client.println("<BODY>");

          client.println("<H1>Zoomkat's simple Arduino button</H1>");
          
          // DIY buttons
          client.println("<a href=\"/?on\"\">ON</a>"); 
          client.println("<a href=\"/?off\"\">OFF</a><br />"); 

          // mousedown buttons
          client.println("<br><input type=\"button\" value=\"ON\" onmousedown=\"location.href ('/?on');\"/>"); 
          client.println("<input type=\"button\" value=\"OFF\" onmousedown=\"location.href ('/?off');\"/>");        
          
          // mousedown radio buttons
          client.println("<br><br><input type=\"radio\" value=\"ON\" onmousedown=\"location.href ('/?on');\"\">ON</>"); 
          client.println("<input type=\"radio\" value=\"OFF\" onmousedown=\"location.href ('/?off');\"\">OFF</>");        

          client.println("</BODY>");
          client.println("</HTML>");

          delay(1);
          //stopping client
          client.stop();

          ///////////////////// control arduino pin
          if(readString.indexOf("on") >0)//checks for on
          {

            digitalWrite(0, HIGH);    // set pin 5 high
            Serial.println("Led On");
          }
          if(readString.indexOf("off") >0)//checks for off
          {
            digitalWrite(0, LOW);    // set pin 5 low
            Serial.println("Led Off");
          }
          //clearing string for next read
          readString="";

        }
      }
    }
  }
}

igrr, 2015-04-02:
This is a [known issue](http://www.esp8266.com/viewtopic.php?f=29&t=2196&p=13039&hilit=__dso_handle#p13039) with Windows toolchain (should be fixed in the next release).
As a workaround, declare the following global variable in your sketch:
```
void* __dso_handle;
```

scargill, 2015-04-02:
Aw thanks – that’s really great.

closed 2015-05-05 by igrr