esp8266/Arduino #41: itoa() function not found

marcoschwartz, 2015-04-03:
I have a sketch that uses the itoa() function, and I got the error when compiling:

'itoa' was not declared in this scope

Is this function already included in the project? Thanks!

igrr, 2015-04-03:
It was included yesterday:
https://github.com/Links2004/Arduino/blob/esp8266/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_noniso.c#L102

marcoschwartz, 2015-04-03:
Cool! But that's not yet merged into the main repository right?

igrr, 2015-04-03:
It's not in the release yet, yes.
But you can download the release, clone this git repository, and then replace the contents of **hardware/esp8266com** directory with the contents of the same directory from git.

marcoschwartz, 2015-04-03:
Got it, thanks!

closed 2015-04-03 by marcoschwartz

igrr, 2015-04-03:
And yeah, it's merged actually:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_noniso.c#L110

marcoschwartz, 2015-04-03:
Yeah, works! I just needed to include:

#include "stdlib_noniso.h"