esp8266/Arduino #43: SSL support

anteph, 2015-04-03:
Hi! 
I would like to know if you plan to include SSL support in the libraries. 
I've tryed to send some https requests with no sucess.

It would be cool if it the ESP could act as a secure server too.

I've checked the some examples from the sdk and they have a code to create a secure server with digital certificate. 

Thanks!

abl, 2015-04-22:
SSL has been ~broken since 0.9.3-patch1 (and you needed a new libssl, separate from patch1, from Espressif.)

It's supposedly fixed again in 1.0.1b2 (you still need a special libssl for the moment) - http://bbs.espressif.com/viewtopic.php?f=5&t=382 - I haven't had a chance to try it yet, but I've heard good reports.

There are still some features missing for practical SSL - iirc there's no way to pin a certificate or do trust validation, it'll silently accept self-signed certs - but it's still a huge positive step.

Links2004, 2015-04-23:
i have add the latest sdk + ssl patches 
see here:
https://github.com/Links2004/Arduino/commit/af6c4008dfffdd2afca4a759cccdea8c7fe4c3ee
pull request is waiting for merged (#93)

igrr, 2015-04-23:
Merged that, thanks.
Just one (obvious) thing to note: adding SSL libraries ([axTLS](http://axtls.sourceforge.net/))is one thing, but to make them useful we also need to add SSL support to WiFiClient/WiFiServer.

vicatcu, 2015-06-17:
@igrr does this issue cover adding HTTPS support to a WebClient application for accessing secure-connection-only APIs? What's the roadmap/timeline for including this feature? Is it *actually* possible?

vicatcu, 2015-06-27:
@Links2004 @igrr Do I read this thread correctly that there *is* support for an HTTPS WebClient built into the core now, but WiFiClient class "just" needs that core functionality integrated into it?

abl, 2015-06-27:
@vicatcu:

> does this issue cover adding HTTPS support to a WebClient application for accessing secure-connection-only APIs?

Yes - HTTPS is just HTTP over SSL sockets, so once SSL sockets are integrated with WiFiClient/WiFiServer, you should be able to talk to HTTPS servers.

> What's the roadmap/timeline for including this feature?

I'm not very familiar with this project's codebase.

> Is it actually possible?

Yes. Other projects (like https://github.com/tuanpmt/esp_bridge) are able to make HTTPS requests from an ESP8266. The primary limiting factor is certificate size and protocol support (TLS v1.2 is not yet supported and is mandatory for some servers) - and that the Espressif SDK doesn't always have working SSL built in to it (it works now!)

> Do I read this thread correctly that there is support for an HTTPS WebClient built into the core now, but WiFiClient class "just" needs that core functionality integrated into it?

It's now possible to write a WiFiClient that opens a secure socket. The existing WebClient shouldn't need major changes.

vicatcu, 2015-06-27:
@abl cool, I need to come up to speed on on getting a dev environment set up so I can contribute...

molda, 2015-07-13:
Hi,
is there any update on this?
I would love to help, unfortunately i'm not a friend with C and not good at C++ either, well i just know enough to do what i need and sometimes struggling a lot.
Looking forward to see the SSL support.

Great job guys.
Thanks

glennalgie, 2015-07-20:
looking forward to SSL support as well for ESP8266. I am looking at using Golgi.io arduino stack as an option for the SSL shim as they provide other time saving and mission critical networking support functions.

WickedDevice, 2015-07-25:
@igrr @Links2004 can you give any pointers on how to use axTLS to implement HTTPS capability into WiFiClient/WiFiServer? Any updates on progress here would be appreciated.

vicatcu, 2015-07-25:
@igrr @Links2004 can you give any pointers on how to use axTLS to implement HTTPS capability into WiFiClient/WiFiServer? Any updates on progress here would be appreciated.

igrr, 2015-07-25:
@vicatcu
WiFiClient needs only TLS, not HTTPS. You can use WiFiClient for things other than HTTP — SSH and SMTP to name a few. The way to do this is to create a new class, let's say WiFiSecureClient, and override read/write methods of WiFiClient to pump data through TLS engine. Some additional methods for certificate management should be added as well. I'm not sure if there is anything that needs to be done on the server side, since WiFiServer's only job is to create a WiFiClient instance for every new connection. I suppose someone more familiar with TLS can comment on this.

vicatcu, 2015-07-25:
@igrr ah yes of course, that (that WiFiClient is not *just* about HTTP) makes a lot of sense.

fmgomes, 2015-07-30:
@vicatcu , have you started to implement the TLS on the WiFiClient? I need to connect to a webserver using https, and I'm interested in this functionality, and could try to help (don't know if I am able to do it, no previous experience with HTTPS or TLS).

Best regards

Fernando

vicatcu, 2015-07-30:
@fmgomes honestly I haven't really gotten anywhere yet - it's a pretty steep learning curve for me so I'm kind of in the same boat as you :-/. It's a bit frustrating to know it's possible for the last few months, but that I haven't had time / know-how / etc to help make it happen. @igrr spelled out what needs to happen pretty well in his last response, but I'm not sure what "certificate management" requires...

abl, 2015-07-30:
When a client connects to a server and performs a TLS handshake (which is what turns a socket in to a secure socket, and HTTPS is just HTTP on a secure socket) the server sends over a certificate. The certificate is what identifies the server; without the certificate validation step, HTTPS prevents against eavesdropping but you could potentially be communicating with an attacker (aka a MitM attack.)

Validating a certificate:

1. Check to see if the name of the server you're trying to connect to matches the server name on the certificate.
2. Check to see if the certificate has expired.
3. Check to see if the certificate was issued by a trusted CA.
4. Check to see if the certificate has been revoked.

1 should be pretty easy to implement; the names should be exactly equal. Wildcard certificates exist but are considered harmful; not supporting them is probably fine for v1.

2 can probably be ignored for now; the vast majority of ESP8266 work will be bound to specific certificates because...

3 is the most complex part. On a normal computer, the OS (and sometimes the browser) maintains a list of trusted root certificates. For Debian, the size of this certificate package is 502kB (see https://packages.debian.org/sid/ca-certificates) - it's possible to shrink this down a bit but it's obviously a nonstarter for the average ESP8266 project. An alternative to this is simply maintaining a list of certificate fingerprints and checking that instead - creating an explicit whitelist of certificates. That's what I'd recommend - and when you're using certificate fingerprints, you can be lax about expiration and revocation and manage it yourself.

4 is a complicated topic - check out CRL and OCSP if you're curious - and way beyond what we'd want to do in an embedded system.

AxTLS probably has simple-ish calls to handle all of this; they'd need to be exposed via SecureWifiClient.

SecureWifiServer would need a new method that allows a user to set the public and private key to use.

mtnbrit, 2015-07-30:
does thread this also cover applying TLS to MQTT eventually?

anteph, 2015-08-02:
For what I've read, adding https support is a little bit hard. But what about simple ciphering, for direct socket communication? Something with a symmetric key, that would be a good start. 

I've read in the Espressif page that ESP8266 has a built in AES engine, but they don't specify if it is hardware or software based. Has anyone tried it?

torntrousers, 2015-08-02:
Just a comment on that list of work from @abl - to get going i'd be quite happy to not have any of that certificate validation stuff. Wont be sending my bank account details or anything top secret just need to post to a remote service that only supports https. So just binning the certificate after its read would be fine for my uses.

drmpf, 2015-08-02:
If you just want to prevent un-authorized use rather then hid the messages, check out this page on using SipHash as a secure hash.
http://www.forward.com.au/pfod/secureChallengeResponse/index.html

cottsak, 2015-08-28:
So what's the latest?

* Can I make a HTTPS request?
* can I use TLS 1.2?
* can I pin a certificate (avoid the CA lookups)?

jacealot, 2015-08-28:
Is it now possible to make a HTTPS request?
Or are there any libraries out there that I could use?

igrr, 2015-08-31:
Started working on TLS support: https://github.com/igrr/axtls-8266
This is the same library that Espressif uses for their libssl, so TLS 1.0/1.1 only.

abl, 2015-09-02:
Going that route, it seems like wolfSSL and mbed TLS (well, especially mbed TLS) might be easier to bring in, although we know axTLS works since Espressif has it working :)

They also support TLS 1.2.

igrr, 2015-09-02:
Both wolfSSL and PolarSSL (aka mbed TLS) are GPL (not LGPL) unless you
apply for a commercial license. I would like to keep the library
LGPL-compatible.

abl, 2015-09-02:
Ah, I missed that on WolfSSL.

For PolarSSL/mbed, they have an explicit exception list that includes the LGPL: https://tls.mbed.org/foss-license-exception

The language appears to prevent the distribution of a non-GPL fork of mbed TLS. Certainly a larger headache than axTLS's BSD-ish license.

cottsak, 2015-09-10:
I can't believe this conversation has come to licencing..

igrr, 2015-09-14:
WiFiClientSecure is mostly ready now. Things left to do:
- seed PRNG properly
- check host name against certificate's common name and alternative DNS names (wildcards are quite common, 4 out of 5 services I used for my tests do use wildcards).
- release memory allocated for certificate storage after check is complete

cottsak, 2015-09-14:
@igrr Coz I'm new, what is the simplest and quickest way to get your changes onto my esp8266?

vicatcu, 2015-09-14:
@igrr that's awesome, I'm looking forward to trying it out. Thanks for your work on this!

igrr, 2015-09-14:
@cottsak You can either [build IDE from source](https://github.com/arduino/Arduino/wiki/Building-Arduino), wait a bit for the next staging version.

cottsak, 2015-09-15:
@igrr So that's it? Once built, run the IDE from the build output location and the `WiFiClientSecure` changes will be available to my sketch and should upload to my esp8266 no problem?

cottsak, 2015-09-17:
@igrr you mean https://github.com/esp8266/Arduino not https://github.com/arduino/Arduino/wiki/Building-Arduino right?

igrr, 2015-09-17:
That link (https://github.com/arduino/Arduino/wiki/Building-Arduino) was to instructions for building the IDE. Do a git clone of [this repository](https://github.com/esp8266/Arduino) and follow the instructions from that link.

cottsak, 2015-09-17:
@igrr got it. thanks

cottsak, 2015-09-17:
@igrr Now that I've built `macosx/arduino-1.6.6-macosx.zip`, how do i manually add the esp8266 board to my IDE installation since I can't use the Board Manager?

igrr, 2015-09-17:
If you have done git clone of github.com/esp8266/Arduino, your IDE will have ESP8266 support built-in.
Just run it with `open build/macosx/work/Arduino.app`.

cottsak, 2015-09-17:
@igrr I had to remove the `esp8266` board from my default IDE install. Then when I started the IDE from the build output the `esp8266` board was back but with the latest source changes.

cottsak, 2015-09-17:
Finally got the `HTTPSRequest` example working and it seemed to connect to Github. However, it then proceeds to appear to crash and dump some debugging context to serial. Anyone have any idea about this?

```

connecting to itsatrap
........
WiFi connected
IP address: 
10.1.1.11
connecting to api.github.com
certificate matches
requesting URL: /repos/esp8266/Arduino/commits/esp8266/status
request sent
headers received
esp8266/Arduino CI successfull!
reply was:
==========
{"state":"success","statuses":[{"url":"https://api.github.com/repos/esp8266/Arduino/statuses/89df2854c8e3e5923b600bcfd70afbaa9964cdfe","id":309793032,"state":"success","description":"The Travis CI build passed","target_url":"https://travis-ci.org/esp8266/Arduino/builds/80635331","context":"continuous-integration/travis-ci/push","created_at":"2015-09-16T13:34:51Z","updated_at":"2015-09-16T13:34:51Z"}],"sha":"89df2854c8e3e5923b600bcfd70afbaa9964cdfe","total_count":1,"repository":{"id":32969220,"name":"Arduino","full_name":"esp8266/Arduino","owner":{"login":"esp8266","id":8943775,"avatar_url":"https://avatars.githubusercontent.com/u/8943775?v=3","gravatar_id":"","url":"https://api.github.com/users/esp8266","html_url":"https://github.com/esp8266","followers_url":"https://api.github.com/users/esp8266/followers","following_url":"https://api.github.com/users/esp8266/following{/other_user}","gists_url":"https://api.github.com/users/esp8266/gists{/gist_id}","starred_url":"https://api.github.com/users/esp8266/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/esp8266/subscriptions","organizations_url":"https://api.github.com/users/esp8266/orgs","repos_url":"https://api.github.com/users/esp8266/repos","events_url":"https://api.github.com/users/esp8266/events{/privacy}","received_events_url":"https://api.github.com/users/esp8266/received_events","type":"Organization","site_admin":false},"private":false,"html_url":"https://github.com/esp8266/Arduino","description":"Arduino IDE for ESP8266","fork":false,"url":"https://api.github.com/repos/esp8266/Arduino","forks_url":"https://api.github.com/repos/esp8266/Arduino/forks","keys_url":"https://api.github.com/repos/esp8266/Arduino/keys{/key_id}","collaborators_url":"https://api.github.com/repos/esp8266/Arduino/collaborators{/collaborator}","teams_url":"https://api.github.com/repos/esp8266/Arduino/teams","hooks_url":"https://api.github.com/repos/esp8266/Arduino/hooks","issue_events_url":"https://api.github.com/repos/esp8266/Arduino/issues/events{/number}","events_url":"https://api.github.com/repos/esp8266/Arduino/events","assignees_url":"https://api.github.com/repos/esp8266/Arduino/assignees{/user}","branches_url":"https://api.github.com/repos/esp8266/Arduino/branches{/branch}","tags_url":"https://api.github.com/repos/esp8266/Arduino/tags","blobs_url":"https://api.github.com/repos/esp8266/Arduino/git/blobs{/sha}","git_tags_url":"https://api.github.com/repos/esp8266/Arduino/git/tags{/sha}","git_refs_url":"https://api.github.com/repos/esp8266/Arduino/git/refs{/sha}","trees_url":"https://api.github.com/repos/esp8266/Arduino/git/trees{/sha}","statuses_url":"https://api.github.com/repos/esp8266/Arduino/statuses/{sha}","languages_url":"https://api.github.com/repos/esp8266/Arduino/languages","stargazers_url":"https://api.github.com/repos/esp8266/Arduino/stargazers","contributors_url":"https://api.github.com/repos/esp8266/Arduino/contributors","subscribers_url":"https://api.github.com/repos/esp8266/Arduino/subscribers","subscription_url":"https://api.github.com/repos/esp8266/Arduino/subscription","commits_url":"https://api.github.com/repos/esp8266/Arduino/commits{/sha}","git_commits_url":"https://api.github.com/repos/esp8266/Arduino/git/commits{/sha}","comments_url":"https://api.github.com/repos/esp8266/Arduino/comments{/number}","issue_comment_url":"https://api.github.com/repos/esp8266/Arduino/issues/comments{/number}","contents_url":"https://api.github.com/repos/esp8266/Arduino/contents/{+path}","compare_url":"https://api.github.com/repos/esp8266/Arduino/compare/{base}...{head}","merges_url":"https://api.github.com/repos/esp8266/Arduino/merges","archive_url":"https://api.github.com/repos/esp8266/Arduino/{archive_format}{/ref}","downloads_url":"https://api.github.com/repos/esp8266/Arduino/downloads","issues_url":"https://api.github.com/repos/esp8266/Arduino/issues{/number}","pulls_url":"https://api.github.com/repos/esp8266/Arduino/pulls{/number}","milestones_url":"https://api.github.com/repos/esp8266/Arduino/milestones{/number}","notifications_url":"https://api.github.com/repos/esp8266/Arduino/notifications{?since,all,participating}","labels_url":"https://api.github.com/repos/esp8266/Arduino/labels{/name}","releases_url":"https://api.github.com/repos/esp8266/Arduino/releases{/id}"},"commit_url":"https://api.github.com/repos/esp8266/Arduino/commits/89df2854c8e3e5923b600bcfd70afbaa9964cdfe","url":"https://api.github.com/repos/esp8266/Arduino/commits/89df2854c8e3e5923b600bcfd70afbaa9964cdfe/status"}
==========
closing connection

ctx: cont 
sp: 3ffeb720 end: 3ffeb9a0 offset: 01b0

>>>stack>>>
3ffeb8d0:  401000b4 3fff5178 3ffe9404 4022c4bb  
3ffeb8e0:  3ffe9268 3fff3180 3fff318c 4022ae7a  
3ffeb8f0:  3ffe9268 3fff6578 3fff5178 4022a060  
3ffeb900:  3ffe93f3 00000000 3fff2ed8 40204572  
3ffeb910:  402075f9 3ffe93e8 3ffeb930 40203f60  
3ffeb920:  3ffe9268 3ffe93e8 3ffeb9f8 4020245e  
3ffeb930:  3ffe95a8 00000000 000003e8 00008df3  
3ffeb940:  00000000 3fff2ea0 3fff2ed8 3fff2de0  
3ffeb950:  00000012 00000012 3fff9150 000011ad  
3ffeb960:  000011ad 3fff3248 0000002d 0000002d  
3ffeb970:  00000000 00000000 00000000 3ffeb9cc  
3ffeb980:  3fffdc20 00000000 3ffeb9c4 4020185f  
3ffeb990:  00000000 00000000 3ffea980 40100398  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)


 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

igrr, 2015-09-17:
Uhm, it worked fine when I wrote this sample. Will check today, perhaps the commit from yesterday broke something.

torntrousers, 2015-09-17:
The sample crashes like that for me too, a couple of seconds after the request has completed ok, but using WiFiClientSecure in other code seems to work ok for me. Had a play around with the sample - moving the line "WiFiClientSecure client;" outside of the setup() function to the top of sketch looks like it fixes the problem. Some garbage collection problem?

cottsak, 2015-09-18:
I can verify that @torntrousers workaround also prevents the crash for me. 

Yeh, sounds like a memory-related problem to me too. There's no compile time error since `client` isn't referenced in `void loop()` but maybe under the hood something else is trying to access the instance and because it's not in global scope? it's barfing.. ?? Just a guess.

cottsak, 2015-09-18:
I'm still getting more of these crashes.. they seem to be host-dependant. ie, some hosts don't trigger them and some do.

Here is another dump:

```
Sending alarm request..

ctx: cont 
sp: 3ffeb7b0 end: 3ffebac0 offset: 01b0

>>>stack>>>
3ffeb960:  401000b4 3fff5298 3ffebb8c 4022c86b  
3ffeb970:  40208bb7 3fff35b8 3fff35c4 4022b22a  
3ffeb980:  000001bb 3fff6698 3fff5298 4022a410  
3ffeb990:  3ffebb80 00000000 3fff2ff8 40204922  
3ffeb9a0:  40208ec4 000001bb 3ffea690 40204828  
3ffeb9b0:  3ffe92f7 40208eac 3ffea690 40203b0e  
3ffeb9c0:  dbca1032 3ffe92f7 402017ec 3ffeaaa0  
3ffeb9d0:  3ffebb18 000001bb 3ffea690 402047e0  
3ffeb9e0:  3ffe96e8 dbca1032 3ffe92f5 3ffebaec  
3ffeb9f0:  3ffebb18 000001bb 3ffea690 40203ad5  
3ffeba00:  3ffe96e8 dbca1032 3ffe96e8 dbca1032  
3ffeba10:  3ffe92f7 00000001 3ffebb18 402070a0  
3ffeba20:  3ffebb18 3ffea690 3ffea7ac 40204800  
3ffeba30:  3ffea6ac 00000000 3ffea7ac 40202370  
3ffeba40:  3ffe93ce 00000000 3ffebb18 4020651d  
3ffeba50:  3ffe9274 3ffebaec 3ffebb18 402070a0  
3ffeba60:  402010ae 000003e8 3ffebb18 4020710c  
3ffeba70:  3fffdc20 00000000 3ffebb18 3ffebaec  
3ffeba80:  3fffdc20 00000000 3ffea7ac 402025ad  
3ffeba90:  4020180d 00000000 3ffebae4 402022e0  
3ffebaa0:  00000000 00000000 3ffebae4 4020186a  
3ffebab0:  00000000 00000000 3ffeaaa0 40100398  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```

cottsak, 2015-09-18:
Also, is this a problem:

```
GET /a/check HTTP/1.1
Host: www.howsmyssl.com
User-Agent: ESP8266
Connection: close

HTTP/1.1 200 OK
Content-Length: 568
Access-Control-Allow-Origin: *
Connection: close
Content-Type: application/json
Date: Fri, 18 Sep 2015 07:50:29 GMT
Strict-Transport-Security: max-age=631138519; includeSubdomains; preload

{"given_cipher_suites":["TLS_RSA_WITH_AES_128_CBC_SHA","TLS_RSA_WITH_AES_256_CBC_SHA","TLS_RSA_WITH_RC4_128_SHA","TLS_RSA_WITH_RC4_128_MD5"],"ephemeral_keys_supported":false,"session_ticket_supported":false,"tls_compression_supported":false,"unknown_cipher_suite_supported":false,"beast_vuln":false,"able_to_detect_n_minus_one_splitting":false,"insecure_cipher_suites":{"TLS_RSA_WITH_RC4_128_MD5":["use RC4 which has insecure biases in its output"],"TLS_RSA_WITH_RC4_128_SHA":["use RC4 which has insecure biases in its output"]},"tls_version":"TLS 1.1","rating":"Bad"}
```

Particularly the RC4 stuff and that the client is only supporting TLS 1.1?

igrr, 2015-09-18:
Found the cause of the crash, fix coming a bit later today.

Regarding only TLS 1.1 being supported — please read the thread above,
axTLS doesn't support 1.2. Regarding cipher suites — there is still much
work to do, i.e. add SHA256, disable outdated stuff... Any help is
appreciated.

cottsak, 2015-09-21:
Thanks @igrr. I'd love to help but much of this c++ networking stack stuff is over my head. I'm only just managing to consume these libs it in my code.

timkay, 2015-09-29:
Running the staging version, I get the error below. Is this error the same as @cottsak reported above? It doesn't look to be.

I tried to run the latest code. I cloned the latest code, installed ant, compiled the Arduino IDE, extracted arduino-1.6.6-windows.zip. It won't run, saying,

java.lang.UnsatisfiedLinkError: C:\Users\timkay\Desktop\arduino-1.6.6\lib\AStylej.dll: Can't load IA 32-bit .dll on a AMD 64-bit platform

connected with 688 Berry Ave, channel 11
dhcp client start...
ip:192.168.8.120,mask:255.255.255.0,gw:192.168.8.254
connecting
State:	sending Client Hello (1)
State:	receiving Server Hello (2)
Error: invalid protocol message
Alert: handshake failure
Alert: close notify
connection failed
Fatal exception (28):
epc1=0x40204011, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, d
Exception (28):
epc1=0x40204011 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffeb5e0 end: 3ffeb840 offset: 01a0

>>>stack>>>
3ffeb780:  3fff2560 40202ee4 3ffeb898 40206bb0  
3ffeb790:epc=0x00000000
  00000000 3ffeb898 3ffea540 40202540  
3ffeb7a0:  6c61686d 7766686c 00736177 00000000  
3ffeb7b0:  3ffeb898 00000001 3ffe92e5 40206a70  
3ffeb7c0:  3ffeb800 00000011 3ffe9285 3ffeb86c  
3ffeb7d0:  3fffdc20 00000006 3ffeb898 40205fc1  
3ffeb7e0:  3ffe92e4 00000000 3ffeb898 3ffeb86c  
3ffeb7f0:  3fffdc20 00000000 3ffeb898 40206b44  
3ffeb800:  00000000 00000000 00000016 40101a6d  
3ffeb810:  4020180d 00000000 3ffeb898 4020249c  
3ffeb820:  00000000 00000000 3ffeb864 4020186a  
3ffeb830:  00000000 00000000 3ffea820 40100398  
<<<stack<<<

cottsak, 2015-09-29:
@timkay Staging is still too old to have included @igrr's latest changes. I would try to help with your buid adventure but I was on OS X and it seemed to work for me.

grahamehorner, 2015-09-30:
@cottsak @igrr I've installed a build from the https://www.arduino.cc/en/Main/Software hourly build link; if staging is to old; is there a more current build of the board package available? to developers for the arduino IDE (hourly build)?

igrr, 2015-09-30:
I've just updated staging to ef26c5f, so now you can just try and pull the latest boards manager package.

grahamehorner, 2015-09-30:
@igrr  your the man! great many thanks :+1:

timkay, 2015-10-01:
@igrr Up and running now with TLS working, talking to Firebase. Thank you!

igrr, 2015-10-01:
Good to know! Don't put this into anything critical yet though, because random number generator is not seeded properly. I'll fix this in the next release.

grahamehorner, 2015-10-01:
just completed my first RESTful call to Azure Service Bus over HTTPS; thanks :dancer: +1

goldbauch, 2015-10-03:
I'm sorry, but I still get a "'WiFiClientSecure' was not declared in this scope" - the most current version of Staging (packaged) seems to be 1.6.5-1160-gef26c5f, which I installed over the BoardManager. Doesn't it work when just adding the link in the IDE Settings, do I have to build it on my own? Any help appreciated, thx

goldbauch, 2015-10-03:
And @igrr - Thanks for the effort on SSL anyway, that's exactly what I needed!

timkay, 2015-10-03:
You need another include, as I show here. 

    #include <ESP8266WiFi.h>
    #define USE_SSL 1
    #if USE_SSL
    #   include <WiFiClientSecure.h>
        WiFiClientSecure client;
    #   define PORT 443
    #else
        WiFiClient client;
    #   define PORT 80
    #endif

goldbauch, 2015-10-03:
Found that out that very moment, digging into ESP8266WiFi.c - =) 

Still: Thanks a lot!

timkay, 2015-10-06:
I can't seem to open two WiFiClientSecure connections at the same time. The second connect() causes a WDT reset. See https://www.dropbox.com/s/u6qma92jwaa65pr/esp_test.ino?dl=0

    scandone
    state: 0 -> 2 (b0)
    state: 2 -> 3 (0)
    state: 3 -> 5 (10)
    add 0
    aid 1
    cnt 
    
    connected with 688 Berry Ave, channel 11
    dhcp client start...
    ip:192.168.8.121,mask:255.255.255.0,gw:192.168.8.254
    About to connect() to www.yahoo.com port 443 uri /error
    connected
    About to connect() to www.yahoo.com port 443 uri /error
    E:M 1048
    
    Soft WDT reset
    
    ctx: cont 
    sp: 3ffeb410 end: 3ffeba20 offset: 01b0
    
    >>>stack>>>
    3ffeb5c0:  3fffb8e0 401000a2 3fff9a40 4022c0a7  
    3ffeb5d0:  3fff9128 3fff9a60 00000081 4022cd0d

igrr, 2015-10-06:
With two connections you are running out of memory, so only one connection
at a time for now. I think Espressif's version of axTLS shares this
limitation.

[99 more replies]

closed 2016-02-27 by igrr