esp8266/Arduino #44: How to detect the use of the ESP8266 ?

marcoschwartz, 2015-04-03:
I am the developer of the aREST library () that implements a REST interface for Arduino & Raspberry Pi boards.

I am currently working on adapting the aREST Arduino library to your project, it nearly works but in the code I need to detect when an ESP8266 is used, and when a classical Arduino is used. For example to detect an Arduino Mega I use:

#if defined(__AVR_ATmega1280__)

Is there something similar I could use to detect that the ESP8266 is selected? Thanks!

Zeron, 2015-04-03:
https://github.com/esp8266/Arduino/commit/422a4469503a681bf6f1d313c41a1ac0b8c5ce61 seems to imply that #if defined(ESP8266) will work once you are using a version with that change incorporated.

marcoschwartz, 2015-04-03:
That worked for me, thanks! Closing the issue.

closed 2015-04-03 by marcoschwartz