esp8266/Arduino #47: Any example about LiquidCrystal I2C Library ?

ly0, 2015-04-04:


ManiackCZ, 2015-04-04:
Not LiquidCrystal library, but I got working OLED i2c display with ssd1306 driver by slightly modding Adafruit libraries. I plan to post it on github soon after some aftertouch.

mperez4, 2015-04-04:
Cool!

sticilface, 2015-04-13:
It would be fantastic if someone could investigate getting the u8glib working.  The chaps over at nodeMCU have it working on their lua stuff...  I tried to look into their code to see what they might have done... but there is no real hint of it.. I wouldn't know what I'm doing anyway!

gaelperon, 2015-05-11:
@ManiackCZ can you share you work with the Oled display ? that would be really appreciated

rvbcrs, 2015-05-11:
@ManiackCZ +1 for me, I really want to use the I2C OLED display on my ESP8266! Would be super if you could share it!

duncan-a, 2015-05-11:
@ManiackCZ <https://github.com/ManiackCZ> and one more please - every time
I try to modify anything, it breaks!!!

ManiackCZ, 2015-05-12:
First of all, thank you for your interest in that library and sorry I didn´t responded earlier, but unfortunetely, I am having my high school leaving exams and I had no time for coding due to preparation for them. But since I am having them next week (19th of May), I promise I will get back into that library as soon as I can after I pass my exams (hopefully). I will try my best to publish that code ASAP, and since I need to do just few tweaks, examples, licence stuff and few tests (i2c library was changed to new one since then), i think its managable.

duncan-a, 2015-05-12:
Good luck in the exams.............

tzapu, 2015-05-21:
Hi,
I am also very interested in this.

Cheers

duncan-a, 2015-05-21:
When using the OLED displays with Arduino, I seem to remember that the
entire display memory is buffered in the Arduino's RAM (?). If the OLED is
being used with the ESP8266, can it use the Flash (the Olimex boards have
16MB, so there's plenty of space).

Just a thought...

tzapu, 2015-05-22:
with the help of the esp8266 guys i ve added this fork that works now for me and can drive a ssd1306 from an esp 
https://github.com/tzapu/Adafruit_SSD1306

maybe it helps someone

tablatronix, 2016-03-31:
Not using I2C but you will have to remove this startup delay from the liquidcrystal library.
`delayMicroseconds(50000);`

Or add a long enough delay before lcd.begin to allow wifi to run.

devyte, 2017-10-20:
Closing per #3655 .

closed 2017-10-20 by devyte