esp8266/Arduino #50: ESP hangs up

iMarvinS, 2015-04-04:
I´m trying to build a wireless PIR sensor from a ESP-07.
I have the code already working but after a undefined time the firmware hangs up and logs some unusual stuff on the UART.
The IDE I´ve used is built by myself and is the most actual version fresh from the esp8266 branch.

The Sketch:
```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>

const int pirPin = 2;
const char* ssid = "Home 7362";
const char* password = "mysecretpasswordwhichnobodyshouldknowxDD";
const char* host = "192.168.178.24";
const int port = 80;
int lastMillis;
bool lastState;
WiFiClient wifiClient;

void pirInterrupt();

void setup() {
  Serial.begin(9600);
  
  WiFi.begin(ssid, password);
  Serial.println("");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  
  pinMode(2, INPUT);
  //attachInterrupt(2, pirInterrupt, FALLING);
  lastMillis = millis();
  
  while(digitalRead(pirPin) == LOW){
    Serial.println("Calibrating...");
    delay(200);
  }
  
}

void loop() {
  if(digitalRead(pirPin) == HIGH && lastState == LOW && millis() - lastMillis >= 5000){
      pirInterrupt();
    lastMillis = millis();
  }
  lastState = digitalRead(pirPin);
  
  delay(20);
}

void pirInterrupt() {
  Serial.println("PIR: alarm triggered!");
  
  if (!wifiClient.connect(host, 80)) {
    Serial.println("Connection failed!");
    return;
  }
  
  String url = "/pirpush/push.php";
  url += "?message=Der+Postmann+ist+da!";
  url+= "&id=1";
  
  Serial.print("Requesting URL: "); 
  Serial.println(url);
  
  wifiClient.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" + 
               "Connection: close\r\n\r\n");
  /*
  delay(50);
  
  while(wifiClient.available()){
    String line = wifiClient.readStringUntil('\r');
    Serial.print(line);
  }
  */
  Serial.println();
  Serial.println("Closing connection.");
}
```

The output on serial:
[image]

What is the issue with it?

derpicknicker1, 2015-04-04:
I have the same  with an ESP-01. I tested both server example sketches (only cachanged SSID and PW). After random time the module does not respond. Terminal looks like it hangs while handling the request. My router shows that the module is not connected anymore. Only a restart helps (reset not tested). Tested with two modules.
When i use the client example sketch everything works fine. I ran it for 4 days every 10 seconds. No problem.

igrr, 2015-04-05:
@iMarvinS 
I see ```LmacRxBlk:1``` appears in the log. This means that the the buffers allocated for the TCP/IP stack  are full with data. It appears that you are not reading all the data that is received on the connection.

Does the same issue happen when you uncomment the following lines near the end of your sketch?
```
delay(50);

  while(wifiClient.available()){
    String line = wifiClient.readStringUntil('\r');
    Serial.print(line);
  }
```

igrr, 2015-04-05:
@derpicknicker1 What are you using for a power supply? I have seen similar behaviour when powering modules from a usb-serial adapter. I have also noticed some random (although quite rare) issues when the GPIO2 pin was not pulled high. 
When I went for an external power supply, plus some decoupling capacitors, and a 10k pull-up on GPIO2, these problems went away.

igrr, 2015-04-05:
@iMarvinS 
Looks like there is indeed a resource leak happening in your case.

Could you please try your sketch again, adding one line here:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/include/ClientContext.h#L92
```
 }
 flush(); // <---- add this line
 delete this;
```
Let me know if this fixes the issue.

Links2004, 2015-04-06:
with this line there is a possibility of an NULL ptr error.

fix:
```cpp
    void flush()
    {
        if (!_rx_buf) {
            return;
        }
        if(_pcb) {
            tcp_recved(_pcb, (size_t) _rx_buf->tot_len);			
        }
	pbuf_free(_rx_buf);
        _rx_buf = 0;
        _rx_buf_offset = 0;
    }
```

may call flush() befor closing the connection is a better

igrr, 2015-04-11:
@iMarvinS
Did you try the latest version? Does it fix your issue?

iMarvinS, 2015-04-12:
@igrr 
Yes and it works now without any interrupts.

Thanks!

closed 2015-04-12 by iMarvinS

lnsriravichandran22, 2015-09-05:
@igrr What values of de-coupling capacitors did you use?

drmpf, 2015-09-05:
This circuit works for me.
http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/index.html