esp8266/Arduino #51: Issue with float values/calculation

ManiackCZ, 2015-04-04:
The calculations with floats doesnt work. It averages to integer.
<code>
float i;
i = 5/2;
</code>
I have hooked up oled display and when I print the value, it shows just 2.00. First, I thought it may be just problem with display, so I get it to send the value back over serial but I got <code>2.0000000000</code> too.

thomasfla, 2015-04-04:
Normal, you do an int operation. Do 5/2.0 or ((float)5)/2

ManiackCZ, 2015-04-04:
Thx, missed that.

closed 2015-04-04 by ManiackCZ

leandrogs, 2015-05-14:
I am also having this issue. I receive different data from sensor using GPIO (Wire library) and at some point I have `scaled.YAxis = ((float)raw.YAxis) + 0.92;` where `scaled.YAxis` is a float and `raw.Yaxis` is an int. I know that `raw.YAxis` is 243, so my output should be 243.92 but what I get is 219.08.