esp8266/Arduino #55: build improvement for Windows users

robertgregor, 2015-04-05:
Hello, I tried to build the arduino-1.6.1-windows.zip from source on Windows 7, 64. I had two issues, which I have fixed by myselve however could be good to maybe improve the ant build process. The first problem was with missing unzip program and the second was with missing tar program. I belive, that most people are building under linux, so this is by default there, but not under windows. I've just copied the unzip.exe and tar.exe to the build directory and everything was fine then.

cimba007, 2015-04-08:
I guess every unzip programm like 7zip or winrar can cope with tar gz

robertgregor, 2015-04-09:
In fact as I wrote, the Ant is strictly calling unzip.exe and tar.exe - so these must be in the execution path, if not, the Ant task will fail.

closed 2015-04-09 by robertgregor