esp8266/Arduino #56: Improvement of ESP8266WiFi class

robertgregor, 2015-04-05:
Hello, in fact I have played with your project, it looks pretty nice! Thanks for the great work! I understand, that you tried to make this class compatible with the Arduino native WiFi shield, however, I am thinking, that it could be nice to maybe add the additional functionalities to enhance it little bit.
E.g. from the ESP AT firmware, I like a lot the functionality to remember the last AP, where the module connected and during the startup connect automatically there. something like methods WiFi.save() and  WiFi.connect(). (save will save all the connection params e.g. also the dhcp, ip address and so on.) Also for the battery powered operations, where the module should be placed to deep sleep mode and wake e.g. each 10 minutes to send the temperature to add some method like WiFi.ulp(int wakeTime). Or modify the WiFiClient class don't know. It could be also combined with the interrupt e.g. to setup interrupt for the specific pin and then do WiFi.ulp(). And the module wakes up when the interrupt is fired.

igrr, 2015-04-21:
Which one is the right SimpleTimer, https://github.com/infomaniac50/SimpleTimer or https://github.com/jfturcot/SimpleTimer?

robertgregor, 2015-04-27:
Hello, I used https://github.com/infomaniac50/SimpleTimer written by mromani@ottotecnica.com
. Working great.

devyte, 2017-10-20:
I think the Ticker lib addresses this. Also, the previous comment says the 3rd party lib works.
Closing.

closed 2017-10-20 by devyte