esp8266/Arduino #59: Add aREST library in list of supported libraries

marcoschwartz, 2015-04-07:
Hello,

I am the main developer of the aREST (http://arest.io/) library, which allows the user to completely control Arduino-compatible boards via HTTP using RESTful commands. For example to change the state of a pin on a board you just need to go to a web browser and type:

http://192.168.1.103/digital/5/1

I just released a new version of the library (https://github.com/marcoschwartz/aREST) which works with the ESP8266 chip & your project. Would you consider adding a link to it into the README file? I think it will be helpful for many people. Thanks!

igrr, 2015-04-07:
Fantastic! I wish I knew about this library before and not reinvented a wheel.

@marcoschwartz, on a slightly unrelated topic, do you think adding automatic argument parsing functionality would make sense for your library?
Something along these lines:
```
int myFunction(int arg1, String arg2, float arg3) {
...
}

rest.function("myfunction", myFunction, "arg1", "arg2", "arg3"); 
```
Which would look up the necessary arguments and convert them to correct types through a bit of template metaprogramming magic.

igrr, 2015-04-07:
Added to readme in 7b54e83.

marcoschwartz, 2015-04-07:
Thanks for adding the library to the Readme file! 

Sure, I am definitely considering extending the functionalities of the library, like having an automatic argument parsing for functions, many people have already asked for that.

Basically I was limited by the Arduino Uno which was my original target for the library. But now I plan to extend the functionalities of the library on more powerful chips like the ESP8266, while keeping the compatibility with 8-bits microcontrollers.

closed 2015-04-07 by marcoschwartz