esp8266/Arduino #65: Wire.endTransmission() wrong return value

igrr, 2015-04-11:
Currently it returns the number of bytes transmitted.
This is how it should work:
```
Returns byte, which indicates the status of the transmission:
0:success
1:data too long to fit in transmit buffer
2:received NACK on transmit of address
3:received NACK on transmit of data
4:other error
```

erniberni, 2015-04-17:
Hi Ivan,
was there any change in the wire lib after this finding?
I think this is the reason the i2c scanner doesn't work and also the BMP180 lib expects "0" for successful module recognition.
Regards
Reinhard

igrr, 2015-05-04:
@erniberni Latest version should have this issue fixed.

erniberni, 2015-05-10:
Thanks Ivan,
I2C scanner now works perfectly. 
Reinhard

closed 2015-05-19 by igrr