esp8266/Arduino #66: libc_replacements.c can't find references to math functions

dmcinnes, 2015-04-11:
Hey just ran into this on OS X, it can't compile `libc_replacements.c` because these math functions don't exist:
Commenting these out fixed the problem (I'm not using them in my current project).

```
core.a(libc_replacements.c.o): In function `strtod':
libc_replacements.c:(.irom0.text+0x338): undefined reference to `sinh'
libc_replacements.c:(.irom0.text+0x342): undefined reference to `sinh'
libc_replacements.c:(.irom0.text+0x350): undefined reference to `hypot'
libc_replacements.c:(.irom0.text+0x35a): undefined reference to `hypot'
core.a(libc_replacements.c.o): In function `__errno':
libc_replacements.c:(.irom0.text+0x368): undefined reference to `hypotf'
libc_replacements.c:(.irom0.text+0x372): undefined reference to `hypotf'
libc_replacements.c:(.irom0.text+0x380): undefined reference to `logf'
core.a(libc_replacements.c.o): In function `__ieee754_sinh':
libc_replacements.c:(.irom0.text+0x38a): undefined reference to `logf'
libc_replacements.c:(.irom0.text+0x398): undefined reference to `log10'
core.a(libc_replacements.c.o): In function `__ieee754_hypot':
libc_replacements.c:(.irom0.text+0x3a2): undefined reference to `log10'
libc_replacements.c:(.irom0.text+0x3b0): undefined reference to `exp'
core.a(libc_replacements.c.o): In function `__ieee754_hypotf':
libc_replacements.c:(.irom0.text+0x3ba): undefined reference to `exp'
libc_replacements.c:(.irom0.text+0x3c8): undefined reference to `cosh'
core.a(libc_replacements.c.o): In function `__ieee754_logf':
libc_replacements.c:(.irom0.text+0x3d2): undefined reference to `cosh'
libc_replacements.c:(.irom0.text+0x3e0): undefined reference to `expf'
core.a(libc_replacements.c.o): In function `__ieee754_log10':
libc_replacements.c:(.irom0.text+0x3ea): undefined reference to `expf'
libc_replacements.c:(.irom0.text+0x3f8): undefined reference to `log10f'
core.a(libc_replacements.c.o): In function `__ieee754_exp':
libc_replacements.c:(.irom0.text+0x402): undefined reference to `log10f'
libc_replacements.c:(.irom0.text+0x410): undefined reference to `atan2'
core.a(libc_replacements.c.o): In function `__ieee754_cosh':
libc_replacements.c:(.irom0.text+0x41a): undefined reference to `atan2'
libc_replacements.c:(.irom0.text+0x428): undefined reference to `sqrtf'
core.a(libc_replacements.c.o): In function `__ieee754_expf':
libc_replacements.c:(.irom0.text+0x432): undefined reference to `sqrtf'
libc_replacements.c:(.irom0.text+0x440): undefined reference to `sinhf'
core.a(libc_replacements.c.o): In function `__ieee754_log10f':
libc_replacements.c:(.irom0.text+0x44a): undefined reference to `sinhf'
libc_replacements.c:(.irom0.text+0x458): undefined reference to `log'
core.a(libc_replacements.c.o): In function `__ieee754_atan2':
libc_replacements.c:(.irom0.text+0x462): undefined reference to `log'
libc_replacements.c:(.irom0.text+0x470): undefined reference to `sqrt'
core.a(libc_replacements.c.o): In function `__ieee754_sqrtf':
libc_replacements.c:(.irom0.text+0x47a): undefined reference to `sqrt'
libc_replacements.c:(.irom0.text+0x488): undefined reference to `coshf'
core.a(libc_replacements.c.o): In function `__ieee754_sinhf':
libc_replacements.c:(.irom0.text+0x492): undefined reference to `coshf'
libc_replacements.c:(.irom0.text+0x4a0): undefined reference to `atan2f'
core.a(libc_replacements.c.o): In function `__ieee754_log':
libc_replacements.c:(.irom0.text+0x4aa): undefined reference to `atan2f'
collect2: error: ld returned 1 exit status
Error compiling.
```

igrr, 2015-04-12:
@dmcinnes there was an update to the platform.txt file which added
```-lm``` flag to link math libarary. Can you please check if your
platform.txt is up to date? I just ran the build locally (OS X also) and
didn't get this issue.

dmcinnes, 2015-04-13:
@igrr Oh my mistake! When I built the OS X version I had to copy platform.txt from the last release because it was blank, must not have had those changes.

igrr, 2015-04-13:
Wow, blank?
Were you using the version from this repository or from one of the forks?

igrr, 2015-04-13:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/platform.txt looks like it's in place

sticilface, 2015-04-13:
I've been missing the esptool from the builds I've been making in the IDE see image.  I can confirm that the programmers file is there and blank. 

[image]

if i add this

```/Users/amelvin/git/Arduino/build/macosx/Arduino````

to the programmers file it works fine!  or well i have to add my modified version that uses esptool.py...

igrr, 2015-04-13:
@sticilface empty programmers file is okay, since two weeks ago esptool is an "upload method", so it is no longer in the programmers menu. See 1d6a7422.

sticilface, 2015-04-13:
Ah ok. Thanks

Still doesn't work for me though. (the one below looked like it was, i got some flash action but then it failed.  I get 100% success rate with esptool.py though.  

`````

Sketch uses 191,520 bytes (36%) of program storage space. Maximum is 524,288 bytes.
/Users/amelvin/git/Arduino/build/macosx/Arduino.app/Contents/Java/hardware/tools/esp8266/esptool -vv -cd ck -cb 115200 -cp /dev/tty.usbserial -ca 0x00000 -cf /var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gn/T/build1846545573157962261.tmp/sketch_apr13a.cpp_00000.bin -ca 0x40000 -cf /var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gn/T/build1846545573157962261.tmp/sketch_apr13a.cpp_40000.bin 
esptool v0.4.2 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/tty.usbserial
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat /var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gn/T/build1846545573157962261.tmp/sketch_apr13a.cpp_00000.bin success
opening port /dev/tty.usbserial at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	espcomm cmd: receiving 2 bytes of data
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_open
Uploading 35520 bytes from /var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gn/T/build1846545573157962261.tmp/sketch_apr13a.cpp_00000.bin to flash at 0x00000000
	erasing flash
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	setting timeout 5000
	setting character timeout 50
	done
	setting timeout 1
	setting character timeout 1
	done
	espcomm cmd: receiving 2 bytes of data
	writing flash
...................................
	setting address from 0x00000000 to 0x00040000
	espcomm_upload_file
	stat /var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gn/T/build1846545573157962261.tmp/sketch_apr13a.cpp_40000.bin success
	espcomm_open
Uploading 156040 bytes from /var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gn/T/build1846545573157962261.tmp/sketch_apr13a.cpp_40000.bin to flash at 0x00040000
	erasing flash
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	setting timeout 5000
	setting character timeout 50
	done
	setting timeout 1
	setting character timeout 1
	done
	espcomm cmd: receiving 2 bytes of data
error: failed reading byte
warning: espcomm_cmd: cant receive slip payload data
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
closing boot loader
```````

dmcinnes, 2015-04-14:
@igrr sorry I meant programmers.txt. :confused: I got it off of master of the repo.
I didn't realize it switched from "Upload Using Programmer", to just "Upload". Works for me now, thanks!

And thank you for this great project! :smile:

closed 2015-05-19 by igrr