esp8266/Arduino #67: Windows build missing xtensa

ToSa27, 2015-04-13:
Seems like the AppVeyor builds failed the last few days for Windows not finding one of the downloads (which is there if you grab it manually so probably temporary unavailability caused this).
I went ahead and did my own build which ended in a SUCCESS message but when I started using the extracted ZIP file I found that the entire xtensa folder is missing (hardware/tools/esp8266/xtensa-lx106-elf). I copied the folder from an earlier build and copied libiconv-2.dll to the arduino root folder and things started working. I had not done my own builds in the past so I'm not sure if this is a new issue or actually by purpose e.g. due to copyrights.

igrr, 2015-04-13:
Looks like a new one to me. Could you please post the full output of ```ant dist```?

ToSa27, 2015-04-13:
I just did that a few minutes ago and found the following - seems like either the tar files are invalid or the version of tar I've installed is bad. I'll dig a bit deeper and let you know:

[code]
untar-unzip-check:

untar-unzip-download:

untar-unzip-checksum:
     [echo] Testing checksum of "windows/dist/gcc-arm-none-eabi-4.8.3-2014q1-windows.tar.gz"

untar:
     [echo] Unzipping into folder windows/work/hardware/tools/
     [exec] tar: Hmm, this doesn't look like a tar archive
     [exec] tar: Skipping to next file header
     [exec] tar: Only read 6249 bytes from archive windows/dist/gcc-arm-none-eabi-4.8.3-2014q1-windows.tar.gz
     [exec] tar: Error is not recoverable: exiting now

untar-unzip-check:

untar-unzip-download:

untar-unzip-checksum:
     [echo] Testing checksum of "windows/dist/win32-xtensa-lx106-elf.tgz"

untar:
     [echo] Unzipping into folder windows/work/hardware/tools/esp8266/
     [exec] tar: Hmm, this doesn't look like a tar archive
     [exec] tar: Skipping to next file header
     [exec] tar: Only read 4674 bytes from archive windows/dist/win32-xtensa-lx106-elf.tgz
     [exec] tar: Error is not recoverable: exiting now
[/code]

ToSa27, 2015-04-13:
Files open fine in 7z - I'll install the latest version of the tar executable and try again.

ToSa27, 2015-04-13:
With a proper tar / gzip version it works fine. I ended up using bsdtar.exe (renamed to tar.exe)... issue can be closed...
While retesting I realized that the arduino team has removed the jre download a few hours ago and instead they use the existing one in $java_home in the build.xml... with that the download does not work anymore and the build process for esp8266 fails :(

closed 2015-04-13 by ToSa27