esp8266/Arduino #68: UDP Multicast example sketch

gerardwr, 2015-04-13:
Hi,

Now that the bug in the UDP is solved ( issue #64) I created a sketch for bidirectional UDP Multicast.

It's partly working.

Similar to issue #64 the sketch receives the Multicast packet OK, but a send confirmation packet is not received by the client.

Perhaps I made an error in the sketch (I'm not a genius in network communication and the Multicast functions are not documented) , or could it be that there's a remaining bug in the Multicast library.

I opened a topic with details and the sketch here:
http://www.esp8266.com/viewtopic.php?f=29&t=2464&p=14358#p14358

Any solutions or suggestions?

igrr, 2015-04-13:
Gerard, from the screenshot posted on the forum I see you are using netcat to test UDP multicast. But netcat doesn't support multicast, i.e. it doesn't know that it needs to join an IGMP group. This explains why you are not receiving any packets on the host side.
There are some equivalents of netcat which do support multicast though (mnc, multicat), perhaps you could try them out?

gerardwr, 2015-04-13:
Hi Ivan,

Thanks for the hint, will try another client, and report back.

gerardwr, 2015-04-13:
Ivan,

As usual, you're right ;-)

I replaced nc by the application Packetsender (available for OSX, Linux, Windows, IOS, etc at packetsender.com) and the confirmation package sent by the ESP is received.

So you fixed the UDP bug for Multicast too !

I updated the topic, and this issue can be closed, right?

Thanks for your help.

igrr, 2015-04-13:
Great, nice to hear that! Thanks for testing this.

closed 2015-04-13 by igrr