esp8266/Arduino #69: compiler warning and system crash on using progmem sample sketch

mvdbro, 2015-04-13:
Sample sketch from arduino website:
============================
const char string_0[] PROGMEM = "String 0";   // "String 0" etc are strings to store - change to suit.
 const char string_1[] PROGMEM = "String 1";
 const char string_2[] PROGMEM = "String 2";
 const char string_3[] PROGMEM = "String 3";
 const char string_4[] PROGMEM = "String 4";
 const char string_5[] PROGMEM = "String 5";
// Then set up a table to refer to your strings.
 const char* const string_table[] PROGMEM = {string_0, string_1, string_2, string_3, string_4, string_5};

char buffer[30];    // make sure this is large enough for the largest string it must hold

void setup()
{
   Serial.begin(9600);
   while(!Serial);
   Serial.println("OK");
}

void loop()
{
   /* Using the string table in program memory requires the use of special functions to retrieve the data.
      The strcpy_P function copies a string from program space to a string in RAM ("buffer").
      Make sure your receiving string in RAM  is large enough to hold whatever
      you are retrieving from program space. */
   for (int i = 0; i < 6; i++)
   {
     strcpy_P(buffer, (char*)pgm_read_word(&(string_table[i]))); // Necessary casts and dereferencing, just copy.
     Serial.println(buffer);
     delay( 500 );
   }
}

Leads to this compiler warning:
=======================
In file included from C:\tools\arduino-1.6.1\hardware\esp8266com\esp8266\cores\esp8266/Arduino.h:30:0,
                 from ESPTestSample.ino:1:
ESPTestSample.ino: In function 'void loop()':
C:\tools\arduino-1.6.1\hardware\esp8266com\esp8266\cores\esp8266/pgmspace.h:33:63: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
 #define pgm_read_word(addr)   (*(const unsigned short *)(addr))
                                                               ^
C:\tools\arduino-1.6.1\hardware\esp8266com\esp8266\cores\esp8266/pgmspace.h:26:47: note: in definition of macro 'strcpy_P'
 #define strcpy_P(dest, src)   strcpy((dest), (src))
                                               ^
ESPTestSample.ino:33:30: note: in expansion of macro 'pgm_read_word'

Sketch uses 190,332 bytes (36%) of program storage space. Maximum is 524,288 bytes.

Leads to this system crash:
=====================
Fatal exception (28):
epc1=0x4000bf0e, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00008750, depc=0x00000000
del if0
bcn 0
del if1
usl
sul 0 0

Any ideas?

robertgregor, 2015-04-15:
I am using F() macro for this purpose, I don't have an issues, however, I am not sure, if it is really working like it should work...

rogerclarkmelbourne, 2015-04-15:
Its kind of dependant in the linker settings, but on ARM I dont think you need the progmem stuff, I think its AVR specific due to the memory architecture if the AVR

Just declaring it as const should be enough on ARM to not have it take up space in RAM, but Ivan can probably confirm this.

robertgregor, 2015-04-15:
So that's good to know, thanks! So it means, that the F() macro is there only for the compatibility purpose and it doesn't make any sense to use it then. Thanks for the info!

Links2004, 2015-04-15:
if you really force it in ROM and to the SPI flash you can use this

```cpp
#define F(str)  []() -> const char * { static const char tmp[] ICACHE_RODATA_ATTR = str; return &tmp[0]; }()
```

mvdbro, 2015-04-17:
I rewrote my sketch using "const char" without the progmem stuff. It's ok now!

closed 2015-04-17 by mvdbro