esp8266/Arduino #70: Some functions missing like strcasecmp, sprintf, dtostrf ?

mvdbro, 2015-04-13:
Using this function call:

strcasecmp

results in compiler error:

error: 'strcasecmp' was not declared in this scope

same issue for these functions:
sprintf
dtostrf
strtol
ltoa

chadouming, 2015-04-13:
have you imported stdio ? cause sprintf works fine here once stdio is imported.

mvdbro, 2015-04-13:
#include <stdio.,h> fixed the sprintf
but the others still have issues.
tried stdlib.h but didn't help.

rogerclarkmelbourne, 2015-04-15:
dtostrf is an AVR specific function.

Some files from the SAM core of the arduino project need to be added to the cores files

See https://github.com/arduino/Arduino/search?utf8=✓&q=dtostrf

Which shows which files need to be copied to the repo and how they get included

I'm not sure why normal c functions are missing e.g ltoa 

Did you try adding your own prototype for these?

Links2004, 2015-04-15:
the functions are implemented and working in the last git version.
to use the functions you need to so the include like this:
```cpp

extern "C" {

#include <stddef.h>
#include <stdarg.h>
#include <stdio.h>

}

```

rogerclarkmelbourne, 2015-04-15:
Can't this be added to one of the central includes e.g. Arduino.h ?

luc-github, 2015-04-17:
strtok neither strtok_r is found even adding 
```
extern "C" {

#include <stddef.h>
#include <stdarg.h>
#include <stdio.h>

}
```
I use the arduino-1.6.1-p1-windows.zip which a part this problem works well so far

mvdbro, 2015-04-17:
Some stuff is solved and found some workarounds for others.

But still can't use this one: error: 'strcasecmp' was not declared in this scope
strcmp is ok, strcasecmp is not ?
I guess these should be in the same library? Or it's just not implemented?

Links2004, 2015-04-17:
you shut update to the latest git version
only need to replace the ".\arduino-nightly\hardware\esp8266com\esp8266" directory.
with the directory from git:
https://github.com/esp8266/Arduino/tree/esp8266/hardware/esp8266com/esp8266

strtok_r is implemented:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/cores/esp8266/libc_replacements.c#L162
and strcasecmp to:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/cores/esp8266/libc_replacements.c#L204

luc-github, 2015-04-17:
Actually I have also used the standalone module of sandeep 0.0.3 which has the file and the modification and still strtok_r is not in scope -  I will update the directory like you suggest  and feedback later
Thanks for your help

luc-github, 2015-04-17:
seems the function is not visible, my test skech is compiling well with a mega board but failed with ESP8266 - 

__test1.ino: In function 'void loop()':
test1.ino:17:37: error: 'strtok_r' was not declared in this scope
Error compiling.__

I have did what was suggested download latest git version - I have the verified that I have the  hardware/esp8266com/esp8266/cores/esp8266/libc_replacements.c with strtok_r inside =>yes

```
extern "C" {
#include <stddef.h>
#include <stdarg.h>
#include <stdio.h>
}
void setup() {
  // put your setup code here, to run once:
}
void loop() {
  // put your main code here, to run repeatedly:
    char * b = strtok_r (NULL,".",NULL);
}
```

Did I missed something ? Thanks

Links2004, 2015-04-17:
strtok_r is part of the string.h you need to include it.

```cpp
extern "C" {
#include <string.h>
}
```

luc-github, 2015-04-17:
do not change the result - I have tested before actually and just redid now to confirm and still 
 ___'strtok_r' was not declared in this scope___
It is weird

Links2004, 2015-04-17:
strange if i try
```cpp

extern "C" {
#include <stddef.h>
#include <stdarg.h>
#include <stdio.h>
#include <string.h>
}
void setup() {
  // put your setup code here, to run once:
}
void loop() {
  // put your main code here, to run repeatedly:
    char * b = strtok_r (NULL,".",NULL);
}
``` 
it is compiling.

i add all includes to the Arduino.h and also implement the "strtok" hope this helps.

https://github.com/Links2004/Arduino/tree/esp8266/hardware/esp8266com/esp8266

luc-github, 2015-04-17:
Yes - I must do something wrong somewhere - I will clone the git to avoid to download the zip 

Thanks for your time

mvdbro, 2015-04-18:
I just do not get it to work with strcasecmp. Must have screwed up something somewhere...

Finally decided to include the function into my sketch:

int strcasecmp(const char * str1, const char * str2) {
    int d = 0;
    while(1) {
        int c1 = tolower(*str1++);
        int c2 = tolower(*str2++);
        if(((d = c1 - c2) != 0) || (c2 == '\0')) {
            break;
        }
    }
    return d;
}
At least i can continue and i managed to finish my entire project!
Thanks for this amazing piece of work!

closed 2015-04-18 by mvdbro