esp8266/Arduino #71: Error: section `.text' will not fit in region `iram1_0_seg'

marvinroger, 2015-04-13:
Hey,

I import some libs in my code:

```c
#include <EEPROM.h>
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <PubSubClient.h>
#include <ArduinoJson.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```
And I am getting the following error at compile-time:

```
Arduino : 1.6.1 (Windows 8.1), Carte : "Generic ESP8266 board"

c:/users/marvin/documents/esp8266-01/arduino-1.6.1 - esp8266/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Marvin\AppData\Local\Temp\build1509245431262001322.tmp/shutters.cpp.elf section `.text' will not fit in region `iram1_0_seg'

c:/users/marvin/documents/esp8266-01/arduino-1.6.1 - esp8266/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libc.a(lib_a-mallocr.o):(.literal+0x2c): undefined reference to `_sbrk_r'

c:/users/marvin/documents/esp8266-01/arduino-1.6.1 - esp8266/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libc.a(lib_a-mallocr.o): In function `malloc_extend_top':

d:\ivan\projects\arduinoesp\toolchain\dl\esp-newlib\build\xtensa-lx106-elf\newlib\libc\stdlib/../../../../../newlib/libc/stdlib/mallocr.c:2165: undefined reference to `_sbrk_r'

d:\ivan\projects\arduinoesp\toolchain\dl\esp-newlib\build\xtensa-lx106-elf\newlib\libc\stdlib/../../../../../newlib/libc/stdlib/mallocr.c:2202: undefined reference to `_sbrk_r'

c:/users/marvin/documents/esp8266-01/arduino-1.6.1 - esp8266/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libc.a(lib_a-freer.o): In function `_malloc_trim_r':

d:\ivan\projects\arduinoesp\toolchain\dl\esp-newlib\build\xtensa-lx106-elf\newlib\libc\stdlib/../../../../../newlib/libc/stdlib/mallocr.c:3325: undefined reference to `_sbrk_r'

d:\ivan\projects\arduinoesp\toolchain\dl\esp-newlib\build\xtensa-lx106-elf\newlib\libc\stdlib/../../../../../newlib/libc/stdlib/mallocr.c:3332: undefined reference to `_sbrk_r'

c:/users/marvin/documents/esp8266-01/arduino-1.6.1 - esp8266/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libc.a(lib_a-freer.o):d:\ivan\projects\arduinoesp\toolchain\dl\esp-newlib\build\xtensa-lx106-elf\newlib\libc\stdlib/../../../../../newlib/libc/stdlib/mallocr.c:3340: more undefined references to `_sbrk_r' follow

collect2.exe: error: ld returned 1 exit status
```

When I remove `#include <ArduinoJson.h>`, everything if fine. So is the ArduinoJson lib too heavy for the ESP8266? I mean, NodeMCU for example handles JSON fine, I don't see how a native small footprint lib would be too heavy...

BTW, thanks for this amazing project!

Links2004, 2015-04-15:
i use the ArduinoJson on the ESP8266.

do you use the latest git version?
the undefined reference to `_sbrk_r' error is fixed there.
may the space problem too.

marvinroger, 2015-04-16:
No I don't, I am using the latest Windows binary or the proxy from Adafruit for the board manager.
I will try the latest git ASAP - if I manage to build it - !

Links2004, 2015-04-16:
no need for build it, only replace the ".\arduino-nightly\hardware\esp8266com\esp8266" directory.
with the dir from git:
https://github.com/esp8266/Arduino/tree/esp8266/hardware/esp8266com/esp8266

marvinroger, 2015-04-16:
Well, thank you! It now works as expected. These ESPs impress me more than ever...

closed 2015-04-16 by marvinroger

burner-, 2015-05-28:
I try copy without build but I got new errors. Do someone have nightly build?

Links2004, 2015-05-28:
latest board manager version will work fine.

burner-, 2015-05-29:
Do you mean that 1.6.4-673-g8cd3697 from json? It does not work.

marvinroger, 2015-05-29:
It does work for me using latest board manager version!

burner-, 2015-05-29:
can you verify what version you mean