esp8266/Arduino #72: WiFiClient example needs a client.close()

cswiger, 2015-04-15:
This is easy - using the WiFiClient.ino example with thingspeak.com - my sensor would only post 4 readings then stop. After putting a client.stop(); right after Serial.println("closing connection"); it now works. 

// Read all the lines of the reply from server and print them to Serial
   while(client.available()){
      String line = client.readStringUntil('\r');
      Serial.print(line);
   }

   Serial.println(); 
   Serial.println("closing connection");
   client.stop();          <-- add 

}

igrr, 2015-04-21:
Did you make any other changes to the WiFiClient.ino?
I'm asking because ```client.stop()``` does the same thing that happens when ```client``` variable goes out of scope. Compare:
```
void ICACHE_FLASH_ATTR WiFiClient::stop() 
{
    if (!_client)
        return;

    _client->unref();
    _client = 0;
}
```
with
```
ICACHE_FLASH_ATTR WiFiClient::~WiFiClient()
{
    if (_client)
        _client->unref();
}
```

So the .stop() call should only be necessary if ```client``` is not a local variable which gets destroyed when the loop function returns.

cswiger, 2015-04-26:
Hmmm, your probably right. I added code to read temperature from a ds18b20 and would get exactly 4 data points then no more (using an esp-03). Added the client.stop() and it started working. My client is still created loop(). 

Just tried basic example and it also worked ok - so no problem then.

closed 2015-04-27 by igrr

xcompass, 2017-01-10:
I got the same problem with esp-12 module and nodemcu. Nothing changed from example except wifi ssid and password. I was only able to do a few queries and then it stopped working. I was able to ping the host but browse stuck on "`waiting for ....`". The last print in the console was "`new client`". So it stuck in the "`while(!client.available())`" loop with "`delay(1);`" running. Have to reset the chip for it to work again (only for a brief moment). 

After adding `client.stop()` after 
```
client.print(s);
delay(1);
```
The module works normally. I figured it out based on https://github.com/esp8266/Arduino/blob/master/doc/esp8266wifi/server-examples.md before I found this issue. In the example, it did close the client. So I guess we do need close statement. I would be happy to send a PR if necessary.

rezaajdarkosh, 2017-09-15:
Hi!
What is the difference between client.stop and client.close and client.flush?

szczukot, 2017-10-17:
I'm confirm. When i added client(stop) on the end of loop() - all working ok. Without thsi line server stopped after few seconds.
Why is this issue close ?

krlf, 2017-11-10:
It does not look like WiFiClient closes connection.
My code as the following:

void SendToNarodmon(......) {

    WiFiClient wifiClient;

    if (wifiClient.connect("narodmon.ru", 8283))
   {
       .....
   }
}

So, the WiFiClient destructor should be called.
I've collected pcap and it is seen that TCP connection is not closed.

192.168.1.172 - ESP
94.142.140.101 - Server.

"10","0.745110","192.168.1.172","94.142.140.101","TCP","58","32737  >  8283 [SYN] Seq=0 Win=5840 Len=0 MSS=1460"
"11","0.756207","94.142.140.101","192.168.1.172","TCP","58","8283  >  32737 [SYN, ACK] Seq=0 Ack=1 Win=14600 Len=0 MSS=1460"
"12","0.811933","192.168.1.172","94.142.140.101","TCP","54","32737  >  8283 [ACK] Seq=1 Ack=1 Win=5840 Len=0"
"13","0.812989","192.168.1.172","94.142.140.101","TCP","138","32737  >  8283 [PSH, ACK] Seq=1 Ack=1 Win=5840 Len=84"
"14","0.824357","94.142.140.101","192.168.1.172","TCP","54","8283  >  32737 [ACK] Seq=1 Ack=85 Win=14600 Len=0"
"15","0.996841","94.142.140.101","192.168.1.172","TCP","57","8283  >  32737 [PSH, ACK] Seq=1 Ack=85 Win=14600 Len=3"
"16","0.997020","94.142.140.101","192.168.1.172","TCP","54","8283  >  32737 [FIN, ACK] Seq=4 Ack=85 Win=14600 Len=0"
"17","1.267880","94.142.140.101","192.168.1.172","TCP","57","[TCP Retransmission] 8283  >  32737 [PSH, ACK] Seq=1 Ack=85 Win=14600 Len=3"
"18","1.807869","94.142.140.101","192.168.1.172","TCP","57","[TCP Retransmission] 8283  >  32737 [PSH, ACK] Seq=1 Ack=85 Win=14600 Len=3"
"19","2.887605","94.142.140.101","192.168.1.172","TCP","57","[TCP Retransmission] 8283  >  32737 [PSH, ACK] Seq=1 Ack=85 Win=14600 Len=3"
"20","5.047557","94.142.140.101","192.168.1.172","TCP","57","[TCP Retransmission] 8283  >  32737 [PSH, ACK] Seq=1 Ack=85 Win=14600 Len=3"

cswiger, 2026-10-18:
Just for giggle I asked Claude Opus 5.5 Medium to look at this thread and offer an opinion :) 

https://claude.ai/share/0aefc7ea-0b22-40e4-93ab-2f6eb0d10cd4

"unref() doesn’t close anything by itself. It only decrements a reference count on the underlying connection object (ClientContext). The TCP connection is torn down only when that count reaches zero and the close path actually runs. If something else still holds a reference, or the close isn’t triggered when the object is released, the connection stays allocated in lwIP, the TCP stack."