esp8266/Arduino #73: Issue with sprintf and float values

mvdbro, 2015-04-16:
Using this sketch::

#include <stdio.h>

void setup() {
  Serial.begin(19200);
}

void loop() {
  char string[40];

  string[0]=0; 
  sprintf(string,"Characters: %c %c", 'a',65);
  Serial.println(string);

  string[0]=0; 
  sprintf(string,"Decimals: %d %ld", 1977, 650000L);
  Serial.println(string);

  string[0]=0; 
  sprintf(string,"%s", "A string");
  Serial.println(string);

  string[0]=0; 
  sprintf(string,"floats: %4.2f %+.0e %E",3.1416, 3.1416, 3.1416);
  Serial.println(string);

  delay(1000);
}

Results in this output;

Characters: a A
Decimals: 1977 650000
A string
floats: %.2f %.0e %E

As you can see, the sprintf function is working for %d, %s, %c but not %f

Using latest github version, downloaded today (april 16th)

igrr, 2015-04-21:
Yep, sprintf (and related functions) from Espressif do not support floating-point formatting.
You can use Arduino String class as a workaround, is has support for floating-point values (although only fixed format IIRC).

mvdbro, 2015-04-21:
Ok that's clear. So it's not a bug, it is by 'design'
I already used some alternate code to solve my issue.
Thanks for the reply.

closed 2015-04-21 by mvdbro

joshglendenning, 2016-12-24:
I guess this would explain the "%!f(MISSING)" output I was getting. Took me forever to find this information, hopefully leaving that string here will improve the google results... At least I know now I'm not crazy.

jourjine, 2016-12-24:
best way to convert float to string but %f not worked.

char *dtostrf(double val, signed char width, unsigned char prec, char *s)  
char str_temp[6];

/* 4 minimum string length, 2 precise; converted value to  в str_temp*/
dtostrf(temp, 4, 2, str_temp);  
sprintf(temperature,"%s F", str_temp);

timw1971, 2017-03-19:
There is a simpler solution if you don't care about exponents in the result:

sprintf("%s", String(myFloat).c_str());

The default is 2 decimal places, but you can override it, e.g.: String(myFloat, 3); // 3 decimal places.

For width formatting etc, use the standard printf modifiers around %s.

milkpirate, 2017-05-29:
For other workarounds (with possibly smaller memory footprint, because of avoiding Strings) have a look here:

[http://yaab-arduino.blogspot.de/2015/12/how-to-sprintf-float-with-arduino.html](http://yaab-arduino.blogspot.de/2015/12/how-to-sprintf-float-with-arduino.html)

aarohlabs01, 2019-03-29:
plese help