esp8266/Arduino #74: Issue with UDP Multicast Client sketch

gerardwr, 2015-04-16:
Hi,

I created an UDP Multicast Server sketch, and that works OK. The sketch accepts UDP packets sent to the Multicast IP address, and sends a confirmation packet back to the UDP Client.
(ref. http://www.esp8266.com/viewtopic.php?f=29&t=2464 ).

Now I would like to create a sketch where the ESP is a client sending packets to an Multicast UDP server, but I am having difficulty understanding the exact use of the functions in WiFiUdp.h.

This code compiles and runs, but packets are not received by the Server. I checked the server, it receives UDP Multicast from another client perfectly.

[image]

Also tried this with the same result:

[image]

So I am confused now, give us a hint!

igrr, 2015-04-17:
```beginMulticast``` is analogous to ```begin``` — it start listening on a UDP port, plus it also joins the multicast group. So you don't need this one to send packets.

```endPacketMulticast``` is like ```endPacket```, except it doesn't use LwIP's ```udp_connect``` internally, but uses ```udp_sendto```. I don't recall why exactly such method was necessary, but in the hindsight I think I could have done it without introducing a new function (```endPacketMulticast```). Nevertheless, I suggest you try again with ```beginPacket``` + ```endPacketMulticast```.
If it doesn't work for some reason then there's likely a bug of some sort.

igrr, 2015-04-17:
Also I assume you are using multicast group address for ```ipMulti```?

gerardwr, 2015-04-17:
Tested as you suggested with `beginPacket` + `endPacketMulticast` using the code snippet below, to no avail. 

The Server never receives the packet, while it DOES accept packets from another Multicast client using the same Multicast IP.

A bug ?, or do you want me to test something else?

```
// Multicast declarations
IPAddress ipMulti(239, 0, 0, 57);
unsigned int portMulti = 12345;      // local port to listen on

void loop()
{
    Udp.beginPacket(ipMulti, portMulti);
    Udp.write("UDP packet received ");
    Udp.print(system_get_chip_id(),HEX);
    Udp.print(":");
    Udp.print(WiFi.localIP());
    Udp.endPacketMulticast(ipMulti, portMulti);
    Udp.stop();
    
    Serial.println("Multicast packet sent");
    delay(5000);
}
```

gerardwr, 2015-04-24:
@igrr 

Any suggestions Ivan?

igrr, 2015-04-25:
Reproduced, looking at this and #105 as well.

gerardwr, 2015-04-25:
OK, thanks for the feedback Ivan, good luck.

igrr, 2015-04-27:
The network stack needs to know which interface is used to send multicast,
so there's an lwIP function udp_set_multicast_netif_addr which needs to be
called for multicast transmission to work. I'm going to add an API to do
that from the sketch.

igrr, 2015-04-27:
Looks like this is going to fix mDNS as well.

gerardwr, 2015-04-27:
Some suggestions:

- Can the lwIP call be "hidden" in beginMulticast, thiw would prevent an additional API call.

- Perhaps the elimination of endpacketMulticast can be done in The fix,

igrr, 2015-04-27:
Yes, sure, I removed endPacketMulticast and added some arguments to
beginPacketMulticast instead.
Just need to test everything and will push the fix shortly.

igrr, 2015-04-27:
This should be fixed now for wifi station interface.

I haven't been able to make multicast work for softAP interface. For multicast destinations over softAP interface udp_send returns error code 1, which is not one of lwIP error codes (which are all negative), so it must be coming from the lower level (netif->output). Perhaps this issue should be raised to Espressif.

gerardwr, 2015-05-04:
@igrr 

Alas, the fix does not solve the issue, behavior is like before.

The sent multicast packet is not received by my server application, while multicast packets sent from another application (Packet Sender) to the same multicast IP+port are received OK.

Here's the complete sketch I use for testing.

```
/*

3-5-2015. Werkt niet. UDP Multicast packet niet ontvangen ( multicast.py / multicastlistener.c )

 */

#include <ESP8266WiFi.h>
#include <WiFiUDP.h>

int status = WL_IDLE_STATUS;
const char* ssid = "ssid";  //  your network SSID (name)
const char* pass = "pw";       // your network password

// A UDP instance to let us send and receive packets over UDP
WiFiUDP Udp;

// Multicast declarations
IPAddress ipMulti(239, 0, 0, 57);
unsigned int portMulti = 12345;      // local port to listen on

void setup()
{
  // Open serial communications and wait for port to open:
  Serial.begin(115200);

  // setting up Station AP
  WiFi.begin(ssid, pass);
  
  // Wait for connect to AP
  Serial.print("[Connecting]");
  Serial.print(ssid);
  int tries=0;
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    tries++;
    if (tries > 30){
      break;
    }
  }

  // print your IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);
}

void loop(){

  Serial.println("Sending UDP multicast packet");
  // send a reply, to the IP address and port that sent us the packet we received
  Udp.beginPacket(ipMulti, portMulti);
  Udp.write("UDP Multicast packet sent by ");
  Udp.println(WiFi.localIP());
  Udp.endPacket();  

  delay(3000);
}
```

gerardwr, 2015-05-10:
@igrr 

My issue is solved.

Today I installed the latest version and changed my example sketch above according the latest README.md.

I changed `Udp.beginPacket(ipMulti, portMulti);`
To  `Udp.beginPacketMulticast(ipMulti, portMulti,WiFi.localIP());`

Now the Multicast packet is received OK by my Multicast client (in this case a Python on my Mac).

Thx for another great fix.

closed 2015-05-19 by igrr

probonopd, 2015-06-27:
I can confirm that sending UDP multicast packets is working well when the ESP8266 is a client in the wireless network,

PyBerger, 2016-01-05:
Has any of you tested sending mutlicast when running in SoftAP mode ?

I have trouble receiving packets on android AP (only get very few packets : losing 80% of packets) while in unicast mode it works well (may loose 1% or even less) 

I use 2.0.0 and the following code:
```
IPAddress     softAPIP;
IPAddress     broadcastIP(192,168,4, 255);
IPAddress     clientIP(192,168,4,2);

WiFi.mode(WIFI_AP);
  WiFi.softAP(espSSID);
  debug_println("ESP8266 Wi-Fi init done!");
  debug_println("");
  softAPIP = WiFi.softAPIP();
 
// This doesn't work well : loosing a lot of packets (more than it should even if UDP has no guarantee  
udpInfo.beginPacketMulticast(broadcastIP, TCP_UDP_INFO_PORT, softAPIP);

// This works well !
 // udpInfo.beginPacket(clientIP, TCP_UDP_INFO_PORT);
  

udpInfo.write(wifi_buf, INFO_PACKET_SIZE);
  udpInfo.endPacket();

mkeyno, 2016-10-25:
@PyBerger  has it fix since new lib release?

Stephan35, 2016-11-03:
Hello,
I also try to send Udp broadcast in AP Mode .
Not working in this AP Mode, i have to call the ip i want to talk with.
If any idea , thanks.
Best regards.

Stephan35, 2016-11-06:
Find a way
I will try to explain 
Router : 192.168.0.1 Broadcast : 192.168.0.255
ESPxxxx1 STA : 192.168.0.10 (On router) AP : 192.168.4.1 
ESPxxxx2 STA: 192.168.0.11 (On router) AP: 192.168.4.1
ESPxxxx4 STA: 192.168.4.2 (On ESPxxxx1) AP 192.168.4.1

Reading UDP for all : 

`
void setup() {

...

  Udp.begin(localUdpPort); // 9999 ?  localUdpPort
  //Udp.beginMulticast(SoftAPIP, IPAddress(192,168,0,255), localUdpPort); 

  Serial.println ( "*** UDP server started" );
  Serial.printf("*** UDP : Now listening at IP %s, UDP port %d\n", WiFi.localIP().toString().c_str(), localUdpPort);
}
`

Sending UDP for all : 
`
void loop() {

// put all your handler
  // Scan des requetes udp
  if (chk_Upd_scan > chk_Upd_scan_time){
    chk_Upd_scan = 0;

    int packetSize = Udp.parsePacket();
    if (packetSize)
    {    
      int len = Udp.read(incomingPacket, 256);
      if (len > 0)
      {
        incomingPacket[len] = 0;
      }
Serial.println(incomingPacket);
  }
  }

  // Envoi du nom de node 1/3
  if (millis() - chk_node > chk_node_time){

    //Serial.print("*** UDP Send :");
    //(String("NEW") + "|" + Node_Name_Prefix + macID + "|" + Node_This + "|" + String(random(3000))).toCharArray(UdpMessage, 80);

    switch (chk_node_Sel){

      case 2:{
        Udp.beginPacketMulticast(IPAddress(192,168,0,255), localUdpPort, SoftAPIP);
        (String(">2> ") + Node_Name_Prefix + macID + " | BroadCastIP :" + "192.168.0.255" + " | SoftAPIP :" + DisplayIP(SoftAPIP) +" |" + String(random(3000))).toCharArray(UdpMessage, 256);
        break;
      }
      case 3:{
        Udp.beginPacketMulticast(IPAddress(192,168,4,255), localUdpPort, SoftAPIP);
        (String(">3> ") + Node_Name_Prefix + macID + " | BroadCastIP :" + DisplayIP(BroadCastIP) +" | BroadCastIP :" + DisplayIP(SoftAPIP) +" |" + String(random(3000))).toCharArray(UdpMessage, 256);
        break;
      }
    }

    Udp.write(UdpMessage);
    Udp.endPacket();
    //Serial.println(String("Sending : ") + UdpMessage);
    yield();
    ++chk_node_Sel;
    if(chk_node_Sel == 4){
      chk_node_Sel = 0;
    }
 
    
    chk_node = millis();
  }

}

`

This is working for me with AP , and , as you can see , i have to specify manually Broadcast.
Now, i don't really know how to get broadcast from the STA where your are connected from . Maybe get it by udp in the side STA to slave AP ... ?

prsolucoes, 2017-01-21:
Any changes? Here it appear one in 4632462734 tries.

R0binBl00d, 2017-09-21:
@igrr 

I'm still having some issues with the Multicast.
Hope someone has a solution for this:

My code looks something like this: ([concerning "Serial << ;"](https://github.com/kachok/arduino-libraries/tree/master/Streaming))

```  
#include <ESP8266WiFi.h>
#include <WiFiUDP.h>
...
  Serial.println("");
  Serial.println("Connecting to UDP Multicast");

  IPAddress rem = CreateIpFromString(multiCastIp);
  IPAddress loc = CreateIpFromString(localIp);

  int result = UdpMulti.beginMulticast(loc, rem, localPortM);
  Serial << "Multicast RemoteIP: " << rem << "; Port: " << localPortM << "; LocalIp: " << loc << endl;
  if(result == 1)
  {
    Serial.println("beginMulticast - Connection successful");
    state = true;
  }
  else
  {
    Serial.println("beginMulticast - Connection failed");
  }
  UdpMulti.beginPacketMulticast(rem, remoteport, loc);
  UdpMulti.write(data);
  UdpMulti.endPacket();
  Serial << "UPD/IPm:\"" << rem << ":" << remoteport << ";" << data << "\"" << endl;
```
which creates following output on the SerialPort:
```
Connecting to UDP Multicast
Multicast RemoteIP: 239.192.0.7; Port: 59998; LocalIp: 172.16.100.192
beginMulticast - Connection successful
UPD/IPm:"239.192.0.7:60007;testdata"
```
unfortunatly Wireshark on the Windows PC is not able to pick it up :-(
I cannot see the packages comming in.

I changed the IPAddress via Serial and got following Output
(**"Connection failed!!!"**)
```
Connecting to UDP Multicast
Multicast RemoteIP: 172.16.255.255; Port: 59998; LocalIp: 172.16.100.192
beginMulticast - Connection failed
UPD/IPm:"172.16.255.255:60007;testdata"
```
Well, Wireshark says something else, the connection is just fine and I received the package on my Windows PC.
```
Frame 5368: 50 bytes on wire (400 bits), 50 bytes captured (400 bits) on interface 0
    Interface id: 0 (\Device\NPF_{672270F9-53EA-4FEC-8F51-41F840AEED43})
    Encapsulation type: Ethernet (1)
    Arrival Time: Sep 21, 2017 12:44:04.572155000 W. Europe Daylight Time
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1505990644.572155000 seconds
    [Time delta from previous captured frame: 3.050718000 seconds]
    [Time delta from previous displayed frame: 3.050718000 seconds]
    [Time since reference or first frame: 4979.351915000 seconds]
    Frame Number: 5368
    Frame Length: 50 bytes (400 bits)
    Capture Length: 50 bytes (400 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: eth:ethertype:ip:udp:data]
    [Coloring Rule Name: UDP]
    [Coloring Rule String: udp]
Ethernet II, Src: Espressi_19:b3:17 (a0:20:a6:19:b3:17), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
    Destination: Broadcast (ff:ff:ff:ff:ff:ff)
    Source: Espressi_19:b3:17 (a0:20:a6:19:b3:17)
    Type: IPv4 (0x0800)
Internet Protocol Version 4, Src: 172.16.100.192, Dst: 172.16.255.255
    0100 .... = Version: 4
    .... 0101 = Header Length: 20 bytes (5)
    Differentiated Services Field: 0x00 (DSCP: CS0, ECN: Not-ECT)
    Total Length: 36
    Identification: 0x0a3d (2621)
    Flags: 0x00
    Fragment offset: 0
    Time to live: 255
    Protocol: UDP (17)
    Header checksum: 0xf4aa [validation disabled]
    [Header checksum status: Unverified]
    Source: 172.16.100.192
    Destination: 172.16.255.255
    [Source GeoIP: Unknown]
    [Destination GeoIP: Unknown]
User Datagram Protocol, Src Port: 4097, Dst Port: 60007
    Source Port: 4097
    Destination Port: 60007
    Length: 16
    Checksum: 0x87e7 [unverified]
    [Checksum Status: Unverified]
    [Stream index: 545]
Data (8 bytes)
    Data: 7465737464617461
    [Length: 8]
```
Not sure what I am supposed to do. 
Since this is a broadcast and not the Multicast I was expecting
Would be great if someone has a solution for this problem.

Thanks and best Regards
Collin

commit 99510e31 (2015-04-27): UDP multicast APIs fix

fix #74, fix #7
commit 3665c874 (2015-04-28): UDP multicast APIs fix

fix #74, fix #7