esp8266/Arduino #76: Change AP static IP configuration

luc-github, 2015-04-17:
Seems there is no function to change AP IP configuration, currently setting IP/GW/Subnet works only for Station.
I have modified ESP8266Wifi.cpp/.h to add the missing function and it works so far, it would be good to add it I think.

```
void ESP8266WiFiClass::configAP(IPAddress local_ip, IPAddress gateway, IPAddress subnet)
{
    struct ip_info info;
    info.ip.addr = static_cast<uint32_t>(local_ip);
    info.gw.addr = static_cast<uint32_t>(gateway);
    info.netmask.addr = static_cast<uint32_t>(subnet);
    wifi_softap_dhcps_stop();
    wifi_set_ip_info(SOFTAP_IF, &info);
    wifi_softap_dhcps_start();
}
```

reaper7, 2015-04-21:
works well!
This is very usefull

igrr, 2015-04-21:
Thanks!
Merged with a minor naming change (configAP -> softAPConfig) to be consistent with other functions.

closed 2015-04-21 by igrr

JanJansen47, 2016-02-25:
From the network point of view there appeared also to be an issue with the lease time of my (TP-Link) router. Standard IP address lease time is 120 minutes.  In case an ESP8266 request the same fixed IP address within these 120 minutes its "refused" and the DHCP return an "ad-hoc" ip address.  So I changed the lease time to one (1) minute.

khalilst, 2016-06-19:
Sorry, i'm a newbie to this field. Could you please write me an example for calling of softAPConfig? I have no idea how to pass the arguments.
Thanks in advance

joemxwolf, 2017-03-19:
what at commands is used for changing ip address of esp8266 (cwmode=3) so that every time the esp connects to the router the router will give static ip for the esp? like having a constant ip for the esp given by the router? big thnx.

commit 307ea873 (2015-04-21): Add method to configure softAP IP address

fix #76