esp8266/Arduino #78: Add Support for nodemcu

rei-vilo, 2015-04-17:
The [nodemcu](http://nodemcu.com/index_en.html) is an ESP8266-based 30-pin DIP form-factor board with USB connector and programmer. By default, it comes with Lua but I'd like to use the Wiring / Arduino framework.

On Mac OS X, it requires the installation of the driver [CH341SER_MAC_ZIP](http://www.wch.cn/download/CH341SER_MAC_ZIP.html) and then launch on Terminal `sudo nvram boot-args="kext-dev-mode=1"`.

Thanks!

gerardwr, 2015-04-17:
I use the Arduino environment from this github on my Mac. My ESP-01 is connected to my Mac via a PL2303 or FTDI board. For the USB-to-to-serial boards you will have to install the Mac OS drivers separately, this is not part of the supplied environment.

Or do I misunderstand?

rei-vilo, 2015-04-17:
We're tlaking about different products.

I'm using a FTDI board for a plain ESP8266-01 and everything goes fine. 

But the noceMCU board has its own programming chip, CH341, and comes pre-loaded with Lua.

sej7278, 2015-04-17:
I believe that the Arduino firmware uses/requires the AT firmware from the official SDK and will overwrite the nodemcu firmware.

Drivers are irrelevant, that's down to the user (and aren't even required on Linux).

If the nodemcu board cannot run non-nodemcu firmware then you should report that as a bug to them.

prasertsakd, 2015-04-17:
i use nodemcu  it different form esp8266 + usb2serial

onboard can auto reset i will ready  experiment on that

only config auto upload key on board.txt form none to ck it work

http://www.ayarafun.com/2015/04/arduino-ide-for-esp8266/

combs, 2015-04-17:
ch341 is a serial to USB adapter chipset, not a programmer, if that helps explain... no special support needed for it.

AllAboutEE, 2015-04-19:
@rei-vilo 

Flashing an Arduino sketch to the NodeMcu board does work, I think you are just getting confused because you think that flashing an Arduino sketch is the same thing as uploading a NodeMcu .lua program.

Uploading an Arduino sketch to the ESP8266 is actually equivalent to flashing a different version of the NodeMcu firmware, NOT uploading a .lua program.

So, to upload an Arduino sketch you need to use the "Flash" button on the board not the "User" button. Once you have uploaded the sketch then yes, use the "User" button to run it.

rei-vilo, 2015-04-19:
Thank you for the answers and pointers. 

Installing `esptool.py` solved my problem ([download](https://github.com/themadinventor/esptool)). It works fine on Mac OS X where `esptools` doesn't work.

Please refer to #3 Got an error when flashing.

Adding  `esptool.py` as described by @sticilface would be a nice addition to Arduino for ESP8266.

So I close this ticket.

closed 2015-04-19 by rei-vilo

hallard, 2015-04-26:
Hi Guys

Just had some headaches on my new NodeMcu, first I had a bad USB cable (which was working fine on Leonardo board), and the 2nd problem is that I was not able to upload any sketch until I realized looking at the board schematics that the 2nd board button named user (1st is named flash) was not a reset one.
Then I succeeded to upload , but with pain because I needed to unplug/plug it back (to reset the board) while pressing flash, and immediately launch upload in IDE at the same time whooo.

Schematic show me that there is an autoreset feature so I should be able to use it. Sorry @prasertsakd just saw your answer now (should I have won time looking here before), but the idea is the same than the one in your anwser, yes !!
What I've done (without doing any change on esptool.py or esptool.exe), I just added NodeMCU board on boards.txt as follow and it works fine on my win 8.1 x64 (same as ESP8266 board but with ck on resetmethod).

```
##############################################################
nodemcu.name=NodeMCU board

nodemcu.upload.tool=esptool
nodemcu.upload.speed=115200
nodemcu.upload.resetmethod=ck
nodemcu.upload.maximum_size=524288
nodemcu.upload.wait_for_upload_port=true

nodemcu.build.mcu=esp8266
nodemcu.build.f_cpu=80000000L
nodemcu.build.board=ESP8266_ESP01
nodemcu.build.core=esp8266
nodemcu.build.variant=esp01
```

If this board could be added to the repo it could be very cool

rei-vilo, 2015-05-02:
Could `esptools.py` be included in the distribution package?

vicriedemann, 2016-03-21:
i changed the default baudrate to 230400 and it worked quite fine. With the 0x40 and 0x180 modification made earlier / i'm using FTDI /YP05) , with a powersupply aside.

ADeeds, 2016-03-26:
@vicriedemann, thanks so much. Wasn't able to get work with my ESP8266 on OSX, even after trying esptools.py, but lowering the lowering the baudrate from what works on Windows (921600 -> 230400) did the trick

vicriedemann, 2016-03-27:
Good to know! Maybe that baudrate 230400 is the one.

Are tou using Lua too? I was able to upload too using luatool at 230400
(default) but i had to make luatool.py to chmod something and worked well,
but i cant - right now- to set the code to start at booting