esp8266/Arduino #81: available RAM check?

mvdbro, 2015-04-18:
On atmega based boards, we were used to retrieve available memory like this:

uint8_t *heapptr, *stackptr;

unsigned long FreeMem(void)
  {
  stackptr = (uint8_t *)malloc(4);        // use stackptr temporarily
  heapptr = stackptr;                     // save value of heap pointer
  free(stackptr);                         // free up the memory again (sets stackptr to 0)
  stackptr =  (uint8_t *)(SP);            // save value of stack pointer
  return (stackptr-heapptr);
  }

It does not compile for  ESP. Anyone that knows how to do it?

On NodeMCU there is Node.Heap()

Links2004, 2015-04-18:
you can use
```cpp
extern "C" {
#include "user_interface.h"
}
uint32_t free = system_get_free_heap_size();
```

igrr, 2015-04-18:
Perhaps we should add this heap() method to the ESP class.

mvdbro, 2015-04-18:
Thanks! Got it to work. Nice to see that we have a lot more mem than with the old Atmega2560...

closed 2015-04-18 by mvdbro