esp8266/Arduino #86: Error when compiling sketch (strtok_r)

marcoschwartz, 2015-04-19:
Hello,

Since the last commit (ec4f53e70508018a1ec644333f4a8045db7eb29f) I have an error with a sketch that was working before:

/Users/marco/Downloads/Arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/libc_replacements.c: At top level:
/Users/marco/Downloads/Arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/libc_replacements.c:166:25: error: conflicting types for 'strtok_r'
 char* ICACHE_FLASH_ATTR strtok_r(char * str, const char * delimiters, char ** temp) {
                         ^
/Users/marco/Downloads/Arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/libc_replacements.c:163:12: note: previous implicit declaration of 'strtok_r' was here
     return strtok_r(str, delimiters, NULL);
            ^
Error compiling.

Any idea about this issue? Thanks

Links2004, 2015-04-19:
can you test the fix above?

marcoschwartz, 2015-04-19:
I just tried the fix, same error.

Links2004, 2015-04-19:
can you provide an example, I don't get the error on reproduced. 
hint use ``` ```cpp 
``` ```  for code.

marcoschwartz, 2015-04-19:
I get the error with my aREST library that implements a REST interface for the chip. You can find the example here:

https://github.com/marcoschwartz/aREST/tree/master/examples/ESP8266

Links2004, 2015-04-19:
is working fine at.
only one error 
```
../ESP8266_aREST.ino:42:23: error: 'ledControl' was not declared in this scope
   rest.function("led",ledControl);
```
i command it out and than it compile.

i only wondering why the implementation is in the .h file and there is no .cpp file for the library?

gregslomin, 2015-04-20:
After pulling the latest, running ant clean and ant dist I get the same error on the mDNS example sketch.

valeros, 2015-04-20:
I have the same error as marcoschwartz:
.pioenvs\esp01_8266\FrameworkArduino\libc_replacements.c: In function 'strtok':
.pioenvs\esp01_8266\FrameworkArduino\libc_replacements.c:164:5: warning: return makes pointer from integer without a cast [enabled by default]
return strtok_r(str, delimiters, NULL);
^
.pioenvs\esp01_8266\FrameworkArduino\libc_replacements.c: At top level:
.pioenvs\esp01_8266\FrameworkArduino\libc_replacements.c:167:25: error: conflicting types for 'strtok_r'
char* ICACHE_FLASH_ATTR strtok_r(char * str, const char * delimiters, char ** temp) {
^
.pioenvs\esp01_8266\FrameworkArduino\libc_replacements.c:164:12: note: previous implicit declaration of 'strtok_r' was here
return strtok_r(str, delimiters, NULL);

jeffryr, 2015-04-20:
I also have the same error as marcoschwartz after doing a clean build.  The gregslomin fix works for me.

marcoschwartz, 2015-04-20:
Just tried again with my aREST library + the gregslomin fix, now it works for me as well.

igrr, 2015-04-21:
merged the fix in 2edc522, thanks!

closed 2015-04-21 by igrr