esp8266/Arduino #88: section `.text' will not fit in region `iram1_0_seg'

einsteinx2, 2015-04-19:
I'm trying to adapt https://github.com/cnlohr/ws2812esp8266 for use with the Arduino IDE, but I keep running into this error: section `.text' will not fit in region `iram1_0_seg' 

I'm using the latest code from this repo compiled myself, so I should have the latest fixes.

igrr, 2015-04-21:
Can you post your work-in-progress code somewhere? I suspect the reason is that the code from .cpp files is automatically placed into .irom0.text (see linker script), whereas code from .c files does not.
So if you are using cnlohr's code as is you are likely filling up the .iram.text section.

einsteinx2, 2015-04-21:
I'll post the code when I get home tonight.

einsteinx2, 2015-04-22:
Turned out that I needed to add ICACHE_FLASH_ATTR to all of the functions to move them out of region.

closed 2015-04-22 by einsteinx2

sticilface, 2015-05-04:
Thought id give compiling the u8glib now that the iram has been increased.  Guess what.. this error has gone.  Only to be replaced by a new one...  error below... 

```````

Build options changed, rebuilding all
/Users/amelvin/Documents/Arduino/libraries/U8glib/utility/u8g_com_arduino_fast_parallel.c: In function 'u8g_com_arduino_fast_parallel_init':
/Users/amelvin/Documents/Arduino/libraries/U8glib/utility/u8g_com_arduino_fast_parallel.c:104:20: warning: assignment from incompatible pointer type [enabled by default]
   u8g_data_port[0] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D0]));
                    ^
/Users/amelvin/Documents/Arduino/libraries/U8glib/utility/u8g_com_arduino_fast_parallel.c:106:20: warning: assignment from incompatible pointer type [enabled by default]
   u8g_data_port[1] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D1]));
                    ^
/Users/amelvin/Documents/Arduino/libraries/U8glib/utility/u8g_com_arduino_fast_parallel.c:108:20: warning: assignment from incompatible pointer type [enabled by default]
   u8g_data_port[2] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D2]));
                    ^
/Users/amelvin/Documents/Arduino/libraries/U8glib/utility/u8g_com_arduino_fast_parallel.c:110:20: warning: assignment from incompatible pointer type [enabled by default]
   u8g_data_port[3] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D3]));
                    ^
/Users/amelvin/Documents/Arduino/libraries/U8glib/utility/u8g_com_arduino_fast_parallel.c:113:20: warning: assignment from incompatible pointer type [enabled by default]
   u8g_data_port[4] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D4]));
                    ^
/Users/amelvin/Documents/Arduino/libraries/U8glib/utility/u8g_com_arduino_fast_parallel.c:115:20: warning: assignment from incompatible pointer type [enabled by default]
   u8g_data_port[5] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D5]));
                    ^
/Users/amelvin/Documents/Arduino/libraries/U8glib/utility/u8g_com_arduino_fast_parallel.c:117:20: warning: assignment from incompatible pointer type [enabled by default]
   u8g_data_port[6] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D6]));
                    ^
/Users/amelvin/Documents/Arduino/libraries/U8glib/utility/u8g_com_arduino_fast_parallel.c:119:20: warning: assignment from incompatible pointer type [enabled by default]
   u8g_data_port[7] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D7]));
                    ^
/Users/amelvin/Documents/Arduino/libraries/U8glib/utility/u8g_com_arduino_no_en_parallel.c: In function 'u8g_com_arduino_no_en_parallel_init':
/Users/amelvin/Documents/Arduino/libraries/U8glib/utility/u8g_com_arduino_no_en_parallel.c:87:20: warning: assignment from incompatible pointer type [enabled by default]
   u8g_data_port[0] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D0]));
                    ^
/Users/amelvin/Documents/Arduino/libraries/U8glib/utility/u8g_com_arduino_no_en_parallel.c:89:20: warning: assignment from incompatible pointer type [enabled by default]
   u8g_data_port[1] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D1]));
                    ^
/Users/amelvin/Documents/Arduino/libraries/U8glib/utility/u8g_com_arduino_no_en_parallel.c:91:20: warning: assignment from incompatible pointer type [enabled by default]
   u8g_data_port[2] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D2]));
                    ^
/Users/amelvin/Documents/Arduino/libraries/U8glib/utility/u8g_com_arduino_no_en_parallel.c:93:20: warning: assignment from incompatible pointer type [enabled by default]
   u8g_data_port[3] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D3]));
                    ^
/Users/amelvin/Documents/Arduino/libraries/U8glib/utility/u8g_com_arduino_no_en_parallel.c:96:20: warning: assignment from incompatible pointer type [enabled by default]
   u8g_data_port[4] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D4]));
                    ^
/Users/amelvin/Documents/Arduino/libraries/U8glib/utility/u8g_com_arduino_no_en_parallel.c:98:20: warning: assignment from incompatible pointer type [enabled by default]
   u8g_data_port[5] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D5]));
                    ^
/Users/amelvin/Documents/Arduino/libraries/U8glib/utility/u8g_com_arduino_no_en_parallel.c:100:20: warning: assignment from incompatible pointer type [enabled by default]
   u8g_data_port[6] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D6]));
                    ^
/Users/amelvin/Documents/Arduino/libraries/U8glib/utility/u8g_com_arduino_no_en_parallel.c:102:20: warning: assignment from incompatible pointer type [enabled by default]
   u8g_data_port[7] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D7]));
                    ^
/Users/amelvin/Documents/Arduino/libraries/U8glib/utility/u8g_com_arduino_t6963.c: In function 'u8g_com_arduino_t6963_init':
/Users/amelvin/Documents/Arduino/libraries/U8glib/utility/u8g_com_arduino_t6963.c:92:27: warning: assignment from incompatible pointer type [enabled by default]
   u8g_output_data_port[0] =  portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D0]));
                           ^
[... cut ...]
/Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_si2c.c:165:7: note: in expansion of macro 'SCL_LOW'
       SCL_LOW();
       ^
/Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_si2c.c:32:22: error: 'GPEC' undeclared (first use in this function)
 #define SCL_HIGH()  (GPEC = (1 << twi_scl)) 
                      ^
/Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_si2c.c:167:7: note: in expansion of macro 'SCL_HIGH'
       SCL_HIGH();
       ^
Error compiling.
```````

probonopd, 2015-06-21:
Nodemcu has u8glib running on ESP8266; maybe it can be ported to ESP8266/Arduino?
https://github.com/nodemcu/nodemcu-firmware/tree/master/app/u8glib

tprochazka, 2015-10-23:
I tried Adafruit_SSD1331 ad found similar problem with portOutputRegister() function

```
Adafruit_SSD1331\Adafruit_SSD1331.cpp:307:15: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment
     csportreg = portOutputRegister(digitalPinToPort(_cs));
               ^
```

It is some common problem or ESP8266 ex port?

Links2004, 2015-10-23:
the ESP is a 32Bit system the pointer can not match the 8Bit of the AVR-
the fastest way for GPIO changes on ESP is the ```GPOS``` and ```GPOC``` registers.
take a look here:
https://github.com/Links2004/Adafruit_ILI9341/blob/master/Adafruit_ILI9341.cpp#L73-L75
https://github.com/Links2004/Adafruit_ILI9341/blob/master/Adafruit_ILI9341.cpp#L159-L171
https://github.com/Links2004/Adafruit_ILI9341/blob/master/Adafruit_ILI9341.h#L213

may this verion of SSD1331 is working
https://github.com/me-no-dev/Adafruit_SSD1306

marvinroger, 2016-03-11:
Sorry to comment on an old issue, but @igrr you said:

> the code from .cpp files is automatically placed into .irom0.text (see linker script), whereas code from .c files does not.

Is this still a true statement? As a library developer I develop in C++, so do I have to add `ICACHE_FLASH_ATTR`?

igrr, 2016-03-11:
That's not true now. This has been fixed somewhere after 2.0.0 i think, so now all code goes into flash by default. You can use ICACHE_RAM_ATTR for functions which have to be in RAM (like interrupt handlers).

marvinroger, 2016-03-11:
So [all these occurences](https://github.com/esp8266/Arduino/search?utf8=%E2%9C%93&q=ICACHE_FLASH_ATTR&type=Code) are redundant? Thanks for your answer!

igrr, 2016-03-11:
Not all of them. Some come from lwip header files, and those are irrelevant. Some files (like umm_malloc.c) are placed into IRAM by linker script, so there ICACHE_FLASH_ATTR works as an override. Some are redundant and will be cleaned up.