esp8266/Arduino #90: warning: espcomm cmd: wrong direction/command

julianh2o, 2015-04-20:
OS: OSX 10.10.3
ESP Firmware: 0018000902-AI03
Arduino IDE Version(s): Tried with 1.6.1, 1.6.3, and the 1.6.4 nightly

Have reset and GPIO0 pins connected to buttons that drop them to ground. Have tried both with and without pull-up resistors on those pins. I've tried programming sequences of holding GPIO0 LOW, pressing restart, and then uploading as well as holding GPIO0 LOW, pressing restart, releasing GPIO0 and then uploading.

GPIO2 is left to float. "Generic ESP8266 Board" confirmed to be selected, as is "esptool" under programmer on the firmware versions that have it.

Of note, attempts to flash a new version of the firmware have also been failing using esptool.py with the error "Failed to connect"

Any help is greatly appreciated! It's quite possible that there's something very obvious that I'm missing, though I've checked many of the usual suspects.

P.S. Powering via a 3.0v regulator connected to the 5v of the USB->TTL FTDI.

Error log: 

    Sketch uses 174,194 bytes (33%) of program storage space. Maximum is 524,288 bytes.
    /Users/julian/Downloads/Arduino161latest.app/Contents/Java/hardware/tools/esp8266/esptool -vv -cd none -cb 115200 -cp /dev/tty.usbserial-A7027CY5 -ca 0x00000 -cf /var/folders/st/h452tm5d6rq99tf8bpqgz0hr0000gn/T/build4384587546977833334.tmp/espblink.cpp_00000.bin -ca 0x40000 -cf /var/folders/st/h452tm5d6rq99tf8bpqgz0hr0000gn/T/build4384587546977833334.tmp/espblink.cpp_40000.bin 
    esptool v0.4.2 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
        setting board to none
        setting baudrate from 115200 to 115200
        setting port from /dev/tty.usbserial to /dev/tty.usbserial-A7027CY5
        setting address from 0x00000000 to 0x00000000
        espcomm_upload_file
        stat /var/folders/st/h452tm5d6rq99tf8bpqgz0hr0000gn/T/build4384587546977833334.tmp/espblink.cpp_00000.bin success
    opening port /dev/tty.usbserial-A7027CY5 at 115200
        tcgetattr
        tcsetattr
        serial open
    opening bootloader
    resetting board
    trying to connect
        setting character timeout 0
        done
        setting character timeout 1
        done
        espcomm_cmd: sending command header
        espcomm_cmd: sending command payload
        espcomm cmd: receiving 36 bytes of data
    warning: espcomm cmd: wrong direction/command: 0x00 0x08, expected 0x01 0x08
    trying to connect
        setting character timeout 0
        done
        setting character timeout 1
        done
        espcomm_cmd: sending command header
        espcomm_cmd: sending command payload
        espcomm cmd: receiving 36 bytes of data
    warning: espcomm cmd: wrong direction/command: 0x00 0x08, expected 0x01 0x08
    trying to connect
        setting character timeout 0
        done
        setting character timeout 1
        done
        espcomm_cmd: sending command header
        espcomm_cmd: sending command payload
        espcomm cmd: receiving 36 bytes of data
    warning: espcomm cmd: wrong direction/command: 0x00 0x08, expected 0x01 0x08
    resetting board
    trying to connect
        setting character timeout 0
        done
        setting character timeout 1
        done
        espcomm_cmd: sending command header
        espcomm_cmd: sending command payload
        espcomm cmd: receiving 36 bytes of data
    warning: espcomm cmd: wrong direction/command: 0x00 0x08, expected 0x01 0x08
    trying to connect
        setting character timeout 0
        done
        setting character timeout 1
        done
        espcomm_cmd: sending command header
        espcomm_cmd: sending command payload
        espcomm cmd: receiving 36 bytes of data
    warning: espcomm cmd: wrong direction/command: 0x00 0x08, expected 0x01 0x08
    trying to connect
        setting character timeout 0
        done
        setting character timeout 1
        done
        espcomm_cmd: sending command header
        espcomm_cmd: sending command payload
        espcomm cmd: receiving 36 bytes of data
    warning: espcomm cmd: wrong direction/command: 0x00 0x08, expected 0x01 0x08
    resetting board
    trying to connect
        setting character timeout 0
        done
        setting character timeout 1
        done
        espcomm_cmd: sending command header
        espcomm_cmd: sending command payload
        espcomm cmd: receiving 36 bytes of data
    warning: espcomm cmd: wrong direction/command: 0x00 0x08, expected 0x01 0x08
    trying to connect
        setting character timeout 0
        done
        setting character timeout 1
        done
        espcomm_cmd: sending command header
        espcomm_cmd: sending command payload
        espcomm cmd: receiving 36 bytes of data
    warning: espcomm cmd: wrong direction/command: 0x00 0x08, expected 0x01 0x08
    trying to connect
        setting character timeout 0
        done
        setting character timeout 1
        done
        espcomm_cmd: sending command header
        espcomm_cmd: sending command payload
        espcomm cmd: receiving 36 bytes of data
    warning: espcomm cmd: wrong direction/command: 0x00 0x08, expected 0x01 0x08
    warning: espcomm_sync failed
    error: espcomm_open failed

gerardwr, 2015-04-20:
You seem to have everything right .

I assume you sourced your power regulator directly from the 5V of the USB connector of the FTDI board, right?

sticilface, 2015-04-20:
This is what I get, try using the python version to flash it. 

Try my solution listed here 

https://github.com/esp8266/Arduino/issues/3

if you are compiling your own version from here, they've changed it a little and you have to select upload using programmer (after selecting pyesptool) from the arduino menu.  OR... change 

generic.upload.tool=esptool

to 

generic.upload.tool=pyesptool

in boards.txt

gerardwr, 2015-04-20:
Other  maybe obvious suggestions:
- CH-PD connected to Vcc
- GPIO15 connected to GND (for ESP07/12)
- FTDI-GND - ESP-GND - Power-GND

`holding GPIO0 LOW, pressing restart, releasing GPIO0 and then uploading` works for me

What output do you get on the serial port when resetting?

julianh2o, 2015-04-20:
I assume you sourced your power regulator directly from the 5V of the USB connector of the FTDI board, right?
Yep

I'll double check these other things and then follow the solution you linked and post back with my findings, thanks!

duncan-a, 2015-04-20:
What regulator are you using? Not all are capable of providing 3.3V from 5V
(which is not always 5V anyway).

I'm not sure if there any current spikes during programming but operation
can be unreliable when powering from USB. I always use a 5V 'wall wart'
power adaptor and an AMS117-based 3.3V regulator with an additional 22MFd
capacitor on the power rails close to the ESP8266.

The other thing to consider is that the ESP8266 is a 3.3V device and *that
applies to the logic levels as well as the power rail* - if you've got 5V
as the Vcc from the USB-FTDI then it's most likely 5V logic and that could
well have blown the ESP's input...

julianh2o, 2015-04-20:
I'm using an LM317 hooked up directly to the FTDI's 5v pin. I've got a scope on the output voltage of the power supply and I'm getting a pretty consistent 3v. Furthermore, interacting with the ESP via the serial monitor is working just fine, so I very much doubt that the problem lies in the power coming in on the FTDI's TX line.

julianh2o, 2015-04-20:
Apparently this is what I get for tinkering at 3am.. Turns out the problem was that my flash button (DTR) was hooked up incorrectly! Thanks for the help guys!

closed 2015-04-20 by julianh2o

zenmanenergy, 2015-05-13:
I get this error ALL the time. But not every time. If I unplug and plug my ftdi *sometimes* it goes away, but not always. Are you suggesting it's a power issue?

Mayoogh, 2018-01-17:
I had similar issue 
When I check the Preferences Tab in the Arduino IDE the additional board manager was blank

http://arduino.esp8266.com/stable/package_esp8266com_index.json

Adding this URL solved the error for my case
[image]
[image]