esp8266/Arduino #91: Incompatibility in pin assignments between esptool and esp8266/Arduino

julianh2o, 2015-04-21:
This is also posted on esptool: https://github.com/themadinventor/esptool/issues/29

I've seen in a few places people complaining about their setup working with esptool but not with the esp8266 Arduino IDE project (https://github.com/esp8266/Arduino) and was having this problem myself, but I think I've found why that is...

**When flashing with esptool the setup sequence looks like this:**
RTS goes LOW
DTR goes LOW
RTS goes HIGH
DELAY
DTR goes HIGH

which means that your wiring looks like this...
RTS -> CH_PD (i.e reset)
DTR -> GPIO0

**When flashing with the Arduino IDE**
RTS and DTR go LOW
DTR goes HIGH
DELAY
RTS goes HIGH

which means that your wiring looks like this...
RTS -> GPIO0
DTR -> CH_PD (i.e reset)

It seems to me that these pins are sorta arbitrarily chosen, but it would be better for everyone if we standardized on the same pin assignments early on. I'm going to post this issue to the Arduino IDE plugin as well, but I guess my suggestion would be that the swap in the esptool code.. Perhaps a --legacy mode switch could be added to allow for older setups?

julianh2o, 2015-04-21:
This seems to have changed recently in the esp8266/Arduino code... I was perhaps running an older version, in the latest version of the esp8266/Arduino it seems that there is agreement with esptool.

closed 2015-04-21 by julianh2o

duncan-a, 2015-04-21:
> This is also posted on esptool: themadinventor/esptool#29

I'm new to this, can't fathom out GitHub and am confused...

I'm using the 1.6.1-p1 ESP8266/Arduino from GitHub and have been manually
grounding GPIO0, then flicking the power on and off (then removing the GPIO
jumper and flicking power again after uploading) - I'm using only Tx, Rx
and the ground between the FTDI and the ESP, with CH_PD pulled to 3V3).

Are you saying that the GPIO0 and Reset can be done automatically - if so,
does it automatically release GPIO and perform another Reset after
programming?

Everything *was* working OK with my setup but I now only get gibberish from
the ESP in response to a reset and I'm suspecting the FTDI Friend has died
(another ESP performs the same) - it's probably unconnected but I'd value
any thoughts...

julianh2o, 2015-04-21:
I'm not sure what to say about the current issues you're having, but GPIO0 and Reset can be done automatically using an FTDI with both the DTR and the RTS pins exposed. These are data terminal output pins that can be set by software on the computer. (such as esptool and esp8266/Arduino)

probonopd, 2015-06-28:
It is working using DTR and RTS in principle now, but the Arduino IDE must be changed to make the Serial Monitor work; see https://github.com/esp8266/Arduino/issues/22#issuecomment-116276423